APS-USAXS/usaxs-bluesky-ended-2023 #4: configure Struck SIS3820 Multi-Channel scaler in ophyd

prjemian, 2018-02-22:
Q: Is there existing ophyd device support for the SIS3820?

@tacaswell writes;

See https://github.com/NSLS-II-TES/profile_collection/blob/c0b86877d5d4fa2a626472136fb9c3d41ae853c4/startup/25-sclr.py#L37

which merges a scalerMCA and scalerCH

It is possible that you just want https://github.com/NSLS-II-TES/profile_collection/blob/master/startup/25-sclr.py#L7

prjemian, 2018-02-22:
EPICS documentation: http://cars9.uchicago.edu/software/epics/mcaStruck.html

prjemian, 2018-06-01:
What about [`EpicsMCA`](https://nsls-ii.github.io/ophyd/builtin-devices.html#epicsmca-and-epicsdxp)

tacaswell, 2018-06-04:
@prjemian We should look at this while I am out in Chicago.

prjemian, 2018-06-04:
also 575c569

prjemian, 2018-06-04:
What else to do?

prjemian, 2018-07-20:
test next week, changing milestone

prjemian, 2018-08-02:
USAXS has `Struck3820` Device.  Closing for now,  Open new issue as needed.

closed 2018-08-02 by prjemian