APS-USAXS/usaxs-bluesky-ended-2023 #16: need autorange capability for amplifiers

prjemian, 2018-04-23:
each one gets autotuning (related to #9)

* one example: `USAXS/spec/macros/local/usaxs_diode.mac`, macro `def autorange_UPDI0I00`
* uses macros from `USAXS/spec/macros/local/usaxs_commands.mac`
* an `USAXS/spec/macros/local/usaxs_IonChambers.mac`

need to interface the amplifier's sequence program support into the amplifier device

* manual, auto, auto+bkg
* change requested range 
* background values are writables as well

[image]

this is another device that *will* be needed but not as soon as the first one

[image]

prjemian, 2018-05-23:
testing (now that IOC for ion chambers is booted):

```
In [3]: I0_struct.auto.reqrange.enum_strs
Out[3]: 
('1e5 low noise',
 '1e6 low noise',
 '1e7 low noise',
 '1e8 low noise',
 '1e9 low noise')
```

One assumption must be relaxed.  Need to adapt to this type of naming.

kmpeters, 2018-05-23:
`reqrange` is set by values 0-4 for all three auto-ranging sequence programs.  The string values of those states obviously differs based on which program is used.

prjemian, 2018-05-23:
Great!

prjemian, 2018-05-23:
@kmpeters, @jilavsky :  see how this works
```
In [1]: I0_struct.auto.setGain(1e7)
2: 1e7 low noise

In [2]: I0_struct.auto.setGain(1e6)
1: 1e6 low noise

In [3]: I0_struct.auto.setGain(1e5)
0: 1e5 low noise

In [4]: I0_struct.auto.setGain(1e7)
2: 1e7 low noise

In [5]: upd_struct.auto.setGain(2)
2: 1e8 V/A gain

In [6]: upd_struct.auto.setGain(1e8)
2: 1e8 V/A gain

In [7]: I00_struct.auto.reqrange.value
Out[7]: 4

In [8]: I00_struct.auto.reqrange.enum_strs
Out[8]: 
('1e5 low noise',
 '1e6 low noise',
 '1e7 low noise',
 '1e8 low noise',
 '1e9 low noise')

In [9]: I00_struct.auto.setGain(3)
3: 1e8 low noise

In [10]: I00_struct.auto.setGain(1e6)
1: 1e6 low noise

In [11]: I00_struct.auto.setGain("1e9 low noise")
4: 1e9 low noise
```

prjemian, 2018-05-23:
measure backgrounds:

```
In [2]: upd_controls.measure_dark_currents()
Measure dark current backgrounds for: PD_USAXS
gain = 1e4 V/A, dark current = 13.0 +/- 0.0
gain = 1e6 V/A, dark current = 12.125 +/- 0.33071891388307384
gain = 1e8 V/A, dark current = 25.5 +/- 1.118033988749895
gain = 1e10 V/A, dark current = 1051.0 +/- 59.508402768012516
gain = 1e12 V/A, dark current = 107052.0 +/- 10865.248858171633

In [3]: I0_controls.measure_dark_currents()
Measure dark current backgrounds for: I0_USAXS
gain = 1e5 V/A, dark current = 30552.125 +/- 0.33071891388307384
gain = 1e6 V/A, dark current = 30552.375 +/- 0.4841229182759271
gain = 1e7 V/A, dark current = 30552.5 +/- 0.5
gain = 1e8 V/A, dark current = 30552.875 +/- 0.33071891388307384
gain = 1e9 V/A, dark current = 30552.625 +/- 0.4841229182759271

In [4]: I00_controls.measure_dark_currents()
Measure dark current backgrounds for: I00_USAXS
gain = 1e5 V/A, dark current = 31753.25 +/- 0.4330127018922193
gain = 1e6 V/A, dark current = 31752.125 +/- 1.0532687216470449
gain = 1e7 V/A, dark current = 31753.125 +/- 0.33071891388307384
gain = 1e8 V/A, dark current = 31752.75 +/- 0.4330127018922193
gain = 1e9 V/A, dark current = 31752.375 +/- 0.4841229182759271

In [5]: trd_controls.measure_dark_currents()
Measure dark current backgrounds for: TR diode
gain = 1e5 V/A, dark current = 1.0 +/- 0.0
gain = 1e6 V/A, dark current = 1.0 +/- 0.0
gain = 1e7 V/A, dark current = 673.375 +/- 70.82184955929914
gain = 1e8 V/A, dark current = 8837.5 +/- 2.3979157616563596
gain = 1e9 V/A, dark current = 89335.625 +/- 75.62231400188703

```

prjemian, 2018-05-23:
autoscale result

```
In [1]: upd_controls.autoscale()
Autoscaling for: PD_USAXS
gain = 1e10 V/A
gain = 1e12 V/A
gain = 1e10 V/A
gain = 1e12 V/A
gain = 1e10 V/A

In [2]: I0_controls.autoscale()
Autoscaling for: I0_USAXS
gain = 1e9 V/A
gain = 1e9 V/A
gain = 1e9 V/A
gain = 1e9 V/A
gain = 1e9 V/A

In [3]: I00_controls.autoscale()
Autoscaling for: I00_USAXS
gain = 1e9 V/A
gain = 1e9 V/A
gain = 1e9 V/A
gain = 1e9 V/A
gain = 1e9 V/A

In [4]: trd_controls.autoscale()
Autoscaling for: TR diode
gain = 1e9 V/A
gain = 1e9 V/A
gain = 1e9 V/A
gain = 1e9 V/A
gain = 1e9 V/A
```

prjemian, 2018-05-23:
@jilavsky : What other features remain to be migrated from SPEC macros?

jilavsky, 2018-05-24:
Not sure if it got to you from web site.

spec autoscales ALL of these amplifiers at once to save time.

we have two versions of similar code:
autoscale UPD+I0+I00+TRD
autoscale I0+I00+TRD

autoscaling sequentially will take more time.

Sincerely

Jan

prjemian, 2018-05-25:
@jilavsky : this is mostly ready to test

```
def autoscale_amplifiers(controls, shutter=None, count_time=1.0)
def measure_background(controls, shutter=None, count_time=1.0, num_readings=8)

controls_list_I0_I00_TRD = [I0_controls, I00_controls, trd_controls]
controls_list_UPD_I0_I00_TRD = [upd_controls] + controls_list_I0_I00_TRD

autoscale_amplifiers(controls_list_UPD_I0_I00_TRD, shutter=mono_shutter, count_time=0.5)
autoscale_amplifiers(controls_list_I0_I00_TRD, mono_shutter, 1.0)
measure_background(controls_list_UPD_I0_I00_TRD, mono_shutter, 1.0, num_readings=8)
```

prjemian, 2018-05-25:
it *has* been tested at the beam line

prjemian, 2018-05-25:
one more feature to implement:

```
#now find the right gain. Use loop until we do not change AND we are not topped up. 
```

Loop is limited to at most 9 iterations.  Arbitrary number?  Why not related to number of allowed gains?

prjemian, 2018-05-25:
```
In [9]: autoscale_amplifiers(controls_list_UPD_I0_I00_TRD)

In [10]: _last_autorange_gain_
Out[10]: 
{'9idcLAX:vsc:c0': {'9idcLAX:pd01:seq02:': 10000000000.0,
  '9idcLAX:pd02:seq01:': 1000000000.0,
  '9idcLAX:pd03:seq01:': 1000000000.0,
  '9idcLAX:pd05:seq01:': 1000000000.0}}
```

jilavsky, 2018-05-25:
This. “9" is tested number.

Amplifiers have recovery time if we accidentally put them in overload. There is no easy way to test for overload. So if you simply limit to number of ranges only, and put it in overload, you get wrong range.

I tested that when we use 9 iterations AND the counts/sleep times, we seem to get correct range practically always. I am sure one can get smarter.

But 9 cycles will be used ONLY in rare cases of this overload. By caching the last known range, at least 80% of times we pass through this with only one or two iterations. We run this _very_ often. Every second here counts many,many times…

This design took me hell of a lot of testing and trying. Simpler designs were either slow or very error prone. Spec code is complex, but finally, it works…

Sincerely

Jan

prjemian, 2018-05-25:
OK, 9 it is, and <950kc/s, in 1c80c4a 

* [at most 9 iterations](https://github.com/APS-USAXS/ipython-usaxs/commit/1c80c4a727aafd8161f841a20173ebc83074e077#diff-e08476e9ed1126bc4c7a01d2a8a426a4R334) - done
* [count rate <= 950,000 c/s](https://github.com/APS-USAXS/ipython-usaxs/commit/1c80c4a727aafd8161f841a20173ebc83074e077#diff-e08476e9ed1126bc4c7a01d2a8a426a4R162) - done
* [amplifier gain settling time](https://github.com/APS-USAXS/ipython-usaxs/commit/1c80c4a727aafd8161f841a20173ebc83074e077#diff-e08476e9ed1126bc4c7a01d2a8a426a4R54) - not yet implemented

prjemian, 2018-05-26:
working, open new issue if further work is identified

closed 2018-05-26 by prjemian