APS-USAXS/usaxs-bluesky-ended-2023 #44: USAXS step scan written to NeXus file

prjemian, 2018-07-25:
this is the future, for now use the SPEC file

prjemian, 2018-11-28:
do this work as part of or following #8

prjemian, 2019-05-31:
Since it will be *the future* and operations starts *real soon now*, put this off to the next milestone.

jilavsky, 2020-05-28:
Is ti time to address this now? Or too late for Saturday?

prjemian, 2020-05-28:
Needs maybe a week of my attention.  Not realistic to have it ready for the weekend.

prjemian, 2020-06-11:
Can't use [NXsas](https://manual.nexusformat.org/classes/applications/NXsas.html#nxsas) for raw data from the USAXS instrument.  NXsas is written for

> raw, monochromatic 2-D SAS data with an area detector

and the requirements of its detector group are wholly unsuited for raw USAXS data.

It is not required to use an application definition when writing raw USAXS data.  Instead, we'll write this data using only the NeXus base classes.

prjemian, 2020-06-11:
In addition to its (opinion: unnecessary and short-sighted) restriction to 2-D detectors, there are sufficient [additional](https://github.com/nexusformat/definitions/issues/522) [concerns](https://github.com/nexusformat/definitions/issues/523) about NXsas that make it easy to drop our use of it.

prjemian, 2020-06-12:
@jilavsky : Here is a first cut at a NeXus/HDF5 file for *uascan* data.  It uses links all over to make the data show up in the correct NeXus structure.  If you know where to look in NeXus, I expect you should find most things.  Otherwise, these are the largish containers for the measured data.

h5 address | description
:--- | :---
`/entry/instrument/bluesky_metadata` | all the metadata (`md`) in the *start* document (complicated metadata structures, such as python dictionaries, are written as json text)
`/entry/instrument/bluesky_streams/primary` | the actual measurements
`/entry/instrument/bluesky_streams/baseline` | other PVs recorded at start and end of scan
`/entry/instrument/bluesky_streams/aps_current_monitor` | storage ring updates, separate from *primary* stream

note: When I turned compression ON (gzip or lzf), the HDF5 file size *grew* by 50%!

Most of the code that writes this is agnostic to what kind of scan we are running, although it was developed only by looking at *one* uascan measurement.  So, it might be rough around the edges.

I know we need:

* indication of the primary signals to plot on X & Y axes

What other data have we missed (and not recorded in bluesky)?

prjemian, 2020-06-12:
@jilavsky : the file:  
[test_raw_usaxs.zip](https://github.com/APS-USAXS/ipython-usaxs/files/4767854/test_raw_usaxs.zip)

prjemian, 2020-06-12:
Switching from [json to yaml](https://stackoverflow.com/questions/1726802/what-is-the-difference-between-yaml-and-json) for writing complex metadata structures.  It's easier on the eyes and easy to process.

Illustrate this with a python dictionary, showing software versions in use:

```
In [13]: d                                                                                                                                  
Out[13]: 
{'apstools': '1.2.4rc3',
 'bluesky': '1.6.0',
 'databroker': '1.0.2',
 'epics': '3.4.2',
 'matplotlib': '3.2.1',
 'numpy': '1.18.5',
 'ophyd': '1.5.0',
 'pyRestTable': '2020.0.2',
 'pymongo': '3.9.0',
 'spec2nexus': '2021.1.7'}
```

*json*

```
In [14]: json.dumps(d)                                                                                                                      
Out[14]: '{"apstools": "1.2.4rc3", "bluesky": "1.6.0", "databroker": "1.0.2", "epics": "3.4.2", "matplotlib": "3.2.1", "numpy": "1.18.5", "ophyd": "1.5.0", "pyRestTable": "2020.0.2", "pymongo": "3.9.0", "spec2nexus": "2021.1.7"}'
```

*yaml*

```
In [15]: yaml.dump(d)                                                                                                                       
Out[15]: 'apstools: 1.2.4rc3\nbluesky: 1.6.0\ndatabroker: 1.0.2\nepics: 3.4.2\nmatplotlib: 3.2.1\nnumpy: 1.18.5\nophyd: 1.5.0\npyRestTable: 2020.0.2\npymongo: 3.9.0\nspec2nexus: 2021.1.7\n'
```

prjemian, 2020-06-15:
While this works for `uascan`, continue to develop the `NXWriterBase` to support use of area detector.  Need to test with more examples.  The development may impact how the `uascan` support is encoded and improve use for more USAXS data acquisition plans.

prjemian, 2020-06-15:
`Further testing of `NXWriterBase` needs more variations of area detector acquisition than WAXS and SAXS.  Work on this offline from the USAXS instrument.

prjemian, 2020-06-18:
Tested today at instrument.  Works for `uascan`!

closed 2020-06-18 by prjemian