APS-USAXS/usaxs-bluesky-ended-2023 #100: compress record arrays returned as floats by PyEpics

prjemian, 2018-12-08:
As noted [previously](https://github.com/APS-USAXS/ipython-usaxs/issues/45#issuecomment-445406763), PVs of record type [compress](https://epics.anl.gov/EpicsDocumentation/AppDevManuals/RecordRef/Recordref-15.html) are being returned as float rather than numpy.ndarray.  This locks up the data processing stream.

Seems to be related to the PyEpics version.  Success: Python 2.7.15 and PyEpics 3.2.5, Fail: Python 3.6.5 and PyEpics 3.3.1

Here is PyEpics code (at 9-ID-C) to test:

```
import epics
epics.__version__
pv = epics.PV("9idcLAX:USAXSfly:I0:ampReqGain")
arr = pv.get()
type(arr)
arr
print arr
```

prjemian, 2018-12-08:
The plot thickens with a data point from another workstation:

## example EPICS compress record

### database

```
# file: compress.db

record(ao, "signal") {
    field(FLNK, "compress")
}

record(compress, "compress") {
    field(INP, "signal")
    field(ALG, "Circular Buffer")
    field(NSAM, 5)
    field(N, 1)
}
```

### EPICS softIoc

```
mintadmin@mint-vm ~ $ softIoc -d compress.db 
Starting iocInit
############################################################################
## EPICS R3.15.5
## EPICS Base built Jan 19 2017
############################################################################
iocRun: All initialization complete
epics> dbl
compress
signal
epics> 
```

### pump some numbers into the `compress` record

```
caput signal 1
caput signal 2
caput signal 3
caput signal 4
caput signal 5
caput signal 6
```

Then, verify the circular buffer has sorted these chronolgically

    mintadmin@mint-vm ~ $ caget compress
    compress 5 2 3 4 5 6

## Python 3.6.6 and PyEpics 3.3.1: success
```
mintadmin@mint-vm ~ $ use_bluesky.sh 
BlueSky executable path: /home/mintadmin/Apps/BlueSky/bin
Python 3.6.6 :: Anaconda custom (64-bit)
conda 4.5.11
pip 18.1 from /home/mintadmin/Apps/BlueSky/lib/python3.6/site-packages/pip (python 3.6)
starting ipython with profile: bluesky
Python 3.6.6 |Anaconda custom (64-bit)| (default, Jun 28 2018, 17:14:51) 
Type 'copyright', 'credits' or 'license' for more information
IPython 6.4.0 -- An enhanced Interactive Python. Type '?' for help.

IPython profile: bluesky
/home/mintadmin/.ipython/profile_bluesky/startup/00-0-checks.py
BlueSky version: 1.4.1
Ophyd version: 1.3.0

...
In [1]: pv = EpicsSignal("compress", name="pv")

In [2]: arr = pv.get()

In [3]: type(arr)
Out[3]: numpy.ndarray

In [4]: arr
Out[4]: array([ 2.,  3.,  4.,  5.,  6.])

In [5]: epics.__version__
Out[5]: '3.3.1'

In [6]: !caget compress.RTYP
compress.RTYP                  compress

In [7]: !caget compress.NSAM
compress.NSAM                  5

In [8]: !caget compress
compress 5 2 3 4 5 6
```

prjemian, 2018-12-08:
Needs more diagnosis on the usaxscontrol workstation, then.  Should not affect operations at all.

prjemian, 2018-12-10:
observation

```
/APSshare/anaconda3/BlueSky/bin/conda create --clone base --name local_bs
Source:      /APSshare/anaconda3/BlueSky
Destination: /share1/bluesky/conda_envs/local_bs
The following packages cannot be cloned out of the root environment:
 - conda-4.5.11-py36_0
 - conda-env-2.6.0-h36134e3_1
Packages: 195
Files: 244

Downloading and Extracting Packages
...

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
#
# To activate this environment, use:
# > source activate local_bs
#
# To deactivate an active environment, use:
# > source deactivate
#

usaxs@usaxscontrol .../bluesky/conda_envs $ source /APSshare/anaconda3/BlueSky/bin/activate  local_bs
(/share1/bluesky/conda_envs/local_bs) usaxs@usaxscontrol .../bluesky/conda_envs $ ipython
Python 3.6.5 |Anaconda, Inc.| (default, Apr 29 2018, 16:14:56)
Type 'copyright', 'credits' or 'license' for more information
IPython 6.4.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import epics

In [2]: pv = epics.PV("9idcLAX:USAXSfly:I0:ampReqGain")

In [3]: arr = pv.get()

In [4]: arr
Out[4]: array([], dtype=float64)

In [5]: pv = epics.PV("compress")

In [6]: arr = pv.get()

In [7]: arr
Out[7]: array([ 2.1,  3. ,  4. ,  5. ,  6. ])

In [8]: epics.__version__
Out[8]: '3.3.1'

In [9]:
```

problem is **not** with PyEpics version but related to the `"9idcLAX:USAXSfly:I0:ampReqGain"` PV itself ... time to check the others

prjemian, 2018-12-10:
from simple testing tools, don't see this problem (something specific to the code when it is called in association with a fly scan)

```
usaxs@usaxscontrol ~/.../startup/usaxs_support $ /APSshare/anaconda3/BlueSky/bin/python ./bs_100.py  | grep -i reqgain
9idcLAX:USAXSfly:I0:ampReqGain <class 'numpy.ndarray'> []
9idcLAX:USAXSfly:I00:ampReqGain <class 'numpy.ndarray'> []
9idcLAX:USAXSfly:DLPCA200:ampReqGain <class 'numpy.ndarray'> []
9idcLAX:USAXSfly:DDPCA300:ampReqGain <class 'numpy.ndarray'> []
```

prjemian, 2018-12-10:
other testing tool

```
usaxs@usaxscontrol ~/.../startup/usaxs_support $ /APSshare/anaconda3/BlueSky/bin/python ./tester_100.py | grep -i reqgain
9idcLAX:USAXSfly:I0:ampReqGain <class 'numpy.ndarray'>
9idcLAX:USAXSfly:I00:ampReqGain <class 'numpy.ndarray'>
9idcLAX:USAXSfly:DLPCA200:ampReqGain <class 'numpy.ndarray'>
9idcLAX:USAXSfly:DDPCA300:ampReqGain <class 'numpy.ndarray'>
```

prjemian, 2018-12-10:
inspection of the HDF5 file written by the software used to save fly scan data:

```
usaxs@usaxscontrol ~/.../startup/usaxs_support $ punx tree test_100.hdf5 | grep -i reqgain
      changes_DDPCA300_ampReqGain:NX_FLOAT64[0] = []
        @epics_pv = 9idcLAX:USAXSfly:DDPCA300:ampReqGain
      changes_DLPCA200_ampReqGain:NX_FLOAT64[0] = []
        @epics_pv = 9idcLAX:USAXSfly:DLPCA200:ampReqGain
      changes_I00_ampReqGain:NX_FLOAT64[0] = []
        @epics_pv = 9idcLAX:USAXSfly:I00:ampReqGain
      changes_I0_ampReqGain:NX_FLOAT64[0] = []
        @epics_pv = 9idcLAX:USAXSfly:I0:ampReqGain
```

prjemian, 2018-12-10:
code in 2aa76de

prjemian, 2018-12-17:
# testing at beam line

```
In [12]: pv = EpicsSignal("9idcLAX:USAXSfly:I0:ampReqGain", name="pv")

In [13]: pv.read()
Out[13]: {'pv': {'value': array([ 2.]), 'timestamp': 1545055051.069777}}

In [14]: pv.value
Out[14]: array([ 2.])
```

While there is only one value in the array, the array is reported by ophyd properly, as an array of one value.

prjemian, 2018-12-17:
... more ...

```
In [15]: pv = EpicsSignal("9idcLAX:USAXSfly:DLPCA200:ampReqGain", name="pv")

In [16]: pv.value
Out[16]: array([ 4.])

In [17]: pv = EpicsSignal("9idcLAX:USAXSfly:DDPCA300:ampReqGain", name="pv")

In [18]: pv.value
Out[18]: array([ 2.,  1.,  1.,  0.,  0.,  1.,  1.,  2.,  2.,  3.,  3.,  4.,  4.])
```

prjemian, 2018-12-17:
the main fly scan data array:

```
In [2]: struck.mca3
Out[2]: EpicsMCARecord(prefix='9idcLAX:3820:mca3', name='struck_mca3', parent='struck', read_attrs=['preset_real_time', 'elapsed_real_time', 'spectrum'], configuration_attrs=['preset_real_time'])

In [3]: struck.mca3.spectrum.value
Out[3]: array([2030, 2110, 1995, ...,  774,  792,  805], dtype=int32)
```

# Summary

The best way forward is to refactor the `usaxs_support.saveFlyData.SaveFlyScan()` code into standard ophyd objects and save that data at the end of a Bluesky USAXS fly scan plan.

prjemian, 2018-12-17:
## observation

Two of the `compress` records behave differently

PV | PyEpics return type | value
---- | ---- | ----
`9idcLAX:USAXSfly:DLPCA200:ampReqGain` | float | 4.0
`9idcLAX:USAXSfly:DDPCA300:ampReqGain` | numpy.ndarray | array([ 2.,  1.,  1.,  0.,  0.,  1.,  1.,  2.,  2.,  3.,  3.,  4.,  4.])

```
In [1]: sfs = SaveFlyScan("/tmp/sfs.h5", config_file="usaxs_support/saveFlyData.xml")
   ...: sfs.preliminaryWriteFile()
   ...: sfs.saveFile()
   ...: 
   ...: 

In [8]: !/APSshare/anaconda3/x86_64/bin/punx tree /tmp/sfs.h5 | grep -i req
/APSshare/anaconda3/x86_64/lib/python3.5/site-packages/h5py/__init__.py:36: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
      changes_DDPCA300_ampReqGain:NX_FLOAT64[13] = [2.0, 1.0, 1.0, '...', 4.0]
        @epics_pv = 9idcLAX:USAXSfly:DDPCA300:ampReqGain
      changes_DLPCA200_ampReqGain:NX_FLOAT64 = 4.0
        @epics_pv = 9idcLAX:USAXSfly:DLPCA200:ampReqGain
      changes_I00_ampReqGain:NX_FLOAT64 = 4.0
        @epics_pv = 9idcLAX:USAXSfly:I00:ampReqGain
      changes_I0_ampReqGain:NX_FLOAT64 = 2.0
        @epics_pv = 9idcLAX:USAXSfly:I0:ampReqGain
      mca_clock_frequency:NX_FLOAT64 = 50000000.0
        @epics_pv = 9idcLAX:3820:clock_frequency
        @epics_description = clock freq for struck

In [9]: pv = epics.PV("9idcLAX:USAXSfly:DDPCA300:ampReqGain")

In [10]: pv
Out[10]: <PV '9idcLAX:USAXSfly:DDPCA300:ampReqGain', count=13/300, type=time_double, access=read/write>

In [11]: !caget 9idcLAX:USAXSfly:DDPCA300:ampReqGain.RTYP
9idcLAX:USAXSfly:DDPCA300:ampReqGain.RTYP compress

In [12]: pv = epics.PV("9idcLAX:USAXSfly:DLPCA200:ampReqGain")

In [13]: pv
Out[13]: <PV '9idcLAX:USAXSfly:DLPCA200:ampReqGain', count=1, type=time_double, access=read/write>

In [14]: !caget 9idcLAX:USAXSfly:DLPCA200:ampReqGain.RTYP
9idcLAX:USAXSfly:DLPCA200:ampReqGain.RTYP compress

```

kmpeters, 2018-12-17:
Testing multiple combinations of python & pyepics:

```
usaxs@usaxscontrol /share1/USAXS_data $ /APSshare/anaconda/x86_64/bin/python
Python 2.7.15 |Anaconda custom (64-bit)| (default, Oct 10 2018, 21:32:13) 
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import epics
>>> epics.__version__
'3.2.5'
>>> epics.caget("9idcLAX:USAXSfly:DDPCA300:ampReqGain")
array([ 2.,  1.,  1.,  0.,  0.,  1.,  1.,  2.,  2.,  3.,  3.,  4.,  4.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.])
>>> epics.caget("9idcLAX:USAXSfly:DLPCA200:ampReqGain")
array([ 4.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,
        0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.])
>>> quit()
```

```
usaxs@usaxscontrol /share1/USAXS_data $ /APSshare/anaconda3/x86_64/bin/python
Python 3.5.4 |Anaconda custom (64-bit)| (default, Nov 20 2017, 18:44:38) 
[GCC 7.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import epics
>>> epics.__version__
'3.2.6'
>>> epics.caget("9idcLAX:USAXSfly:DDPCA300:ampReqGain")
array([2., 1., 1., 0., 0., 1., 1., 2., 2., 3., 3., 4., 4.])
>>> epics.caget("9idcLAX:USAXSfly:DLPCA200:ampReqGain")
4.0
>>> quit()
```

```
usaxs@usaxscontrol /share1/USAXS_data $ /APSshare/anaconda3/BlueSky/bin/ipython
Python 3.6.5 |Anaconda, Inc.| (default, Apr 29 2018, 16:14:56) 
Type 'copyright', 'credits' or 'license' for more information
IPython 6.4.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import epics

In [2]: epics.__version__
   ...: 
Out[2]: '3.3.1'

In [3]: epics.caget("9idcLAX:USAXSfly:DDPCA300:ampReqGain")
Out[3]: array([ 2.,  1.,  1.,  0.,  0.,  1.,  1.,  2.,  2.,  3.,  3.,  4.,  4.])

In [4]: epics.caget("9idcLAX:USAXSfly:DLPCA200:ampReqGain")
Out[4]: 4.0

In [5]: quit()
```

prjemian, 2018-12-17:
@kmpeters Thanks.  Summarizing: the problem can be reproduced using only PyEpics and Python

Python | PyEpics | PV | type(value) | correct?
---- | ---- | ---- | ---- | ----
2.7.15 | 3.2.5 | `9idcLAX:USAXSfly:DLPCA200:ampReqGain` | float | Ok
2.7.15 | 3.2.5 | `9idcLAX:USAXSfly:DDPCA300:ampReqGain` | numpy.ndarray | Ok
3.5.4 | 3.2.6 | `9idcLAX:USAXSfly:DLPCA200:ampReqGain` | float | **WRONG**
3.5.4 | 3.2.6 | `9idcLAX:USAXSfly:DDPCA300:ampReqGain` | numpy.ndarray | Ok
3.6.5 | 3.3.1 | `9idcLAX:USAXSfly:DLPCA200:ampReqGain` | float | **WRONG**
3.6.5 | 3.3.1 | `9idcLAX:USAXSfly:DDPCA300:ampReqGain` | numpy.ndarray | Ok

kmpeters, 2018-12-17:
I wanted to test the as_numpy argument to caget, and I found that my first caget of the one-element array behaved differently than subsequent ones.  Also, the as_numpy argument was not helpful.

```
Python 3.6.5 |Anaconda, Inc.| (default, Apr 29 2018, 16:14:56) 
[GCC 7.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import epics
>>> epics.__version__
'3.3.1'
>>> epics.caget("9idcLAX:USAXSfly:DLPCA200:ampReqGain")
array([ 4.])
>>> epics.caget("9idcLAX:USAXSfly:DLPCA200:ampReqGain", as_numpy=True)
4.0
>>> epics.caget("9idcLAX:USAXSfly:DLPCA200:ampReqGain", as_numpy=False)
4.0
>>> epics.caget("9idcLAX:USAXSfly:DLPCA200:ampReqGain")
4.0
>>> epics.caget("9idcLAX:USAXSfly:DLPCA200:ampReqGain")
4.0
>>> epics.caget("9idcLAX:USAXSfly:DLPCA200:ampReqGain.NUSE")
1.0
>>> 
```

If NUSE is set to 2, do the cagets start behaving correctly?

kmpeters, 2018-12-17:
> I found that my first caget of the one-element array behaved differently than subsequent ones.

I'm also not able to reproduce this.

prjemian, 2018-12-17:
`NUSE` is the number of items currently in the array.  It cannot be modified by the CA client.  See  https://epics.anl.gov/EpicsDocumentation/AppDevManuals/RecordRef/Recordref-15.html#HEADING15-44

I also tested cases where `NUSE=1`, hypothesizing that might be a special case (compared with `NUSE>1`), no difference.

prjemian, 2018-12-18:
# work-in-progress

Trying to find the combination of PyEpics and libca libraries that works with Python 3.6.5 but not having success.  This short session reveals the problem with this investigation:

```
usaxs@usaxscontrol .../USAXS_data/test $ export PYEPICS_LIBCA=
usaxs@usaxscontrol .../USAXS_data/test $ conda create -n py365 -y python=3.6.5
Solving environment: done

## Package Plan ##

  environment location: /share1/bluesky/conda_envs/py365

  added / updated specs: 
    - python=3.6.5

The following NEW packages will be INSTALLED:

    ca-certificates: 2018.03.07-0           
    certifi:         2018.11.29-py36_0      
    libedit:         3.1.20170329-h6b74fdf_2
    libffi:          3.2.1-hd88cf55_4       
    libgcc-ng:       8.2.0-hdf63c60_1       
    libstdcxx-ng:    8.2.0-hdf63c60_1       
    ncurses:         6.1-he6710b0_1         
    openssl:         1.0.2p-h14c3975_0      
    pip:             18.1-py36_0            
    python:          3.6.5-hc3d631a_2       
    readline:        7.0-h7b6447c_5         
    setuptools:      40.6.3-py36_0          
    sqlite:          3.26.0-h7b6447c_0      
    tk:              8.6.8-hbc83047_0       
    wheel:           0.32.3-py36_0          
    xz:              5.2.4-h14c3975_4       
    zlib:            1.2.11-h7b6447c_3      

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
#
# To activate this environment, use
#
#     $ conda activate py365
#
# To deactivate an active environment, use
#
#     $ conda deactivate

usaxs@usaxscontrol .../USAXS_data/test $ source /APSshare/anaconda/x86_64/bin/activate py365
(/share1/bluesky/conda_envs/py365) usaxs@usaxscontrol .../USAXS_data/test $ 
(/share1/bluesky/conda_envs/py365) usaxs@usaxscontrol .../USAXS_data/test $ pip install git+git://github.com/pyepics/pyepics@master
Collecting git+git://github.com/pyepics/pyepics@master
  Cloning git://github.com/pyepics/pyepics (to revision master) to /tmp/pip-req-build-39c_hsp3
Building wheels for collected packages: pyepics
  Running setup.py bdist_wheel for pyepics ... done
  Stored in directory: /tmp/pip-ephem-wheel-cache-3td066cn/wheels/35/19/2e/81dcbf7355f681b137b8f8e4eb7a5daa698d66ac66c7371237
Successfully built pyepics
Installing collected packages: pyepics
Successfully installed pyepics-3.3.2rc1+6.g0051a29
(/share1/bluesky/conda_envs/py365) usaxs@usaxscontrol .../USAXS_data/test $ ./test_100.py
Python version: 3.6.5 |Anaconda, Inc.| (default, Apr 29 2018, 16:14:56) 
[GCC 7.2.0]
PyEpics version: 3.3.2rc1+6.g0051a29
libca version: /share1/bluesky/conda_envs/py365/lib/python3.6/site-packages/epics/clibs/linux64/libca.so
[4.0]
```
Repeating that test *immediately* (without changing any code or PV values) produces a different result!  This is consistent with the [previous](https://github.com/APS-USAXS/ipython-usaxs/issues/100#issuecomment-447976685) report by @kmpeters.

```
(/share1/bluesky/conda_envs/py365) usaxs@usaxscontrol .../USAXS_data/test $ ./test_100.py
Python version: 3.6.5 |Anaconda, Inc.| (default, Apr 29 2018, 16:14:56) 
[GCC 7.2.0]
PyEpics version: 3.3.2rc1+6.g0051a29
libca version: /share1/bluesky/conda_envs/py365/lib/python3.6/site-packages/epics/clibs/linux64/libca.so
4.0
```

prjemian, 2018-12-18:
Tried various *different* libca libraries but could not produce an array output consistently.

prjemian, 2018-12-18:
actual differences:

field | DDPCA300 | DLPCA200
---- | ---- | ----
BPTR | 0x8256b50 | 0x82589b0
INP | DB_LINK 9idcLAX:USAXSfly:DDPCA300:recordGain.C NPP NMS   | DB_LINK 9idcLAX:USAXSfly:DLPCA200:recordGain.C NPP NMS  
LSET | 0x98a7f50 | 0x98a7f80
MLOK | 08 25 6b 20 | 08 25 89 80
NAME | 9idcLAX:USAXSfly:DDPCA300:ampReqGain | 9idcLAX:USAXSfly:DLPCA200:ampReqGain
NUSE | 13 | 1
OFF | 13 | 1
OUSE | 13 | 1
TIME | 2018-12-17 07:58:01.636443100 | 2018-12-17 07:57:31.069777656
WPTR | 0x4fbe2b0 | 0x4fe1ef0

Real differences: none that a CA client should treat them as different data types.  DDPCA300 has 13 elements in use while DLCPA200 only has one.  That is typical.

kmpeters, 2018-12-18:
I added a compress record to my development soft-ioc (synApps_5_7, base-3.14.12.3) and requested cainfo of the compress record from python (v3.6.5, /APSshare/anaconda3/BlueSky/bin/python) and found consistent behavior.

If the number of elements in the compress record (NUSE) was 1, the cainfo reported that the value was a float and NELM=1:
```
>>> epics.cainfo("kmp3:compress2:ampGain")
== kmp3:compress2:ampGain  (time_double) ==
   value      = 7
   char_value = '7'
   count      = 1
   nelm       = 1
   type       = time_double
   units      = 
   precision  = 0
   host       = kmp-linux2.aps.anl.gov:41479
   access     = read/write
   status     = 0
   severity   = 0
   timestamp  = 1545167451.339 (2018-12-18 15:10:51.33914)
   posixseconds        = 1545167451.0
   nanoseconds= 339136762
   upper_ctrl_limit    = 0.0
   lower_ctrl_limit    = 0.0
   upper_disp_limit    = 0.0
   lower_disp_limit    = 0.0
   upper_alarm_limit   = nan
   lower_alarm_limit   = nan
   upper_warning_limit = nan
   lower_warning_limit = nan
   PV is internally monitored, with 0 user-defined callbacks:
=============================
```
If the number of elements in the compress record (NUSE) was greater than 1, the cainfo reported that the value was an array and NELM=300:
```
>>> epics.cainfo("kmp3:compress2:ampGain")
== kmp3:compress2:ampGain  (time_double) ==
   value      = array  [7,14]
   char_value = '<array size=2, type=time_double>'
   count      = 2
   nelm       = 300
   type       = time_double
   units      = 
   precision  = 0
   host       = kmp-linux2.aps.anl.gov:41479
   access     = read/write
   status     = 0
   severity   = 0
   timestamp  = 1545168642.370 (2018-12-18 15:30:42.37041)
   posixseconds        = 1545168642.0
   nanoseconds= 370413788
   upper_ctrl_limit    = 0.0
   lower_ctrl_limit    = 0.0
   upper_disp_limit    = 0.0
   lower_disp_limit    = 0.0
   upper_alarm_limit   = nan
   lower_alarm_limit   = nan
   upper_warning_limit = nan
   lower_warning_limit = nan
   PV is internally monitored, with 0 user-defined callbacks:
=============================
```

newville, 2018-12-18:
@prjemian @kmpeters   I'm not entirely sure I have absorbed all the details, so let me try to restate the issue and see if I'm getting it right.

It seems like this a case of a "one element array" vs "scalar".   That is, with `count` of 1, the value is being coerced to be a float scalar, even though `nelm` is set and bigger than 1  so that the value really should be an array.  Is that the main idea?   I do see where that we are using `count<=1` to test whether a PV is a scalar or an array.  I think that should be using `nelm` which will probably fix this problem.   

For working libca for Linuxes, I just compiled (and now included in github master) shared libs from base 3.16.2 build on Centos 6.9 (gcc 4.4.7, glib 2.12).  I'd be curious to know if those work for you, or what we should include so that it "just works" on APSshare.

prjemian, 2018-12-19:
@newville Your summary is correct.  This issue has a lot of cruft.  The situation may be more clear on issue #102.

prjemian, 2018-12-20:
We can install PyEpics from GiHub master branch once https://github.com/pyepics/pyepics/pull/139 is merged.

prjemian, 2019-01-24:
This problem is now tested and resolved.  Thanks @newville!

```
(base) usaxs@usaxscontrol ~/Documents/jemian $ ipython
Python 3.6.6 |Anaconda, Inc.| (default, Oct  9 2018, 12:34:16) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.2.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: !caget compress                                                                                                                       
compress 5 2 3 4 5 6

In [2]: import epics 
   ...: epics.__version__ 
   ...: pv = epics.PV("compress") 
   ...: arr = pv.get() 
   ...: print(type(arr), arr)                                                                                                                 
<class 'numpy.ndarray'> [2. 3. 4. 5. 6.]

```

prjemian, 2019-01-25:
It's back...  something with how the data is written?

prjemian, 2019-01-25:
Perhaps this problem is really solved.  With no beam, the arrays we want have garbage data (and only one value).  HDF5 regards an array of one element the same as a scalar.  Wait and see with real beam...

```
In [11]: f = h5py.File("/tmp/arrays.h5", "w")                                                                                                                                       

In [12]: f.create_dataset(name="float", data=1.234)                                                                                                                                 
Out[12]: <HDF5 dataset "float": shape (), type "<f8">

In [13]: f.create_dataset(name="arr0", data=[])                                                                                                                                     
Out[13]: <HDF5 dataset "arr0": shape (0,), type "<f8">

In [14]: f.create_dataset(name="arr1", data=[1])                                                                                                                                    
Out[14]: <HDF5 dataset "arr1": shape (1,), type "<i8">

In [15]: f.create_dataset(name="arr2", data=[1, 2])                                                                                                                                 
Out[15]: <HDF5 dataset "arr2": shape (2,), type "<i8">

In [16]: f.create_dataset(name="arr3", data=[1, 2, 3])                                                                                                                              
Out[16]: <HDF5 dataset "arr3": shape (3,), type "<i8">

In [17]: f.close()                                                                                                                                                                  

In [18]: exit                                                                                                                                                                       
(base) usaxs@usaxscontrol ~/.../startup/usaxs_support $ cp /tmp/arrays.h5 /share1/USAXS_data/test/
(base) usaxs@usaxscontrol ~/.../startup/usaxs_support $ punx tree /tmp/arrays.h5

!!! WARNING: this program is not ready for distribution.

/tmp/arrays.h5
  arr0:float64[0] = []
  arr1:int64 = 1
  arr2:int64[2] = [1, 2]
  arr3:int64[3] = [1, 2, 3]
  float:float64[] = 
```

closed 2019-01-25 by prjemian