APS-USAXS/usaxs-bluesky-ended-2023 #102: PyEpics and compress record with only one element

prjemian, 2018-12-18:
Issue #100 shows problems with reading several EPICS [compress](https://epics.anl.gov/EpicsDocumentation/AppDevManuals/RecordRef/Recordref-15.html) records as arrays.  One common factor is that the problems show up when the compress record has one element, where the record is returned from PyEpics as type float rather than numpy.ndarray.

Test PyEpics with a compress record where the number of elements in the array varies from zero up to NSAM.

prjemian, 2018-12-18:
test database

```
# file: compress.db

record(ao, "signal") {
    field(FLNK, "compress")
}

record(compress, "compress") {
    field(INP, "signal")
    field(ALG, "Circular Buffer")
    field(NSAM, 5)
    field(N, 1)
}
```

run this in a softIoc (same as first two steps shown [previously](https://github.com/APS-USAXS/ipython-usaxs/issues/100#issuecomment-445473110)):

```
usaxs@usaxscontrol .../test/issue_100 $ softIoc -d compress.db
Starting iocInit
############################################################################
## EPICS R3.14.12.3-ext1 $Date: 2012/12/20 19:45:40 $
## EPICS Base built Jul 10 2013
############################################################################
iocRun: All initialization complete
epics> dbl
signal
compress
epics>
```

prjemian, 2018-12-18:
test the compress record value using PyEpics:

```python
#!/usr/bin/env python

import epics, sys
print("Python version:", sys.version)
print("PyEpics version:", epics.__version__)
print("libca version:", epics.ca.find_libca())
for field in ("N", "NSAM", "NUSE", "VAL"):
    pvname = "compress.%s" % field
    print(pvname, epics.caget(pvname))
```

prjemian, 2018-12-18:
Demonstrates the problem clearly for Python 3 and is consistent with [report](https://github.com/APS-USAXS/ipython-usaxs/issues/100#issuecomment-448387237) of @kmpeters.  When a compress record has only one element (NSAM=5, NUSE=1), PyEpics (in python 3) reports it as a single value rather than an array.  Using Python 2, it is always reported as an array.

Other tests on this are described in #100 and #101.

@newville - Any ideas how we can resolve this?

kmpeters, 2018-12-18:
My compress record has NSAM=300 and NUSE=1.

If I use pdb to debug a call to epics.caget() on the compress record, I see the following while inside pv.py:get():
```
(Pdb) p self._args
{'pvname': 'kmp3:compress2:ampGain', 'value': 14.0, 'char_value': '14.0', 'status': 0, 'ftype': 20, 'chid': 94739566223744, 'host': 'kmp-linux2.aps.anl.gov:41479', 'count': 1, 'access': 'read/write', 'write_access': True, 'read_access': True, 'severity': 0, 'timestamp': 1545173541.959358, 'posixseconds': 1545173541.0, 'nanoseconds': 959358080, 'precision': None, 'units': None, 'enum_strs': None, 'upper_disp_limit': None, 'lower_disp_limit': None, 'upper_alarm_limit': None, 'lower_alarm_limit': None, 'lower_warning_limit': None, 'upper_warning_limit': None, 'upper_ctrl_limit': None, 'lower_ctrl_limit': None, 'nelm': 300, 'type': 'time_double', 'typefull': 'time_double'}
```
It appears that only the count is taken into account to determine whether or not the value is an array, even though the nelm is known:
https://github.com/pyepics/pyepics/blob/498d803be41e43921bd787f95d88e20eb11a4d72/epics/pv.py#L425

prjemian, 2018-12-19:
I can try testing with libca.so from 3.16.2.  I've already tested with 3.16.6 and am prepared to test with 7.0.2rc1

prjemian, 2018-12-19:
TD;DR - same problem using current pyepics@master

@newville - here are the results of that [test](https://github.com/APS-USAXS/ipython-usaxs/issues/100#issuecomment-448416845):

```
(/share1/bluesky/conda_envs/issue100) usaxs@usaxscontrol .../test/issue_100 $ ./compress100.sh 
Old : compress.RES                   0
New : compress.RES                   0
compress
    State:            connected
    Host:             usaxscontrol.xray.aps.anl.gov:5064
    Access:           read, write
    Native data type: DBF_DOUBLE
    Request type:     DBR_DOUBLE
    Element count:    5
Python version: 3.6.7 |Anaconda, Inc.| (default, Oct 23 2018, 19:16:44) 
[GCC 7.3.0]
PyEpics version: 3.3.2rc1+8.g498d803
libca version: /share1/bluesky/conda_envs/issue100/lib/python3.6/site-packages/epics/clibs/linux64/libca.so
compress.N 1.0
compress.NSAM 5.0
compress.NUSE 0.0
compress.VAL []
Python version: 3.6.5 |Anaconda, Inc.| (default, Apr 29 2018, 16:14:56) 
[GCC 7.2.0]
PyEpics version: 3.3.1
libca version: /APSshare/anaconda3/BlueSky/lib/python3.6/site-packages/epics/clibs/linux64/libca.so
compress.N 1.0
compress.NSAM 5.0
compress.NUSE 0.0
compress.VAL []
###############################
Old : signal                         0
New : signal                         0.1
compress 5 0.1 0 0 0 0
Python version: 3.6.7 |Anaconda, Inc.| (default, Oct 23 2018, 19:16:44) 
[GCC 7.3.0]
PyEpics version: 3.3.2rc1+8.g498d803
libca version: /share1/bluesky/conda_envs/issue100/lib/python3.6/site-packages/epics/clibs/linux64/libca.so
compress.N 1.0
compress.NSAM 5.0
compress.NUSE 1.0
compress.VAL 0.1
Python version: 3.6.5 |Anaconda, Inc.| (default, Apr 29 2018, 16:14:56) 
[GCC 7.2.0]
PyEpics version: 3.3.1
libca version: /APSshare/anaconda3/BlueSky/lib/python3.6/site-packages/epics/clibs/linux64/libca.so
compress.N 1.0
compress.NSAM 5.0
compress.NUSE 1.0
compress.VAL 0.1
###############################
Old : signal                         0.1
New : signal                         0.2
compress 5 0.1 0.2 0 0 0
Python version: 3.6.7 |Anaconda, Inc.| (default, Oct 23 2018, 19:16:44) 
[GCC 7.3.0]
PyEpics version: 3.3.2rc1+8.g498d803
libca version: /share1/bluesky/conda_envs/issue100/lib/python3.6/site-packages/epics/clibs/linux64/libca.so
compress.N 1.0
compress.NSAM 5.0
compress.NUSE 2.0
compress.VAL [0.1 0.2]
Python version: 3.6.5 |Anaconda, Inc.| (default, Apr 29 2018, 16:14:56) 
[GCC 7.2.0]
PyEpics version: 3.3.1
libca version: /APSshare/anaconda3/BlueSky/lib/python3.6/site-packages/epics/clibs/linux64/libca.so
compress.N 1.0
compress.NSAM 5.0
compress.NUSE 2.0
compress.VAL [ 0.1  0.2]
###############################
Old : signal                         0.2
New : signal                         0.3
compress 5 0.1 0.2 0.3 0 0
Python version: 3.6.7 |Anaconda, Inc.| (default, Oct 23 2018, 19:16:44) 
[GCC 7.3.0]
PyEpics version: 3.3.2rc1+8.g498d803
libca version: /share1/bluesky/conda_envs/issue100/lib/python3.6/site-packages/epics/clibs/linux64/libca.so
compress.N 1.0
compress.NSAM 5.0
compress.NUSE 3.0
compress.VAL [0.1 0.2 0.3]
Python version: 3.6.5 |Anaconda, Inc.| (default, Apr 29 2018, 16:14:56) 
[GCC 7.2.0]
PyEpics version: 3.3.1
libca version: /APSshare/anaconda3/BlueSky/lib/python3.6/site-packages/epics/clibs/linux64/libca.so
compress.N 1.0
compress.NSAM 5.0
compress.NUSE 3.0
compress.VAL [ 0.1  0.2  0.3]
###############################
Old : signal                         0.3
New : signal                         0.4
compress 5 0.1 0.2 0.3 0.4 0
Python version: 3.6.7 |Anaconda, Inc.| (default, Oct 23 2018, 19:16:44) 
[GCC 7.3.0]
PyEpics version: 3.3.2rc1+8.g498d803
libca version: /share1/bluesky/conda_envs/issue100/lib/python3.6/site-packages/epics/clibs/linux64/libca.so
compress.N 1.0
compress.NSAM 5.0
compress.NUSE 4.0
compress.VAL [0.1 0.2 0.3 0.4]
Python version: 3.6.5 |Anaconda, Inc.| (default, Apr 29 2018, 16:14:56) 
[GCC 7.2.0]
PyEpics version: 3.3.1
libca version: /APSshare/anaconda3/BlueSky/lib/python3.6/site-packages/epics/clibs/linux64/libca.so
compress.N 1.0
compress.NSAM 5.0
compress.NUSE 4.0
compress.VAL [ 0.1  0.2  0.3  0.4]
###############################
Old : signal                         0.4
New : signal                         0.5
compress 5 0.1 0.2 0.3 0.4 0.5
Python version: 3.6.7 |Anaconda, Inc.| (default, Oct 23 2018, 19:16:44) 
[GCC 7.3.0]
PyEpics version: 3.3.2rc1+8.g498d803
libca version: /share1/bluesky/conda_envs/issue100/lib/python3.6/site-packages/epics/clibs/linux64/libca.so
compress.N 1.0
compress.NSAM 5.0
compress.NUSE 5.0
compress.VAL [0.1 0.2 0.3 0.4 0.5]
Python version: 3.6.5 |Anaconda, Inc.| (default, Apr 29 2018, 16:14:56) 
[GCC 7.2.0]
PyEpics version: 3.3.1
libca version: /APSshare/anaconda3/BlueSky/lib/python3.6/site-packages/epics/clibs/linux64/libca.so
compress.N 1.0
compress.NSAM 5.0
compress.NUSE 5.0
compress.VAL [ 0.1  0.2  0.3  0.4  0.5]
###############################
Old : signal                         0.5
New : signal                         0.6
compress 5 0.2 0.3 0.4 0.5 0.6
Python version: 3.6.7 |Anaconda, Inc.| (default, Oct 23 2018, 19:16:44) 
[GCC 7.3.0]
PyEpics version: 3.3.2rc1+8.g498d803
libca version: /share1/bluesky/conda_envs/issue100/lib/python3.6/site-packages/epics/clibs/linux64/libca.so
compress.N 1.0
compress.NSAM 5.0
compress.NUSE 5.0
compress.VAL [0.2 0.3 0.4 0.5 0.6]
Python version: 3.6.5 |Anaconda, Inc.| (default, Apr 29 2018, 16:14:56) 
[GCC 7.2.0]
PyEpics version: 3.3.1
libca version: /APSshare/anaconda3/BlueSky/lib/python3.6/site-packages/epics/clibs/linux64/libca.so
compress.N 1.0
compress.NSAM 5.0
compress.NUSE 5.0
compress.VAL [ 0.2  0.3  0.4  0.5  0.6]
```

newville, 2018-12-19:
@prjemian  @kmpeters  OK, Yes, this is a pyepics problem that I can believe we broke recently and/or are not testing for well enough.  We'll fix (or, try to) before releasing 3.3.2 which should be soon: maybe not tomorrow, but well before the start of the next run.

newville, 2018-12-19:
@prjemian @kmpeters  This might be solved with https://github.com/pyepics/pyepics/pull/139.  At least, it works for me with a simple example use of a compress record....