APS-USAXS/usaxs-bluesky-ended-2023 #107: verify that existing fly scan data capture generates useful file

prjemian, 2019-01-24:
we're in a shutdown, the data will be garbage, it's the structure we need right, espcially with respect ot the [gain changes, *et al.](#100).

prjemian, 2019-01-24:
Moved this file to: `/share1/USAXS_data/test/2019-01-24_test_100.hdf5` so it can be test with IgorPro

prjemian, 2019-01-24:
@jilavsky Will this test file load into IgorPro?  `<USAXS_data>/test/2019-01-24_test_100.hdf5`

jilavsky, 2019-01-25:
I tried and for some reason the mca channels which should contain time, I) and upd counts all have 0 points value.

That is surprising. Time channel should be always usable and the others should all have some noise in them.

In other words, did you MCA run and advance as expected???

Sincerely

Jan
*****************************************************************
Dr. Jan Ilavsky, physicist
X-Ray Science Division
Advanced Photon Source
Argonne National Laboratory
9700 S. Cass Avenue, bldg 433A002
Argonne, IL 60439, USA
Phone:  (630) 252-0866
e-mail:  ilavsky@aps.anl.gov<mailto:ilavsky@aps.anl.gov>
Skype:  Honzail
Facetime:ilavsky@aps.anl.gov<mailto:ilavsky@aps.anl.gov>

web page:
https://usaxs.xray.aps.anl.gov/staff/jan-ilavsky
Indra web page: https://usaxs.xray.aps.anl.gov/software/indra
Irena web page: https://usaxs.xray.aps.anl.gov/software/irena
Nika web page: https://usaxs.xray.aps.anl.gov/software/nika
USAXS web page: https://usaxs.xray.aps.anl.gov
APS small-angle page: https://small-angle.aps.anl.gov/
Youtube channel: https://www.youtube.com/channel/UCDTzjGr3mAbRi3O4DJG7xHA/about
Shipping instructions: https://usaxs.xray.aps.anl.gov/documentation/shipping-instructions
*****************************************************************
Suggestion No1 to users:
If the model is incompatible with your data, it is not the data you need to change.
The fact, that A model fits your data, it is NOT proof that it is THE appropriate model.

prjemian, 2019-01-25:
Did not yet run a scan, just created the file with the output routine.
Tomorrow, the fly scan...

prjemian, 2019-01-30:
File loaded into IgorPro, data were missing any changes in photodiode amplifier gain.  Was amplifier in "automatic" mode?

commit d12753db (2019-01-30): fixes #107
closed 2019-01-30 by prjemian