APS-USAXS/usaxs-bluesky-ended-2023 #122: retune_needed needs to be set by calculations

prjemian, 2019-01-30:
There are a few places where `retune_needed` should be calculated, based on some EPICS PVs.  @jilavsky will help identify where and the logic.

prjemian, 2019-01-30:
```terms.preUSAXStune.summary()
data keys (* hints)
-------------------
 terms_preUSAXStune_epoch_last_tune
 terms_preUSAXStune_num_scans_last_tune
 terms_preUSAXStune_req_num_scans_between_tune
 terms_preUSAXStune_req_time_between_tune
 terms_preUSAXStune_run_tune_next
 terms_preUSAXStune_run_tune_on_qdo

read attrs
----------
num_scans_last_tune  EpicsSignal         ('terms_preUSAXStune_num_scans_last_tune')
epoch_last_tune      EpicsSignal         ('terms_preUSAXStune_epoch_last_tune')
req_num_scans_between_tune EpicsSignal         ('terms_preUSAXStune_req_num_scans_between_tune')
req_time_between_tune EpicsSignal         ('terms_preUSAXStune_req_time_between_tune')
run_tune_on_qdo      EpicsSignal         ('terms_preUSAXStune_run_tune_on_qdo')
run_tune_next        EpicsSignal         ('terms_preUSAXStune_run_tune_next')

config keys
-----------

configuration attrs
----------

Unused attrs
------------

```

prjemian, 2019-01-30:
*pretune_needed* 

```
= 1 when run_tune_next
= 1 when num_scans_last_tune  >req_num_scans_between_tune
= 1 when epoch now > epoch_last_tune + req_time_between_tune
```

* and reset the various components during the subsequent *preUSAXStune*
* also adjust the appropriate components during the appropriate scans

jilavsky, 2019-02-05:
We tested this last week and it was working "quite reasonably."

closed 2019-02-05 by jilavsky