APS-USAXS/usaxs-bluesky-ended-2023 #129: Need how to set some epics PVs on start

jilavsky, 2019-02-04:
USAXS_conf.mac sets not only constants (now stored in constants.py, but also calculate some epics PVs. There are calculated based on energy etc. Since USAXS_conf is loaded with every data collection, these get updates as necessary. I see no way to do so at this time.

prjemian, 2019-04-18:
Is this complete yet?

prjemian, 2019-04-19:
Still work-in-progress:

tune_ar chooses a width based on the monochromator energy

similar for the other tunable axes

Likely we can implement this in the `AXIS_pretune_hook()` but we need a cross-reference of energies and tune widths - consult the SPEC implementation for guidance

prjemian, 2019-06-03:
This should also consider the new PV for each axis tune width. 

In SPEC, this is run when user asks for a job (`qdo USAXS.conf`).  Any user conf will override the USAXS.conf values.  At this time, EPICS will be updated with the final values.

  - instrument defaults are energy-dependent
  - any user settings, if provided, override instrument defaults
  - this goes to EPICS

Initial tuning starts with these values.  The user can update values during measurement from EPICS and that will pass through to BS.  The next `qdo` will reset EPICS with the above procedure.

prjemian, 2019-06-04:
relates to #211, #230, #232

prjemian, 2019-06-04:
Also setting instrument defaults for DX & DY widths in `instrument_default_tune_ranges()` (in file `29-axis_tuning.py`)

PR #238 (2019-06-05): (re)set default tune ranges and other settings
fixes #129
closed 2019-06-05 by prjemian