APS-USAXS/usaxs-bluesky-ended-2023 #132: WAXS terms missing

jilavsky, 2019-02-04:
WAXS and SAXS use the same PVs for some information, but should we add these same PVs into terms.WAXS to avoid using terms.SAXS inside WAXS code?  Even though these refer to the same PVs? 

```
        terms.SAXS.start_exposure_time, ts,
        terms.SAXS.I0, scaler1.channels.chan02.s.value, 
        terms.SAXS.diode_transmission, scaler0.channels.chan04.s.value,
        terms.SAXS.diode_gain, trd_controls.femto.gain.value,
        terms.SAXS.I0_transmission, scaler0.channels.chan02.s.value,
        terms.SAXS.I0_gain, I0_controls.femto.gain.value,
```

In [1]: terms.WAXS                                                                                                                                                      
Out[1]: Parameters_WAXS(prefix='', name='terms_WAXS', parent='terms', read_attrs=['x_in', 'x_out', 'x_limit_offset', 'filters', 'filters.Al', 'filters.Ti', 'base_dir', 'num_images', 'acquire_time'], configuration_attrs=['filters'])

In [2]: terms.SAXS                                                                                                                                                      
Out[2]: Parameters_SAXS(prefix='', name='terms_SAXS', parent='terms', read_attrs=['z_in', 'z_out', 'z_limit_offset', 'y_in', 'y_out', 'y_limit_offset', 'ax_in', 'ax_out', 'ax_limit_offset', 'dx_in', 'dx_out', 'dx_limit_offset', 'usaxs_h_size', 'usaxs_v_size', 'v_size', 'h_size', 'usaxs_guard_h_size', 'usaxs_guard_v_size', 'guard_v_size', 'guard_h_size', 'filters', 'filters.Al', 'filters.Ti', 'diode_transmission', 'I0_transmission', 'diode_gain', 'I0_gain', 'base_dir', 'UsaxsSaxsMode', 'num_images', 'acquire_time', 'start_exposure_time', 'end_exposure_time', 'I0'], configuration_attrs=['filters'])

prjemian, 2019-06-05:
@jilavsky : What do you think?

Since there is one set of PVs that is used for both SAXS & WAXS detectors, seems weird to place this specifically under `terms.SAXS` yet also use it for WAXS.  Advice from bluesky developers is not to define the same PV in more than one `EpicsSignal`.

I suggest that you pick from one of these possibilities:

* [ ] move them to new subgrouping `terms.SWAXS`
* [ ] move them to new subgrouping `terms.SAXS_WAXS`
* [ ] move them to main group `terms`
* [ ] leave them in the same place and use `terms.SAXS` for both SAXS and WAXS
* [ ] your suggestion: _____________________

prjemian, 2019-06-06:
@jilavsky : pick one of [these](https://github.com/APS-USAXS/ipython-usaxs/issues/132#issuecomment-499087918)

jilavsky, 2019-06-10:
terms.SAXS_WAXS will be easiest to split later if we decide we need second set of PVs. It has not been necessary yet. We already have a lot of PVs.

prjemian, 2019-06-10:
not really a bug, just something to do, possibly an enhancement

PR #245 (2019-06-10): gather terms used by both SAXS & WAXS
fixes #132
closed 2019-06-10 by prjemian