APS-USAXS/usaxs-bluesky-ended-2023 #140: design how to describe loops in BS scan input file

jilavsky, 2019-02-05:
Sometimes, users need to repeat scans as sample changes.

jilavsky, 2019-02-05:
probably best to teach them how to write a plan, provide some good starting suggestions

jilavsky, 2019-02-05:
SPEC example:

```
def MeasureAllThree '{
   if( $# != 4) {
    	  printf ("usage: MeasureAllThree sx sy thickness \"Sample name\"\n")
        exit;
   }
   comment "USAXS SAXS WAXS scan"
   USAXSscan        $1     $2   $3    \'$4\'
   saxsExp          $1     $2   $3    \'$4\' 
   waxsExp          $1     $2   $3    \'$4\' 
}'
```

then later

```
	 __UserStartExpTime = time()
	 MeasureAllThree     __LinkSX     __LinkSY    __LinkTH   __LinkSampleName

	 ### At 400C sp= 401.65
	 set_LinkamTemp 402			### temperature in C
	 set_LinkamRate 100  		### rate in deg C/min
     WaitLinkamTemp
	 for(__MattIVal=1 ; __MattIVal<9 ; __MattIVal++){
	   #print __MattIVal
	   MeasureAllThree     __LinkSX     __LinkSY    __LinkTH   __LinkSampleName
     }
```

prjemian, 2019-05-30:
Use python.  Here is an example:

```
def MeasureAllThree(sx, sy, thickness, sample_name, md={}):
    print("USAXS SAXS WAXS scan")
    yield from FlyScan(sx, sy, thickness, sample_name, md=md)
    yield from SAXS(sx, sy, thickness, sample_name, md=md)
    yield from WAXS(sx, sy, thickness, sample_name, md=md)

def myPlan(sx, sy, thickness, sample_name, iterations=9, md={}):
    t0 = time.time()
    md = {
        "user_procedure": "USAXS SAXS WAXS scans",
        "iteration": 0,
        "total_iteration": iterations,
        }
    yield from MeasureAllThree(sx, sy, thickness, sample_name, md=md)
    yield from bps.mv(linkam.set_rate, 100)			# degrees C/minute
    yield from linkam.set_target(402, wait=True)	# degrees C
    for i in range(iterations):
        print(f"Iteration {i+1} of {iterations}, elapsed time = {time.time() - elapsed:.3f}s")
        md["iteration"] = i+1
        yield from MeasureAllThree(sx, sy, thickness, sample_name, md=md)
```

closed 2019-05-30 by prjemian

prjemian, 2019-06-05:
This example was added (with corre3ction) into #239