APS-USAXS/usaxs-bluesky-ended-2023 #141: show how users can write their own local python code to use in BS session

jilavsky, 2019-02-05:


prjemian, 2019-05-30:
Users can always write a python source code file in their present working directory and import from it.  Say they have a file `heater.py`, then `import heater` and `RE(heater.my_overnight())` might be one example.

prjemian, 2019-05-30:
The real issue to overcome is how to *re-*import a file that has previously been imported, then edited.  Takes a little more work to do that.

prjemian, 2019-05-30:
That **has** to be a common problem so web search turns up:

* https://stackoverflow.com/questions/1254370/reimport-a-module-in-python-while-interactive

With the example `heater.py` file, we'll need to use the `importlib` package:

```
import importlib
importlib.reload(heater)
```

or simpler:

```
from importlib import reload
reload(heater)
```

commit 7e538259 (2019-05-30): fixes #141
closed 2019-05-30 by prjemian