APS-USAXS/usaxs-bluesky-ended-2023 #142: support Linkam soft IOC

jilavsky, 2019-02-05:
These soft IOCs support Linkam temperature controllers

* CI94 (the old one)
* T96 (the new one)

prjemian, 2019-05-21:
notes:

```
jemian@wow .../oxygen18/JEMIAN $ /APSshare/bin/alivedb | grep -i linkam | grep -i 9idc
9idcLINKAM (164.54.53.132) 0 - Down time: > 35 days, 12 hr
  TOP = D:/epics/linkamDevel/ioc/9idcLINKAM
ioc9idcLINKAM (164.54.109.130) 0 - Down time: 23 hr, 36 min
  TOP = C:/epics/ioc/9idcLINKAM
  TOP = D:/epics/linkamDevel/ioc/9idcLINKAM
```

conclusion: this is a Windows IOC

prjemian, 2019-05-21:
PVs:

name | id | description | value | units | timestamp
-- | -- | -- | -- | -- | --
9idcLINKAM:tc1:vacuumLimit_RBV | linkamT96_Pressure | Linkam Pressure, mBar | None |   | 2019-05-20 14:43:10.119654
9idcLINKAM:tc1:vacuumStatus_RBV | linkamT96_VaccumStatus | Linkam Vacuum Status | None |   | 2019-05-20 14:43:16.120195
9idcLINKAM:tc1:statusError_RBV | linkamT96_errors | Linkam errors | None |   | 2019-05-20 14:42:58.118710
9idcLINKAM:tc1:rampLimit_RBV | linkamT96_limit | Linkam Limit, C | None |   | 2019-05-20 14:42:52.118280
9idcLINKAM:tc1:lnpStatus_RBV | linkamT96_ln2Status | Linkam LN2 Status, C/min | None |   | 2019-05-20 14:43:04.119146
9idcLINKAM:tc1:rampRate_RBV | linkamT96_rate | Linkam Rate, C/min | None |   | 2019-05-20 14:42:46.117842
9idcLINKAM:tc1:heating | linkamT96_status | Linkam Heating | None |   | 2019-05-20 14:42:40.117397
9idcLINKAM:tc1:temperature_RBV | linkamT96_temp | Linkam T96 Temp., C | None |   | 2019-05-20 14:42:34.117017

prjemian, 2019-05-21:
same for the *old* one:

name | id | description | value | units | timestamp
-- | -- | -- | -- | -- | --
9idcLAX:ci94:errorByte | linkam_errors | Linkam errors | 65408 |   | 2019-05-20 14:42:28.116689
9idcLAX:ci94:setLimit | linkam_limit | Linkam Limit, C | 35.0 |   | 2019-05-20 14:42:28.092192
9idcLAX:ci94:setRate | linkam_rate | Linkam Rate, C/min | 150.0 |   | 2019-05-20 14:42:28.062935
9idcLAX:ci94:status | linkam_status | Linkam Status | 3 |   | 2019-05-20 14:42:28.036129
9idcLAX:ci94:temp | linkam_temp1 | Linkam Temp. 1, C | 35.0 |   | 2019-05-20 14:42:27.982602
9idcLAX:ci94:temp2 | linkam_temp2 | Linkam Temp. 2, C | 35.0 |   | 2019-05-20 14:42:28.008787

prjemian, 2019-05-21:
But there is more to the EPICS database.  See `/net/s9dserv/xorApps/epics/synApps_5_7/ioc/9idcLAX/9idcLAXApp/Db/ci94.db`

prjemian, 2019-05-21:
That database looks like:

RTYP | PV
-- | --
ao | $(P)ci94:setRate
ao | $(P)ci94:setLimit
longout | $(P)ci94:setSpeed
bo | $(P)ci94:clearBuffer
bo | $(P)ci94:endAfterProfile
bo | $(P)ci94:endOnStop
bo | $(P)ci94:start
bo | $(P)ci94:stop
bo | $(P)ci94:hold
bo | $(P)ci94:pumpMode
mbbi | $(P)ci94:status
mbbi | $(P)ci94:errorByte
mbbi | $(P)ci94:genStat
asyn | $(P)ci94:asyn
longin | $(P)ci94:pumpSpeedIn
calc | $(P)ci94:pumpSpeed
bo | $(P)ci94:scanDis
ai | $(P)ci94:tempIn
calc | $(P)ci94:temp
ai | $(P)ci94:dscIn
calc | $(P)ci94:dsc
ai | $(P)ci94:temp2In
calc | $(P)ci94:temp2
longin | $(P)ci94:statusIn
longout | $(P)ci94:test
ai | $(P)ci94:DCmd
ai | $(P)ci94:TCmd

prjemian, 2019-05-21:
@kmpeters : Can you post the contents of the EPICS database for the Linkam T96 support?

prjemian, 2019-05-21:
SPEC macros in directory: `/spec/macros/local`

* **`usaxs_devicesupport.mac`**
    macros for CI94 and T96 controllers
* **`User_created_Commands.mac`**
    for CI94: `RecordLinkamTemp`, `set_LinkamTemp`, `WaitLinkamTemp`

prjemian, 2019-05-21:
@kmpeters: never mind, found it and copied to `/share1/tmp` for myself.

commit 6db03b9c (2019-05-24): fixes #142
PR #208 (2019-05-24): support for Linkam temperature controllers
fixes #142
commit 366cab31 (2019-05-24): Merge pull request #208 from APS-USAXS/Linkam-142

support for Linkam temperature controllers, fixes #142
closed 2019-05-24 by prjemian