APS-USAXS/usaxs-bluesky-ended-2023 #143: update live data software to understand BS

jilavsky, 2019-02-05:
Update live data software to manage to work with BS generated data.

prjemian, 2019-04-19:
Also, the specplots do not update properly and this is related.  The code that recognizes "plottable" data needs to handle the way that Bluesky writes a SPEC data file *and* those files need to have sufficient information to identify the plottable data.

prjemian, 2019-05-02:
These issues are related: #187, #193, #194, #195

prjemian, 2019-05-02:
With `spec2nexus.__version__ 2019.0501.0'`, *specplot.py* recognizes and correctly plots the data from `tune_ar()` using today's edits for [apstools](https://github.com/BCDA-APS/apstools): #100 and #126.

prjemian, 2019-05-02:
#198 was also related

prjemian, 2019-05-03:
https://github.com/prjemian/spec2nexus/issues/142 is related

prjemian, 2019-05-06:
looks finished here

closed 2019-05-06 by prjemian