APS-USAXS/usaxs-bluesky-ended-2023 #145: Correct PVs renaming

jilavsky, 2019-02-15:
USAXS plan for March includes renaming some PVs from ugly "USAXS_Pin" to more reasonable "SAXS" and "WAXS". This will need to be propagated to spec and qtdm, and also to BS. List will be provided when available.

prjemian, 2019-03-15:
@jilavsky (@kmpeters)

For starters on this, here is the list of PVs *in LAX* with `pin` in their name:

```
usaxs@usaxscontrol .../iocBoot/ioc9idcLAX $ grep -i pin dbl-all.txt | sort
9idcLAX:ci94:tempIn
9idcLAX:USAXS_Pin:AcquireTime
9idcLAX:USAXS_Pin:ax_in
9idcLAX:USAXS_Pin:ax_limit_offset
9idcLAX:USAXS_Pin:ax_out
9idcLAX:USAXS_Pin:BeamCenterX
9idcLAX:USAXS_Pin:BeamCenterX:NXdetector
9idcLAX:USAXS_Pin:BeamCenterY
9idcLAX:USAXS_Pin:BeamCenterY:NXdetector
9idcLAX:USAXS_Pin:CalibrationConstant
9idcLAX:USAXS_Pin:DetectorTiltX
9idcLAX:USAXS_Pin:DetectorTiltY
9idcLAX:USAXS_Pin:directory
9idcLAX:USAXS_Pin:Distance
9idcLAX:USAXS_Pin:dx_in
9idcLAX:USAXS_Pin:dx_limit_offset
9idcLAX:USAXS_Pin:dx_out
9idcLAX:USAXS_Pin:EmptyFileName
9idcLAX:USAXS_Pin:EndExposureTime
9idcLAX:USAXS_Pin:Exp_Al_Filter
9idcLAX:USAXS_Pin:Exp_Ti_Filter
9idcLAX:USAXS_Pin:I0
9idcLAX:USAXS_Pin:NumImages
9idcLAX:USAXS_Pin:Pin_hgslit_ap
9idcLAX:USAXS_Pin:Pin_hslit_ap
9idcLAX:USAXS_Pin:PinPixSizeX
9idcLAX:USAXS_Pin:PinPixSizeY
9idcLAX:USAXS_Pin:Pin_TrI0
9idcLAX:USAXS_Pin:Pin_TrI0gain
9idcLAX:USAXS_Pin:Pin_TrPD
9idcLAX:USAXS_Pin:Pin_TrPDgain
9idcLAX:USAXS_Pin:Pin_vgslit_ap
9idcLAX:USAXS_Pin:Pin_vslit_ap
9idcLAX:USAXS_Pin:Pin_y_in
9idcLAX:USAXS_Pin:Pin_y_limit_offset
9idcLAX:USAXS_Pin:Pin_y_out
9idcLAX:USAXS_Pin:Pin_z_in
9idcLAX:USAXS_Pin:Pin_z_limit_offset
9idcLAX:USAXS_Pin:Pin_z_out
9idcLAX:USAXS_Pin:sampleTitle
9idcLAX:USAXS_Pin:StartExposureTime
9idcLAX:USAXS_Pin:state
9idcLAX:USAXS_Pin:timeStamp
9idcLAX:USAXS_Pin:USAXS_hgslit_ap
9idcLAX:USAXS_Pin:USAXS_hslit_ap
9idcLAX:USAXS_Pin:USAXSSAXSMode
9idcLAX:USAXS_Pin:USAXS_vgslit_ap
9idcLAX:USAXS_Pin:USAXS_vslit_ap
9idcLAX:USAXS_Pin:userDir
9idcLAX:USAXS_Pin:userName
9idcLAX:USAXS_Pin:waxs_x_in
9idcLAX:USAXS_Pin:waxs_x_limit_offset
9idcLAX:USAXS_Pin:waxs_x_out
9idcLAX:USAXS:TR_MeasurePinTrans
9idcLAX:USAXS:TR_pinCounts
9idcLAX:USAXS:TR_pinGain
9idcLAX:USAXS_WAXS:PinPixSizeX
9idcLAX:USAXS_WAXS:PinPixSizeY
```

Need strong suggestions for renaming here

jilavsky, 2019-03-15:
This was discussed in e-mail on February 26th. 
Important parts are here:

String replacements :

USAXS_Pin -> SAXS
Pin_... -> SAXS_...
USAXS_WAXS -> WAXS

******
Following PVs need to be added (I can do it if needed, but al least check that syntax is OK):

grecord(bo,        "9idcLAX:USAXS:UseSpecificTuneLocation")
{
    field(DESC, "Use specific preUSAXStune location?")
    field(ZNAM, "no")
    field(ONAM, "yes")
}

grecord(ao,        "9idcLAX:USAXS:preUSAXStuneSX")
{
    field(DESC, "Specific preUSAXStune sx value")
    field(EGU,  "mm")
}
grecord(ao,        "9idcLAX:USAXS:preUSAXStuneSY")
{
    field(DESC, "Specific preUSAXStune sy value")
    field(EGU,  "mm")
}
*****
Pete requested change related to BlueSky, is this going to break anything elsewhere? Can you do it, not sure about syntax here...

9idcLAX:USAXS:userDir 		should be waveform

prjemian, 2019-03-15:
The change in `userDir` is issue #151 

------------------

@jkirchman should be aware of these changes, as well

prjemian, 2019-03-18:
Re [previous comment](https://github.com/APS-USAXS/ipython-usaxs/issues/145#issuecomment-473384447), these are the current PVs that match:

<details>
<summary>USAXS_Pin</summary>
<p>

```
jemian@otz .../9idcLAXApp/Db $ grep USAXS_Pin ../../iocBoot/ioc9idcLAX/dbl-all.txt 
9idcLAX:USAXS_Pin:AcquireTime
9idcLAX:USAXS_Pin:BeamCenterX
9idcLAX:USAXS_Pin:BeamCenterY
9idcLAX:USAXS_Pin:CalibrationConstant
9idcLAX:USAXS_Pin:DetectorTiltX
9idcLAX:USAXS_Pin:DetectorTiltY
9idcLAX:USAXS_Pin:Distance
9idcLAX:USAXS_Pin:Exp_Al_Filter
9idcLAX:USAXS_Pin:Exp_Ti_Filter
9idcLAX:USAXS_Pin:I0
9idcLAX:USAXS_Pin:NumImages
9idcLAX:USAXS_Pin:PinPixSizeX
9idcLAX:USAXS_Pin:PinPixSizeY
9idcLAX:USAXS_Pin:Pin_TrI0
9idcLAX:USAXS_Pin:Pin_TrI0gain
9idcLAX:USAXS_Pin:Pin_TrPD
9idcLAX:USAXS_Pin:Pin_TrPDgain
9idcLAX:USAXS_Pin:Pin_hgslit_ap
9idcLAX:USAXS_Pin:Pin_hslit_ap
9idcLAX:USAXS_Pin:Pin_vgslit_ap
9idcLAX:USAXS_Pin:Pin_vslit_ap
9idcLAX:USAXS_Pin:Pin_y_in
9idcLAX:USAXS_Pin:Pin_y_limit_offset
9idcLAX:USAXS_Pin:Pin_y_out
9idcLAX:USAXS_Pin:Pin_z_in
9idcLAX:USAXS_Pin:Pin_z_limit_offset
9idcLAX:USAXS_Pin:Pin_z_out
9idcLAX:USAXS_Pin:USAXS_hgslit_ap
9idcLAX:USAXS_Pin:USAXS_hslit_ap
9idcLAX:USAXS_Pin:USAXS_vgslit_ap
9idcLAX:USAXS_Pin:USAXS_vslit_ap
9idcLAX:USAXS_Pin:ax_in
9idcLAX:USAXS_Pin:ax_limit_offset
9idcLAX:USAXS_Pin:ax_out
9idcLAX:USAXS_Pin:dx_in
9idcLAX:USAXS_Pin:dx_limit_offset
9idcLAX:USAXS_Pin:dx_out
9idcLAX:USAXS_Pin:waxs_x_in
9idcLAX:USAXS_Pin:waxs_x_limit_offset
9idcLAX:USAXS_Pin:waxs_x_out
9idcLAX:USAXS_Pin:BeamCenterX:NXdetector
9idcLAX:USAXS_Pin:BeamCenterY:NXdetector
9idcLAX:USAXS_Pin:USAXSSAXSMode
9idcLAX:USAXS_Pin:EmptyFileName
9idcLAX:USAXS_Pin:EndExposureTime
9idcLAX:USAXS_Pin:StartExposureTime
9idcLAX:USAXS_Pin:sampleTitle
9idcLAX:USAXS_Pin:state
9idcLAX:USAXS_Pin:timeStamp
9idcLAX:USAXS_Pin:userName
9idcLAX:USAXS_Pin:directory
9idcLAX:USAXS_Pin:userDir
```

</p>
</details>

<details>
<summary>Pin_</summary>
<p>

```
jemian@otz .../9idcLAXApp/Db $ grep Pin_ ../../iocBoot/ioc9idcLAX/dbl-all.txt 
9idcLAX:USAXS_Pin:Pin_TrI0
9idcLAX:USAXS_Pin:Pin_TrI0gain
9idcLAX:USAXS_Pin:Pin_TrPD
9idcLAX:USAXS_Pin:Pin_TrPDgain
9idcLAX:USAXS_Pin:Pin_hgslit_ap
9idcLAX:USAXS_Pin:Pin_hslit_ap
9idcLAX:USAXS_Pin:Pin_vgslit_ap
9idcLAX:USAXS_Pin:Pin_vslit_ap
9idcLAX:USAXS_Pin:Pin_y_in
9idcLAX:USAXS_Pin:Pin_y_limit_offset
9idcLAX:USAXS_Pin:Pin_y_out
9idcLAX:USAXS_Pin:Pin_z_in
9idcLAX:USAXS_Pin:Pin_z_limit_offset
9idcLAX:USAXS_Pin:Pin_z_out
```

</p>
</details>

<details>
<summary>USAXS_WAXS</summary>
<p>

```
jemian@otz .../9idcLAXApp/Db $ grep USAXS_WAXS ../../iocBoot/ioc9idcLAX/dbl-all.txt 
9idcLAX:USAXS_WAXS:AcquireTime
9idcLAX:USAXS_WAXS:BeamCenterX
9idcLAX:USAXS_WAXS:BeamCenterY
9idcLAX:USAXS_WAXS:CalibrationConstant
9idcLAX:USAXS_WAXS:DetectorTiltX
9idcLAX:USAXS_WAXS:DetectorTiltY
9idcLAX:USAXS_WAXS:Distance
9idcLAX:USAXS_WAXS:Exp_Al_Filter
9idcLAX:USAXS_WAXS:Exp_Ti_Filter
9idcLAX:USAXS_WAXS:I0
9idcLAX:USAXS_WAXS:NumImages
9idcLAX:USAXS_WAXS:PinPixSizeX
9idcLAX:USAXS_WAXS:PinPixSizeY
9idcLAX:USAXS_WAXS:BeamCenterX:NXdetector
9idcLAX:USAXS_WAXS:BeamCenterY:NXdetector
9idcLAX:USAXS_WAXS:EmptyFileName
9idcLAX:USAXS_WAXS:EndExposureTime
9idcLAX:USAXS_WAXS:StartExposureTime
9idcLAX:USAXS_WAXS:sampleTitle
9idcLAX:USAXS_WAXS:directory
```

</p>
</details>

prjemian, 2019-03-18:
@jilavsky : These are the proposed name changes.  Should we create aliases (attn: @kmpeters, that needs your help) back to the old names?  Adding alises might be best so close to upcoming operations period.

existing PV  |  new PV
---- | ----
9idcLAX:USAXS_Pin:AcquireTime  |  9idcLAX:SAXS:AcquireTime
9idcLAX:USAXS_Pin:BeamCenterX  |  9idcLAX:SAXS:BeamCenterX
9idcLAX:USAXS_Pin:BeamCenterY  |  9idcLAX:SAXS:BeamCenterY
9idcLAX:USAXS_Pin:CalibrationConstant  |  9idcLAX:SAXS:CalibrationConstant
9idcLAX:USAXS_Pin:DetectorTiltX  |  9idcLAX:SAXS:DetectorTiltX
9idcLAX:USAXS_Pin:DetectorTiltY  |  9idcLAX:SAXS:DetectorTiltY
9idcLAX:USAXS_Pin:Distance  |  9idcLAX:SAXS:Distance
9idcLAX:USAXS_Pin:Exp_Al_Filter  |  9idcLAX:SAXS:Exp_Al_Filter
9idcLAX:USAXS_Pin:Exp_Ti_Filter  |  9idcLAX:SAXS:Exp_Ti_Filter
9idcLAX:USAXS_Pin:I0  |  9idcLAX:SAXS:I0
9idcLAX:USAXS_Pin:NumImages  |  9idcLAX:SAXS:NumImages
9idcLAX:USAXS_Pin:PinPixSizeX  |  9idcLAX:SAXS:PinPixSizeX
9idcLAX:USAXS_Pin:PinPixSizeY  |  9idcLAX:SAXS:PinPixSizeY
9idcLAX:USAXS_Pin:Pin_TrI0  |  9idcLAX:SAXS:Pin_TrI0
9idcLAX:USAXS_Pin:Pin_TrI0gain  |  9idcLAX:SAXS:Pin_TrI0gain
9idcLAX:USAXS_Pin:Pin_TrPD  |  9idcLAX:SAXS:Pin_TrPD
9idcLAX:USAXS_Pin:Pin_TrPDgain  |  9idcLAX:SAXS:Pin_TrPDgain
9idcLAX:USAXS_Pin:Pin_hgslit_ap  |  9idcLAX:SAXS:Pin_hgslit_ap
9idcLAX:USAXS_Pin:Pin_hslit_ap  |  9idcLAX:SAXS:Pin_hslit_ap
9idcLAX:USAXS_Pin:Pin_vgslit_ap  |  9idcLAX:SAXS:Pin_vgslit_ap
9idcLAX:USAXS_Pin:Pin_vslit_ap  |  9idcLAX:SAXS:Pin_vslit_ap
9idcLAX:USAXS_Pin:Pin_y_in  |  9idcLAX:SAXS:Pin_y_in
9idcLAX:USAXS_Pin:Pin_y_limit_offset  |  9idcLAX:SAXS:Pin_y_limit_offset
9idcLAX:USAXS_Pin:Pin_y_out  |  9idcLAX:SAXS:Pin_y_out
9idcLAX:USAXS_Pin:Pin_z_in  |  9idcLAX:SAXS:Pin_z_in
9idcLAX:USAXS_Pin:Pin_z_limit_offset  |  9idcLAX:SAXS:Pin_z_limit_offset
9idcLAX:USAXS_Pin:Pin_z_out  |  9idcLAX:SAXS:Pin_z_out
9idcLAX:USAXS_Pin:USAXS_hgslit_ap  |  9idcLAX:SAXS:USAXS_hgslit_ap
9idcLAX:USAXS_Pin:USAXS_hslit_ap  |  9idcLAX:SAXS:USAXS_hslit_ap
9idcLAX:USAXS_Pin:USAXS_vgslit_ap  |  9idcLAX:SAXS:USAXS_vgslit_ap
9idcLAX:USAXS_Pin:USAXS_vslit_ap  |  9idcLAX:SAXS:USAXS_vslit_ap
9idcLAX:USAXS_Pin:ax_in  |  9idcLAX:SAXS:ax_in
9idcLAX:USAXS_Pin:ax_limit_offset  |  9idcLAX:SAXS:ax_limit_offset
9idcLAX:USAXS_Pin:ax_out  |  9idcLAX:SAXS:ax_out
9idcLAX:USAXS_Pin:dx_in  |  9idcLAX:SAXS:dx_in
9idcLAX:USAXS_Pin:dx_limit_offset  |  9idcLAX:SAXS:dx_limit_offset
9idcLAX:USAXS_Pin:dx_out  |  9idcLAX:SAXS:dx_out
9idcLAX:USAXS_Pin:waxs_x_in  |  9idcLAX:SAXS:waxs_x_in
9idcLAX:USAXS_Pin:waxs_x_limit_offset  |  9idcLAX:SAXS:waxs_x_limit_offset
9idcLAX:USAXS_Pin:waxs_x_out  |  9idcLAX:SAXS:waxs_x_out
9idcLAX:USAXS_Pin:BeamCenterX:NXdetector  |  9idcLAX:SAXS:BeamCenterX:NXdetector
9idcLAX:USAXS_Pin:BeamCenterY:NXdetector  |  9idcLAX:SAXS:BeamCenterY:NXdetector
9idcLAX:USAXS_Pin:USAXSSAXSMode  |  9idcLAX:SAXS:USAXSSAXSMode
9idcLAX:USAXS_Pin:EmptyFileName  |  9idcLAX:SAXS:EmptyFileName
9idcLAX:USAXS_Pin:EndExposureTime  |  9idcLAX:SAXS:EndExposureTime
9idcLAX:USAXS_Pin:StartExposureTime  |  9idcLAX:SAXS:StartExposureTime
9idcLAX:USAXS_Pin:sampleTitle  |  9idcLAX:SAXS:sampleTitle
9idcLAX:USAXS_Pin:state  |  9idcLAX:SAXS:state
9idcLAX:USAXS_Pin:timeStamp  |  9idcLAX:SAXS:timeStamp
9idcLAX:USAXS_Pin:userName  |  9idcLAX:SAXS:userName
9idcLAX:USAXS_Pin:directory  |  9idcLAX:SAXS:directory
9idcLAX:USAXS_Pin:userDir  |  9idcLAX:SAXS:userDir
9idcLAX:USAXS_Pin:Pin_TrI0  |  9idcLAX:SAXS:TrI0
9idcLAX:USAXS_Pin:Pin_TrI0gain  |  9idcLAX:SAXS:TrI0gain
9idcLAX:USAXS_Pin:Pin_TrPD  |  9idcLAX:SAXS:TrPD
9idcLAX:USAXS_Pin:Pin_TrPDgain  |  9idcLAX:SAXS:TrPDgain
9idcLAX:USAXS_Pin:Pin_hgslit_ap  |  9idcLAX:SAXS:hgslit_ap
9idcLAX:USAXS_Pin:Pin_hslit_ap  |  9idcLAX:SAXS:hslit_ap
9idcLAX:USAXS_Pin:Pin_vgslit_ap  |  9idcLAX:SAXS:vgslit_ap
9idcLAX:USAXS_Pin:Pin_vslit_ap  |  9idcLAX:SAXS:vslit_ap
9idcLAX:USAXS_Pin:Pin_y_in  |  9idcLAX:SAXS:y_in
9idcLAX:USAXS_Pin:Pin_y_limit_offset  |  9idcLAX:SAXS:y_limit_offset
9idcLAX:USAXS_Pin:Pin_y_out  |  9idcLAX:SAXS:y_out
9idcLAX:USAXS_Pin:Pin_z_in  |  9idcLAX:SAXS:z_in
9idcLAX:USAXS_Pin:Pin_z_limit_offset  |  9idcLAX:SAXS:z_limit_offset
9idcLAX:USAXS_Pin:Pin_z_out  |  9idcLAX:SAXS:z_out
9idcLAX:USAXS_WAXS:AcquireTime  |  9idcLAX:WAXS::AcquireTime
9idcLAX:USAXS_WAXS:BeamCenterX  |  9idcLAX:WAXS::BeamCenterX
9idcLAX:USAXS_WAXS:BeamCenterY  |  9idcLAX:WAXS::BeamCenterY
9idcLAX:USAXS_WAXS:CalibrationConstant  |  9idcLAX:WAXS::CalibrationConstant
9idcLAX:USAXS_WAXS:DetectorTiltX  |  9idcLAX:WAXS::DetectorTiltX
9idcLAX:USAXS_WAXS:DetectorTiltY  |  9idcLAX:WAXS::DetectorTiltY
9idcLAX:USAXS_WAXS:Distance  |  9idcLAX:WAXS::Distance
9idcLAX:USAXS_WAXS:Exp_Al_Filter  |  9idcLAX:WAXS::Exp_Al_Filter
9idcLAX:USAXS_WAXS:Exp_Ti_Filter  |  9idcLAX:WAXS::Exp_Ti_Filter
9idcLAX:USAXS_WAXS:I0  |  9idcLAX:WAXS::I0
9idcLAX:USAXS_WAXS:NumImages  |  9idcLAX:WAXS::NumImages
9idcLAX:USAXS_WAXS:PinPixSizeX  |  9idcLAX:WAXS::PinPixSizeX
9idcLAX:USAXS_WAXS:PinPixSizeY  |  9idcLAX:WAXS::PinPixSizeY
9idcLAX:USAXS_WAXS:BeamCenterX:NXdetector  |  9idcLAX:WAXS::BeamCenterX:NXdetector
9idcLAX:USAXS_WAXS:BeamCenterY:NXdetector  |  9idcLAX:WAXS::BeamCenterY:NXdetector
9idcLAX:USAXS_WAXS:EmptyFileName  |  9idcLAX:WAXS::EmptyFileName
9idcLAX:USAXS_WAXS:EndExposureTime  |  9idcLAX:WAXS::EndExposureTime
9idcLAX:USAXS_WAXS:StartExposureTime  |  9idcLAX:WAXS::StartExposureTime
9idcLAX:USAXS_WAXS:sampleTitle  |  9idcLAX:WAXS::sampleTitle
9idcLAX:USAXS_WAXS:directory  |  9idcLAX:WAXS::directory

prjemian, 2019-03-18:
Caught the `WAXS::` error

prjemian, 2019-03-18:
Best to create a script that can make these changes.  The script can be further adapted to change other client software such as SPEC, caQtDM, livedata pages, Bluesky, ...

prjemian, 2019-03-18:
spreadsheet with PV renames

[USAXS-PV-renames.xlsx](https://github.com/APS-USAXS/ipython-usaxs/files/2980701/USAXS-PV-renames.xlsx)

prjemian, 2019-03-19:
@jilavsky Look!  Look! Look!  Do you agree with these proposed PV name changes?  (I'll make aliases in a separate file back to the old names.)

```
usaxs@usaxscontrol .../9idcLAXApp/Db $ python ./rename_201903.py | tee changes.txt

reading:LAXterms.db

(LAXterms.db,45):
--- grecord(ao,        "9idcLAX:USAXS_Pin:BeamCenterX")
+++ grecord(ao,        "9idcLAX:SAXS:BeamCenterX")

(LAXterms.db,49):
---     field(FLNK, "9idcLAX:USAXS_Pin:BeamCenterX:NXdetector")
+++     field(FLNK, "9idcLAX:SAXS:BeamCenterX:NXdetector")

(LAXterms.db,51):
--- grecord(ao,        "9idcLAX:USAXS_Pin:PinPixSizeX")
+++ grecord(ao,        "9idcLAX:SAXS:PinPixSizeX")

(LAXterms.db,55):
---     field(FLNK, "9idcLAX:USAXS_Pin:BeamCenterX:NXdetector")
+++     field(FLNK, "9idcLAX:SAXS:BeamCenterX:NXdetector")

(LAXterms.db,57):
--- grecord(ao,        "9idcLAX:USAXS_Pin:DetectorTiltX")
+++ grecord(ao,        "9idcLAX:SAXS:DetectorTiltX")

(LAXterms.db,62):
--- grecord(ao,        "9idcLAX:USAXS_Pin:DetectorTiltY")
+++ grecord(ao,        "9idcLAX:SAXS:DetectorTiltY")

(LAXterms.db,67):
--- record(calcout, "9idcLAX:USAXS_Pin:BeamCenterX:NXdetector")
+++ record(calcout, "9idcLAX:SAXS:BeamCenterX:NXdetector")

(LAXterms.db,72):
---     field(INPA, "9idcLAX:USAXS_Pin:BeamCenterX")
+++     field(INPA, "9idcLAX:SAXS:BeamCenterX")

(LAXterms.db,73):
---     field(INPB, "9idcLAX:USAXS_Pin:PinPixSizeX")
+++     field(INPB, "9idcLAX:SAXS:PinPixSizeX")

(LAXterms.db,77):
--- grecord(ao, "9idcLAX:USAXS_Pin:BeamCenterY")
+++ grecord(ao, "9idcLAX:SAXS:BeamCenterY")

(LAXterms.db,81):
---     field(FLNK, "9idcLAX:USAXS_Pin:BeamCenterY:NXdetector")
+++     field(FLNK, "9idcLAX:SAXS:BeamCenterY:NXdetector")

(LAXterms.db,83):
--- grecord(ao,        "9idcLAX:USAXS_Pin:PinPixSizeY")
+++ grecord(ao,        "9idcLAX:SAXS:PinPixSizeY")

(LAXterms.db,87):
---     field(FLNK, "9idcLAX:USAXS_Pin:BeamCenterY:NXdetector")
+++     field(FLNK, "9idcLAX:SAXS:BeamCenterY:NXdetector")

(LAXterms.db,89):
--- record(calcout, "9idcLAX:USAXS_Pin:BeamCenterY:NXdetector")
+++ record(calcout, "9idcLAX:SAXS:BeamCenterY:NXdetector")

(LAXterms.db,94):
---     field(INPA, "9idcLAX:USAXS_Pin:BeamCenterY")
+++     field(INPA, "9idcLAX:SAXS:BeamCenterY")

(LAXterms.db,95):
---     field(INPB, "9idcLAX:USAXS_Pin:PinPixSizeY")
+++     field(INPB, "9idcLAX:SAXS:PinPixSizeY")
[... 590 lines cut ...]
LAXterms.db
LAXterms_settings.req
USAXSterms.db
USAXS_Pinterms.db
"""

for fname in files.strip().split():
    with open(fname, "r") as fin:
        print("")
        print("reading:{}".format(fname))
        fname_out = fname + ".new"
        with open(fname_out, "w") as fout:
            for line_count, text in enumerate(fin.readlines()):
                line_count += 1
                in_text = text
                for k, v in replacements.items():
                    text = text.replace(k, v)
                fout.write(text)
                if in_text != text:
                    msg = "\n({},{}):".format(fname, line_count)
                    msg += "\n{} {}".format("---", in_text.rstrip())
                    msg += "\n{} {}".format("+++", text.rstrip())
                    print(msg)
    print("wrote:{}".format(fname_out))
```

prjemian, 2019-03-19:
locate here:

```
usaxs@usaxscontrol .../9idcLAXApp/Db $ pwd
/net/s9dserv/xorApps/epics/synApps_5_7/ioc/9idcLAX/9idcLAXApp/Db
```

jilavsky, 2019-03-19:
This should be changed: 
(USAXS_Pinterms.db,8):
--- grecord(ao,        "9idcLAX:USAXS_Pin:waxs_x_out")
+++ grecord(ao,        "9idcLAX:SAXS:waxs_x_out")
>> 9idcLAX:WAXS:WAXS_x_out"

(USAXS_Pinterms.db,9):
--- grecord(ao,        "9idcLAX:USAXS_Pin:waxs_x_in")
+++ grecord(ao,        "9idcLAX:SAXS:waxs_x_in")
>>9idcLAX:WAXS:WAXS_x_in

(USAXS_Pinterms.db,10):
--- grecord(ao,        "9idcLAX:USAXS_Pin:waxs_x_limit_offset")
+++ grecord(ao,        "9idcLAX:SAXS:waxs_x_limit_offset")
9idcLAX:WAXS:WAXS_x_limit_offset

Do NOT change SAXS:SAXS, that makes sense.

prjemian, 2019-03-19:
revised python code to rename and also create PV aliases:

```
#!/usr/bin/env python

"""
cleanup some PV names

USAXS_Pin -> SAXS
Pin_... -> SAXS_...
USAXS_WAXS -> WAXS
"""

from collections import OrderedDict

replacements = OrderedDict()
replacements["USAXS_Pin"] = "SAXS"
replacements["Pin_"] = "SAXS_"
replacements["USAXS_WAXS"] = "WAXS"
replacements["SAXS:waxs_"] = "SAXS:WAXS_"

files = """
LAXterms.db
LAXterms_settings.req
USAXSterms.db
USAXS_Pinterms.db
"""

aliases = []

for fname in files.strip().split():
    with open(fname, "r") as fin:
        print("")
        print("reading:{}".format(fname))
        fname_out = fname + ".new"
        with open(fname_out, "w") as fout:
            for line_count, text in enumerate(fin.readlines()):
                line_count += 1
                in_text = text
                for k, v in replacements.items():
                    text = text.replace(k, v)
                fout.write(text)
                if in_text != text:
                    msg = "\n({},{}):".format(fname, line_count)
                    msg += "\n{} {}".format("---", in_text.rstrip())
                    msg += "\n{} {}".format("+++", text.rstrip())
                    if fname.endswith(".db"):
                        if in_text.find("record(") >= 0:
                            old_pv = in_text.split('"')[1]
                            new_pv = text.split('"')[1]
                            alias = 'alias("{}", "{}")'.format(new_pv, old_pv)
                            aliases.append(alias + "\n")
                            msg += '\n{} {}'.format("+++", alias)
                    print(msg)
    print("wrote:{}".format(fname_out))

if len(aliases) > 0:
    with open("aliases_201903.db", "w") as f:
        f.write("# aliases to PVs which changed names, per GitHub issue\n")
        f.write("# see: https://github.com/APS-USAXS/ipython-usaxs/issues/145\n")
        f.write("\n")
        map(f.write, aliases)
```

prjemian, 2019-03-19:
IOC was rebooted today with new PV names and aliases to old PV names, correct function verified in caQtDM, inspected autosave/auto_settings.sav file and that looked OK

Client programs should refer to the <LAX TOP>/9idcLAXApp/Db/aliases_201903.db` for the complete list of PV name changes.

closed 2019-03-19 by prjemian