APS-USAXS/usaxs-bluesky-ended-2023 #146: Add tune location controls

jilavsky, 2019-02-15:
Added to spec already - and will be added in March to epics: control where preUSAXStune is run. This is bit touchy, so three PVs are added:
Variable/PV which controls where preUSAXStune is done. If set to 0, preUSAXStune is done wherever the instrument is. If set to 1, preUSAXStune moves  before tuning to position defined by the two other PVs. 
The control variable/PV is set by default always to 0 by "startup script" and needs to be set in spec command file (BS "plan"?) to 1 by user intentionally. It cannot be set automatcially or left in place - it could cause disasters. 
PV names will be provided when added to LAX.

prjemian, 2019-04-18:
Thinking we did this in early April.  We'll check and verify tomorrow.

prjemian, 2019-04-19:
Still a problem.  Here's how we tested it.

- screen usaxs_user_main.ui, tab Fly scans/SAXS/WAXS
- check Use Specific location?
- set new values for SX, SY positions (set them to different values then current for diagnostic purposes)

  - checkbox: `9idcLAX:USAXS:UseSpecificTuneLocation`  
  - SX: `9idcLAX:USAXS:preUSAXStuneSX`
  - SY: `9idcLAX:USAXS:preUSAXStuneSY`

- `RE(preUSAXStune())`  -- did SX and SY move to expected positions?  (not today)

prjemian, 2019-06-03:
Add to the `preUSAXStune()` code that if `9idcLAX:USAXS:UseSpecificTuneLocation` is set, then move sample to the X & Y positions indicated.

prjemian, 2019-06-04:
FYI, the signals are all available now:

```
In [1]: terms.preUSAXStune.use_specific_location.value                                                                                      
Out[1]: 0

In [2]: terms.preUSAXStune.sx.value                                                                                                         
Out[2]: 2.0

In [3]: terms.preUSAXStune.sy.value                                                                                                         
Out[3]: -3.0
```

commit e444be61 (2019-06-04): fixes #146
PR #237 (2019-06-05): enable option to move SX & SY in preUSAXStune()
fixes #146
closed 2019-06-05 by prjemian