APS-USAXS/usaxs-bluesky-ended-2023 #150: need Device to contain signals from PSS, BLEPS, FE-EPS, ...

prjemian, 2019-02-21:
Instrument scientist suggests one location to find this pile of PVs seems most logical.

prjemian, 2019-02-21:
This might refactor some existing support such as `28-protection_plc.py`

prjemian, 2019-02-21:
started on this with commit 615f1de

prjemian, 2019-06-06:
https://github.com/APS-USAXS/ipython-usaxs/blob/bcce7a610b0c2008835f4e7c18699aa350721923/profile_bluesky/startup/28-protection_plc.py#L122

based on https://github.com/APS-USAXS/ipython-usaxs/blob/bcce7a610b0c2008835f4e7c18699aa350721923/profile_bluesky/startup/28-protection_plc.py#L6-L19

Also:

```
10-devices.py:class PSS_Parameters(Device):
21-aps.py:bc_station_beam_stop = PSS_Parameters(name="bc_station_beam_stop")
21-aps.py:    The PSS has a beam plug just before the C station
21-filters_shutters.py:    FE_shutter = ApsPssShutterWithStatus(
21-filters_shutters.py:    mono_shutter = ApsPssShutterWithStatus(
21-filters_shutters.py:    FE_shutter = SimulatedApsPssShutterWithStatus(name="FE_shutter")
21-filters_shutters.py:    mono_shutter = SimulatedApsPssShutterWithStatus(name="mono_shutter")
21-filters_shutters.py:    usaxs_shutter = SimulatedApsPssShutterWithStatus(name="usaxs_shutter")
```

prjemian, 2019-06-06:
@jilavsky, @jkirchman , @kmpeters : What other BL-EPS or FE-EPS signals are interesting to USAXS data collection?

jilavsky, 2019-06-06:
Look at 9idcLAX:blCalc:userCalc1 "BeamInHutch". It has most PVs today used to evaluate if instrument has chance to operate. These are PVs telling us if hutch shutters are opened. 
We will also should have BLEPS main "OK" value and probably permit for white beam shutter. Those should be included in any BS installation anyway. 

The main reason for all of these is post-mortem evaluation  : Why did it failed?

prjemian, 2019-06-06:
info on that PV:

```
usaxs@usaxscontrol ~/.../profile_bluesky/startup $ cainfo 9idcLAX:blCalc:userCalc1 
9idcLAX:blCalc:userCalc1
    State:            connected
    Host:             ioc9idclax.xray.aps.anl.gov:5064
    Access:           read, write
    Native data type: DBF_DOUBLE
    Request type:     DBR_DOUBLE
    Element count:    1
usaxs@usaxscontrol ~/.../profile_bluesky/startup $ caget 9idcLAX:blCalc:userCalc1.RTYP
9idcLAX:blCalc:userCalc1.RTYP  swait
```

prjemian, 2019-06-06:
That's a `swait` record.  Good since there is support for that in *apstools*.  Here are the non-default values:

```
In [1]: import apstools.synApps_ophyd                                                                                                       

In [2]: calcs = apstools.synApps_ophyd.userCalcsDevice("9idcLAX:blCalc:", name="calcs")                                                     

In [3]: device_read2table(calcs.calc1, show_ancient=False, fmt="markdown")                                                                  
```

name                                 | value                         | datetime                  
------------------------------------ | ----------------------------- | --------------------------
calcs_calc1_desc                     | BeamInHutch                   | 2019-06-06 10:35:07.342112
calcs_calc1_scan                     | 9                             | 2019-06-06 10:35:07.342112
calcs_calc1_calc                     | (A&&!B&&(E>20)&&(ABS(F-H)<I)) | 2019-06-06 10:35:07.342112
calcs_calc1_val                      | 1.0                           | 2019-06-06 10:35:07.342112
calcs_calc1_prec                     | 5                             | 2019-06-06 10:35:07.342112
calcs_calc1_process                  | 0                             | 2019-06-06 10:35:07.342112
calcs_calc1_oevt                     | 0                             | 2019-06-06 10:35:07.342112
calcs_calc1_outn                     |                               | 2019-06-06 10:35:07.342112
calcs_calc1_odly                     | 0.0                           | 2019-06-06 10:35:07.342112
calcs_calc1_doln                     |                               | 2019-06-06 10:35:07.342112
calcs_calc1_dold                     | 0.0                           | 2019-06-06 10:35:07.342112
calcs_calc1_dopt                     | 0                             | 2019-06-06 10:35:07.342112
calcs_calc1_oopt                     | 0                             | 2019-06-06 10:35:07.342112
calcs_calc1_flnk                     | 0                             | 2019-06-06 10:35:07.342112
calcs_calc1_channels_A_value         | 1.0                           | 2019-06-06 10:35:07.342112
calcs_calc1_channels_A_input_pv      | PA:09ID:STA_A_FES_OPEN_PL     | 2019-06-06 10:35:07.342112
calcs_calc1_channels_A_input_trigger | 0                             | 2019-06-06 10:35:07.342112
calcs_calc1_channels_B_value         | 0.0                           | 2019-06-06 10:35:07.342112
calcs_calc1_channels_B_input_pv      | PB:09ID:STA_B_SBS_CLSD_PL     | 2019-06-06 10:35:07.342112
calcs_calc1_channels_B_input_trigger | 1                             | 2019-06-06 10:35:07.342112
calcs_calc1_channels_C_value         | 0.0                           | 2019-06-06 10:35:07.342112
calcs_calc1_channels_C_input_pv      | PA:09ID:STA_C_NO_ACCESS.VAL   | 2019-06-06 10:35:07.342112
calcs_calc1_channels_C_input_trigger | 1                             | 2019-06-06 10:35:07.342112
calcs_calc1_channels_D_value         | 1.0                           | 2019-06-06 10:35:07.342112
calcs_calc1_channels_D_input_pv      | 9idcLAX:plc:Y0                | 2019-06-06 10:35:07.342112
calcs_calc1_channels_D_input_trigger | 1                             | 2019-06-06 10:35:07.342112
calcs_calc1_channels_E_value         | 101.995777528604              | 2019-06-06 10:35:07.342112
calcs_calc1_channels_E_input_pv      | S:SRcurrentAI                 | 2019-06-06 10:35:07.342112
calcs_calc1_channels_E_input_trigger | 1                             | 2019-06-06 10:35:07.342112
calcs_calc1_channels_F_value         | 10.10059118270874             | 2019-06-06 10:35:07.342112
calcs_calc1_channels_F_input_pv      | ID09:Energy                   | 2019-06-06 10:35:07.342112
calcs_calc1_channels_F_input_trigger | 1                             | 2019-06-06 10:35:07.342112
calcs_calc1_channels_G_value         | 1.0                           | 2019-06-06 10:35:07.342112
calcs_calc1_channels_G_input_pv      | 9idcLAX:userCalc1             | 2019-06-06 10:35:07.342112
calcs_calc1_channels_G_input_trigger | 1                             | 2019-06-06 10:35:07.342112
calcs_calc1_channels_H_value         | 10.0                          | 2019-06-06 10:35:07.342112
calcs_calc1_channels_H_input_pv      | 9ida:BraggEAO                 | 2019-06-06 10:35:07.342112
calcs_calc1_channels_H_input_trigger | 1                             | 2019-06-06 10:35:07.342112
calcs_calc1_channels_I_value         | 1.0                           | 2019-06-06 10:35:07.342112
calcs_calc1_channels_I_input_pv      |                               | 2019-06-06 10:35:07.342112
calcs_calc1_channels_I_input_trigger | 1                             | 2019-06-06 10:35:07.342112
calcs_calc1_channels_J_value         | 0.0                           | 2019-06-06 10:35:07.342112
calcs_calc1_channels_J_input_pv      |                               | 2019-06-06 10:35:07.342112
calcs_calc1_channels_J_input_trigger | 1                             | 2019-06-06 10:35:07.342112
calcs_calc1_channels_K_value         | 0.0                           | 2019-06-06 10:35:07.342112
calcs_calc1_channels_K_input_pv      |                               | 2019-06-06 10:35:07.342112
calcs_calc1_channels_K_input_trigger | 1                             | 2019-06-06 10:35:07.342112
calcs_calc1_channels_L_value         | 0.0                           | 2019-06-06 10:35:07.342112
calcs_calc1_channels_L_input_pv      |                               | 2019-06-06 10:35:07.342112
calcs_calc1_channels_L_input_trigger | 1                             | 2019-06-06 10:35:07.342112

prjemian, 2019-06-06:
Probably answers my questions for now.

prjemian, 2019-06-10:
Note that if we just add a Device using the generic swait record support, then we get generic names in the database.  Not very useful for post-mortem diagnostics.  Will define custom device with obvious names for these signals.  May *also* include a copy of the swait Device representation.

prjemian, 2019-06-10:
Note: *permit for white beam shutter* is described by `aps.shutter_permit ` but this applies to all beam lines.  Should also have signal indicating that station is permitted to have open shutter.

prjemian, 2019-06-10:
Need the exact PV that shutter open is permitted for 9-ID-C

prjemian, 2019-06-10:
Gather up existing signals that describe if operations are permitted in 9-ID-C and refactor code in `operations_in_9idc()`.

prjemian, 2019-06-10:
@jilavsky @kmpeters : What is the [*BLEPS main "OK" value*](https://github.com/APS-USAXS/ipython-usaxs/issues/150#issuecomment-499527192)?

PR #244 (2019-06-10): Ophyd signals for beam line diagnostics and post-mortem analyses
fixes #150
closed 2019-06-10 by prjemian