APS-USAXS/usaxs-bluesky-ended-2023 #151: 9idcLAX:USAXS:userDir should be a waveform string

prjemian, 2019-02-21:
Currently:

```
In [17]: !caget 9idcLAX:USAXS:userDir.RTYP                                                                                                           
9idcLAX:USAXS:userDir.RTYP     stringout
```

We need to watch out for string too long for EPICS! 
Solution is to make this PV an EPICS waveform string.

prjemian, 2019-02-21:
@kmpeters : this is probably for you

prjemian, 2019-03-15:
@jilavsky : noting:

```
usaxs@usaxscontrol .../iocBoot/ioc9idcLAX $ grep userDir dbl-all.txt 
9idcLAX:USAXS:userDir
9idcLAX:USAXS_Pin:userDir
```

prjemian, 2019-03-15:
these are located in :

```
USAXS_Pinterms.db:grecord(stringout, "9idcLAX:USAXS_Pin:userDir")
USAXSterms.db:grecord(stringout, "9idcLAX:USAXS:userDir")
```

and `stringout` should change to `waveform` with additional fields such as this proposal:

existing:

    grecord(stringout, "9idcLAX:USAXS:userDir")

proposed:

    grecord(waveform, "9idcLAX:USAXS:userDir") {
        field(DESC, "directory name for user files")
        field(FTVL, "CHAR")
        field(NELM, 1024)
    }

@kmpeters: Should use FTVL="CHAR" or FTVL="STRING"?
@jilavsky: Is 1024-character string long enough?  2k?

`DESC` field is optional

jilavsky, 2019-03-15:
1024-character limit is fine, its already too long for Igor usability...

prjemian, 2019-03-15:
Making subversion changes now.  @jkirchman, @kmpeters: take note as EPICS clients may need to be modified (SPEC, caQtDM, ...)

<details>
<summary>svn diff (click to see)</summary>
<p>

```
usaxs@usaxscontrol .../9idcLAXApp/Db $ svn diff USAXS_Pinterms.db USAXSterms.db
Index: USAXS_Pinterms.db
===================================================================
--- USAXS_Pinterms.db	(revision 11512)
+++ USAXS_Pinterms.db	(working copy)
@@ -67,7 +67,12 @@
 
 grecord(stringout, "9idcLAX:USAXS_Pin:state")
 grecord(stringout, "9idcLAX:USAXS_Pin:timeStamp")
-grecord(stringout, "9idcLAX:USAXS_Pin:userDir")
+#grecord(stringout, "9idcLAX:USAXS_Pin:userDir")
+grecord(waveform,  "9idcLAX:USAXS_Pin:userDir") {
+    field(DESC, "directory name for user files")
+    field(FTVL, "CHAR")
+    field(NELM, 1024)
+}
 grecord(stringout, "9idcLAX:USAXS_Pin:userName")
 
 grecord(ao,        "9idcLAX:USAXS_Pin:AcquireTime")
Index: USAXSterms.db
===================================================================
--- USAXSterms.db	(revision 10948)
+++ USAXSterms.db	(working copy)
@@ -9,7 +9,12 @@
 grecord(longout,   "9idcLAX:USAXS:specScan")
 grecord(stringout, "9idcLAX:USAXS:state")
 grecord(stringout, "9idcLAX:USAXS:timeStamp")
-grecord(stringout, "9idcLAX:USAXS:userDir")
+#grecord(stringout, "9idcLAX:USAXS:userDir")
+grecord(waveform,  "9idcLAX:USAXS:userDir") {
+    field(DESC, "directory name for user files")
+    field(FTVL, "CHAR")
+    field(NELM, 1024)
+}
 grecord(stringout, "9idcLAX:USAXS:userName")
 grecord(stringout, "9idcLAX:USAXS:macroFile")
 grecord(stringout, "9idcLAX:USAXS:macroFileTime")
```
</p></details>

prjemian, 2019-03-15:
We can close this issue once the SPEC, caQtDM, areaDetector, and Bluesky EPICS clients have been checked.

* [ ] SPEC
* [ ] caQtDM
* [ ] areaDetector
* [ ] Bluesky

prjemian, 2019-03-19:
checked in caQtDM today, no problems observed

prjemian, 2019-03-20:
@jkirchman : Can you check that SPEC can render `9idcLAX:USAXS:sampleTitle` as a string?  It was changed [recently](https://github.com/APS-USAXS/livedata/issues/6) to be a longer string by chaning the PV from stringout to waveform (with FTVL="CHAR").

jkirchman, 2019-03-20:
I tested this on my SPEC. No problems... as long as you specify that it's a string.

65.SPEC60801> p epics_get("9idcLAX:USAXS:sampleTitle","string")
multi word test

66.SPEC60801> p epics_get("9idcLAX:USAXS:sampleTitle")
{109, 117, 108, 116, 105, 32, 119, 111, 114, 100, 32, 116, 101, 115, 116, 0 <241 repeats>}

jilavsky, 2019-03-21:
Jeff,

does this impact also epics_put commands in any way?

Sincerely

Jan
*****
Dr. Jan Ilavsky, physicist
APS, Argonne National Laboratory
IL, USA

jkirchman, 2019-03-21:
No. There's no change needed for epics_put.

jilavsky, 2019-03-29:
Urgent: Area Detector has problem with waveform for sampleTitile. 
SAXS/WAXS sample name is written as 8-bit value ("85") instead of string. We need to fix this before Tuesday.

jilavsky, 2019-03-30:
Fixed, AD manual points here: https://cars9.uchicago.edu/software/epics/areaDetectorDoxygenHTML/class_p_v_attribute.html
and that explains, that we need to force string type on epics PV read, cahnge DBL_NATIVE to DBL_STRING.

closed 2019-03-30 by jilavsky