APS-USAXS/usaxs-bluesky-ended-2023 #154: PV for macroFile : file name or file contents?

prjemian, 2019-03-15:
in LAX, there is a PV for the (SPEC) macro file

```
grecord(stringout, "9idcLAX:USAXS:macroFile")
grecord(stringout, "9idcLAX:USAXS:macroFileTime")
```

Is this a PV name or is it expected to hold the contents of the file?

Either way, this should probably be a waveform string (see #151) instead of a stringout.

prjemian, 2019-03-15:
There is a *different* PV for specFile (name of the SPEC data file):

```
grecord(stringout, "9idcLAX:USAXS:specFile")
```

jilavsky, 2019-03-15:
Macro file is name of command file submitted to do the job. It should be somethign like "usaxs.mac" but lots of time it is more complicated file name.
spec file is name of spec working file, generally something in  line with "03_15_testingStuff.dat"

prjemian, 2019-03-15:
OK, these do not need changes to become waveform strings, not expected to get too long for [*stringout* record](https://epics.anl.gov/EpicsDocumentation/AppDevManuals/RecordRef/3.12/stringout.html).

closed 2019-03-15 by prjemian