APS-USAXS/usaxs-bluesky-ended-2023 #161: LAX IOC PVs recently changed from ao to bo record type

prjemian, 2019-04-02:
Some of the PVs recently changed from EPICS `ao` (float) records to `bo` (boolean) records.  This has some consequences for code that expects the value in a certain way.  The changes were motivated by how these PVs are used and how they are represented in the Bluesky software.

Text values should be supplied to represent when the `bo` records have values of zero and one.  Today, additional fields were added to `StopBeforeNextScan`.  Other `bo` records in this list need similar additions.  

**Also**, `FlyScanNotSaved` should be changed to a `bo` record based on how it is used.  Then, it, too, will need the ZNAM and ONAM fields to be defined.  A couple other `ao` records are added here.  Check if they should be `bo` records instead.  If so, what ZNAM & ONAM values to be used with each?

From file `USAXSterms.db`:

```
grecord(ao,        "9idcLAX:USAXS:TR_MeasurePinTrans")

# 2014-08-26, JIL   Add switch to change between step and fly scans
grecord(ao, "9idcLAX:USAXS:UseFlyscan") 

grecord(bo, "9idcLAX:USAXS:RunPreUSAXStuneNext") 
{
    field(DESC, "Force preUSAXStune")
}

grecord(bo, "9idcLAX:USAXS:RunPreUSAXStuneOnQdo") 
{
    field(DESC, "preUSAXStune on qdo")
}

# Stop on next scan handling PVs
grecord(bo, "9idcLAX:USAXS:StopBeforeNextScan") 
{
    field(DESC, "Stop USAXS SAXS WAXS")
    field(ZNAM, "No")
    field(ONAM, "Stop")
}
# Pause before next scan handling PVs
grecord(bo, "9idcLAX:USAXS:PauseBeforeNextScan") 
{
    field(DESC, "Pause USAXS SAXS WAXS")
}

grecord(ao,        "9idcLAX:USAXS:FlyScanNotSaved")
{
    field(DESC, "0 when Python wrote data file")
}
```

prjemian, 2019-06-03:
Is this work complete?

prjemian, 2019-06-06:
Note: PV names were also changed with PV aliases back to previous names.  Configuration files are here:

```
usaxs@usaxscontrol /share1/USAXS_data $ /APSshare//bin/alivedb | grep -i lax
ioc9idcLAX (164.54.109.181) 0 - Up time: 2 days, 16 hr
  TOP = /net/s9dserv/xorApps/epics/synApps_5_7/ioc/9idcLAX
  target name = ioc9idcLAX
  startup script = /xorApps/epics/synApps_5_7/ioc/9idcLAX/iocBoot/ioc9idcLAX/st.cmd
usaxs@usaxscontrol /share1/USAXS_data $ grep RunPreUSAXStuneNext /net/s9dserv/xorApps/epics/synApps_5_7/ioc/9idcLAX/9idcLAXApp/Db/*.db
/net/s9dserv/xorApps/epics/synApps_5_7/ioc/9idcLAX/9idcLAXApp/Db/aliases_201903.db:alias("9idcLAX:RunPreUSAXStuneNext", "9idcLAX:USAXS:RunPreUSAXStuneNext")
/net/s9dserv/xorApps/epics/synApps_5_7/ioc/9idcLAX/9idcLAXApp/Db/LAXterms.db:grecord(bo, "9idcLAX:RunPreUSAXStuneNext") 
```

prjemian, 2019-06-06:
See that `USAXSterms.db` is now `LAXterms.db`

prjemian, 2019-06-06:
Aaaack!  *Except for* this:

```
usaxs@usaxscontrol .../9idcLAXApp/Db $ grep UseFlyscan *.db
USAXSterms.db:grecord(ao, "9idcLAX:USAXS:UseFlyscan") 
```

prjemian, 2019-06-06:
@kmpeters, @jilavsky, @jkirchman : I will change PV `9idcLAX:USAXS:UseFlyscan` from an `ao` (analog output) record to a `bo` (binary output) record.  We only use it to choose between two values, zero and one.  That's a clear `bo` record use.

prjemian, 2019-06-06:
Observe some PV anomalies in `LAXterms.db` (which is in subversion so this is the only place to log this issue):

```
grecord(bo,"9idcLAX:ccdProtection")
{
      field(DESC, "ccd protection bit")
      field(ZNAM, "On")
      field(ONAM, "Off")
      field(VAL,  "0")
}
grecord(bo,"9idcLAX:collectingSAXS")
{
     field(DESC, "Collecting SAXS")
     field(ZNAM, "On")
     field(ONAM, "Off")
     field(VAL,  "0")
}
grecord(bo,"9idcLAX:collectingWAXS")
{
     field(DESC, "Collecting WAXS")
     field(ZNAM, "On")
     field(ONAM, "Off")
     field(VAL,  "0")
}
```

@jilavsky, @kmpeters, @jkirchman :
Looks to me that the ONAM (one value name) and ZNAM are switched.
This is a typo I frequently catch myself making.  Sometimes not.
Or are any of these properly configured as shown?

prjemian, 2019-06-06:
subversion commits:

```
jemian@wow .../9idcLAXApp/Db $ svn ci -m "change PV type per https://github.com/APS-USAXS/ipython-usaxs/issues/161#issuecomment-499494418"
Sending        USAXSterms.db
Transmitting file data .
Committed revision 11685.
jemian@wow .../9idcLAXApp/Db $ svn ci -m "add ZNAM & ONAM fields per https://github.com/APS-USAXS/ipython-usaxs/issues/161"
Sending        LAXterms.db
Transmitting file data .
Committed revision 11686.
```

We'll be done once [this question](https://github.com/APS-USAXS/ipython-usaxs/issues/161#issuecomment-499503182) is resolved.

prjemian, 2019-06-06:
One more change as noted in original statement:

```
grecord(bo, "9idcLAX:USAXS:FlyScanNotSaved")
{
    field(DESC, "0 when Python wrote data file")
    field(ZNAM, "saved")
    field(ONAM, "unsaved")
}
```

prjemian, 2019-06-06:
```
jemian@wow .../9idcLAXApp/Db $ svn ci -m "change to bo record per https://github.com/APS-USAXS/ipython-usaxs/issues/161#issuecomment-499505736"
Sending        USAXSterms.db
Transmitting file data .
Committed revision 11687.
```

prjemian, 2019-06-06:
@jilavsky responds by email:

Seems like typos for the “collecting" PVs,

even though… "collecting SAXS” and WAXS to have On and Off is confusing.

collecting SAXS/WAXS is used that if it is 1 (=Yes?), it is collecting whatever that should be.

So, IMHO, ONAM should be Yes and ZNAM should be No. I am not sure if Yes = On in minds of humans… ‘'

In any case, I suggest these are fixed to whatever is appropriate as what we have is wrong. 

>>  field(DESC, "ccd protection bit”)

this is NOT used in spec code, so this is used by PLC ioc. This needs to be sorted out by Kevin.

prjemian, 2019-06-06:
Two down, one (`9idcLAX:ccdProtection`) to go:

```
jemian@wow .../9idcLAXApp/Db $ svn ci -m "change ZNAM & ONAM values per https://github.com/APS-USAXS/ipython-usaxs/issues/161#issuecomment-499529638"
Sending        LAXterms.db
Transmitting file data .
Committed revision 11690.
```

kmpeters, 2019-06-06:
The ccdProtection record was added at the request of David Ilavsky:
> On 6/24/11 4:04 PM, David Ilavsky wrote:
>> Hey Kevin,
>> I need a new PV. It needs to be a binary PV value 0/1 since I want to use this PV as safety for CCD shutter but I also need to have an override for certain users. So I developed some stuff in the CSS and it works but at this point I am using a local CSS generated PV which can be different for each computer. Which is bad.  I also this PV to maintain a value of 0 usually but I write to it. The value of 1 will stay for a minute and then revert to the value 0. This is so the users can override the protection but it returns after minute. Let me know if this is possible otherwise I will try to come up with something else.
>>
>> Example:
>> PVname                                       PVvalue                         time(sec)
>> 15iddLAX:ccdprotection                   0                                    0                      (protection active)
>> 15iddLAX:ccdprotection                   1                                    1                      (protection deactivated)
>> 15iddLAX:ccdprotection                   1                                    50                    (protection deactivated)
>> 15iddLAX:ccdprotection                   0                                    61                    (protection active)
>>
>> Thanks,
>> David 

I added the PV and recommended that the meanings of the values be inverted:

> On 6/27/11 12:54 PM, Kevin Peterson wrote:
>> It's not too hard, I just don't have time to implement it at the moment.
>> 
>> I added the PV to LAX.  It'll be available the next time LAX reboots. 
>> The PV name is 15iddLAX:ccdProtection.
>> 
>> You might want to consider inverting the meaning of the values of this 
>> new PV.  Usually "1" means "true" and "0" means "false".  A value of "1" 
>> for a PV named "ccdProtection" implies that protection is active.
>> 
>> Kevin
>> 
>> On 6/27/11 8:52 AM, David Ilavsky wrote:
>>> Hey Kevin,
>>> if it is too hard to get that behavior, don't worry about it just add 
>>> the PV and then email me. I will figure out a different way to do it.
>>>
>>> Thanks,
>>> David
>>>

I don't use the PV for anything and have no objections to swapping the meaning of its values.

prjemian, 2019-06-06:
@kmpeters : Thanks for that context.  So the value of the PV is not controlled by the PLC?  Sounds like the PV's value is controlled by code in Control System Studio, which we don't use any longer.  If there truly are no more clients that use this PV, we could just delete it.  @jilavsky: What do you think?

prjemian, 2019-06-06:
@jilavsky @jkirchman : I vote to remove PV `9idcLAX:ccdProtection` since it is no longer used anywhere.  Any last minute reprieve from either of you?

jkirchman, 2019-06-06:
I don't have a dog in this fight. No objections here.

prjemian, 2019-06-06:
I will delete that PV (`9idcLAX:ccdProtection`), then the *9idcLAX* IOC may be rebooted to catch all these changes.

prjemian, 2019-06-06:
done:

```
jemian@wow .../iocBoot/ioc9idcLAX $ svn ci -m "removed ccdProtection PV per https://github.com/APS-USAXS/ipython-usaxs/issues/161#issuecomment-499544999"
Sending        auto_settings.req
Transmitting file data .
Committed revision 11692.
jemian@wow .../iocBoot/ioc9idcLAX $ cd ../../9idcLAXApp/Db/
jemian@wow .../9idcLAXApp/Db $ svn ci -m "removed ccdProtection PV per https://github.com/APS-USAXS/ipython-usaxs/issues/161#issuecomment-499544999"
Sending        LAXterms.db
Sending        LAXterms_settings.req
Transmitting file data ..
Committed revision 11693.
```

Why is this in two separate repos?  Only curious.

prjemian, 2019-06-06:
All work is complete.

closed 2019-06-06 by prjemian

jilavsky, 2019-06-06:
OK, fine with me. I will reboot LAX when suitable for me - we are running some test scans in spec at this time to shake down the stages. WIll be tuning motors soon. SO leave reboot to me later today.