APS-USAXS/usaxs-bluesky-ended-2023 #167: Excel file sample thickness is not handled properly

prjemian, 2019-04-03:
on analysis, sample thickness is not correct when Excel batch is run

prjemian, 2019-04-03:
We expect this value to be here: 9idcLAX:sampleThickness, while collecting SAXS data using BS this is value is 1 instead of 0.09 or whatever we set in Excel sheet.

prjemian, 2019-04-03:
We have PV problem. We have 9idcLAX:USAXS:SampleThickness and 9idcLAX:sampleThickness, where Flyscan is using the USAXS one and SAXS/WAXS are using the one without LAX. 

This needs to be resolved.

prjemian, 2019-04-11:
@jilavsky How will we resolve this?  One PV for sample thickness or three?

jilavsky, 2019-04-11:
We will fix this by going to one sampleThickness, one sampleName etc. At this time we have two sampleThickness variables and three sampleNames.  On to do for May shutdown. No chance of fixing while in operations...

prjemian, 2019-04-19:
The BS code is working fine.  Need to fix the PVs themselves.

prjemian, 2019-05-31:
Recent work done in the IOC to change PV names (and alias back to old ones).  Perhaps that work addresses this problem as well?

jilavsky, 2019-06-02:
This is addressed by PV changes. 
9idcLAX:sampleThickness is the one to use. All others are aliases to prevent code failure this cycle. Same for 9idcLAX:sampleName. 
Report any more PV changes needed, will sort out.

closed 2019-06-03 by prjemian