APS-USAXS/usaxs-bluesky-ended-2023 #193: user support tools needed

jilavsky, 2019-04-24:
For user operations in BlueSky we need to have following plans / tools:
1. energy change for beamline & USAXS (spec: mv_USAXS_E)
1. a plan which will do the pre & post scan steps as spec CollectData does
1. configuration file where we can setup various values manually so they get loaded when scans are started. 
1. Linkam controls for both Linkams and example plans for data collections
1. GUI tool to write out either Excel or suitable text plan file

prjemian, 2019-05-28:
Linkam: #208, #214, #215

prjemian, 2019-06-03:
Item 3 needs more thought.  Let's start with users writing custom python code in experiment directory.  Users could [`reload()`](https://github.com/APS-USAXS/ipython-usaxs/issues/141#issuecomment-497500593) that file as needed.

prjemian, 2019-06-03:
We're done with this.

closed 2019-06-03 by prjemian