APS-USAXS/usaxs-bluesky-ended-2023 #211: Tuning macros now have PVs for ranges

jilavsky, 2019-05-23:
We added PVs for tunning ranges. 
9idcLAX:USAXS:tune_mr_range
9idcLAX:USAXS:tune_m2rp_range
9idcLAX:USAXS:tune_msrp_range
9idcLAX:USAXS:tune_ar_range
9idcLAX:USAXS:tune_a2rp_range
9idcLAX:USAXS:tune_asrp_range

These need to be used by tuning.
Also, these need to be set to suitable defult values (see below) when new job is submitted to the instrument, unless specifically overwritten by user. Depends on X-ray energy.

  USAXS_tune_mr_range   =  0.004    # range for tune mr for Si 220 crystals, this is energy independent
  if(__CurrentSystemEn<10.99){  
    USAXS_tune_ar_range   =  0.002    # range for tune ar for about 10keV 
    USAXS_tune_m2rp_range =  10        # range for tune m2rp for about 10keV
    USAXS_tune_a2rp_range =  7        # range for tune a2rp for about 10keV
    USAXS_tune_msr_range  =  5        # range for tune msr for about 10keV
    USAXS_tune_asr_range  =  3        # range for tune asr for about 10keV 
    set_USAXS_MINSTEP 0.000045
  }
  if(__CurrentSystemEn>10.99 && __CurrentSystemEn<12.99){  
 	### these are for ~12keV
     USAXS_tune_mr_range    =   0.004    # range for tune mr for Si 220 crystals
     USAXS_tune_ar_range    =   0.0015   # range for tune ar for about 12keV 
     USAXS_tune_m2rp_range  =   9        # range for tune m2rp for about 12keV
     USAXS_tune_a2rp_range  =   5        # range for tune a2rp for about 12keV - Prague
     USAXS_tune_msr_range   =   3        # range for tune msr for about 12keV
     USAXS_tune_asr_range   =   3      # range for tune asr for about 12keV 
     set_USAXS_MINSTEP 0.000035
  }
  if(__CurrentSystemEn>13 && __CurrentSystemEn<18.099){  
	### these are for 16-18keV
    USAXS_tune_mr_range    =   0.002    # range for tune mr for Si 220 crystals
    USAXS_tune_ar_range    =   0.0012   # range for tune ar for about 17keV 
    USAXS_tune_m2rp_range  =   8        # range for tune m2rp for about 17keV
    USAXS_tune_a2rp_range  =   5        # range for tune a2rp for about 17keV - Prague
    USAXS_tune_msr_range   =   3        # range for tune msr for about 17keV
    USAXS_tune_asr_range   =   3        # range for tune asr for about 17keV 
    set_USAXS_MINSTEP 0.000025
 }  
  if(__CurrentSystemEn>18.1){  
	### these are for 18-21keV, assume 220
    USAXS_tune_mr_range    =   0.0020    # range for tune mr for Si 220 crystals
    USAXS_tune_ar_range    =   0.0010     # range for tune ar for about 21keV 
    USAXS_tune_m2rp_range  =   6        # range for tune m2rp for about 21keV
    USAXS_tune_a2rp_range  =   5        # range for tune a2rp for about 21keV - Prague
    USAXS_tune_msr_range   =   3        # range for tune msr for about 21keV
    USAXS_tune_asr_range   =   3        # range for tune asr for about 21keV 
    set_USAXS_MINSTEP 0.000025
 }
  if(__CurrentSystemEn>20.8){  
	### these are for 21 and higherkeV, assume 220
    USAXS_tune_mr_range    =   0.0020    # range for tune mr for Si 220 crystals
    USAXS_tune_ar_range    =   0.0007     # range for tune ar for about 21keV 
    USAXS_tune_m2rp_range  =   4        # range for tune m2rp for about 21keV
    USAXS_tune_a2rp_range  =   4        # range for tune a2rp for about 21keV - Prague
    USAXS_tune_msr_range   =   3        # range for tune msr for about 21keV
    USAXS_tune_asr_range   =   3        # range for tune asr for about 21keV 
    set_USAXS_MINSTEP 0.000025
 }

prjemian, 2019-06-04:
related: #230, #231, #232

prjemian, 2019-06-04:
This looks like a restatement of #129, closing as duplicate

closed 2019-06-04 by prjemian