APS-USAXS/usaxs-bluesky-ended-2023 #216: run a list of samples from a text file

prjemian, 2019-05-28:
Like #71 (and #124), read a list of samples from a text file, ad noted in #193

This is an alternative to #139 (which is to read SPEC-style usaxs.mac file)

Derive this from the `run_Excel_file()` plan.

prjemian, 2019-05-28:
This will be a bit less flexible for extra columns other than the expected ones.

jilavsky, 2019-05-28:
Suggestion : engineer way to add BlueSKy command line somehow. This way simpler experiments with small external control needs do not need to be written as separate plan. 

Identify BS line by adding a specific character to start of the line or simply by looking for lines which start with RE( . Simply, allow users to add lines which will be directly passed to Run Engine so we can control other equipment. Like LInkam. 

Note: this could be added into Excel macro also.

prjemian, 2019-05-29:
Recognizing that this has the very signature that starts a [*scope creep*](https://en.wikipedia.org/wiki/Scope_creep) wherein we will be creating a new scripting language, the first reaction is that all efforts spent on creating and maintaing that simple allowance for users to add Bluesky commands could better be spent teaching them how to use the existing bluesky scripting language tools.  At that point, they have access to all the capabilities available.

On the other hand, if we can set out now the specific additional commands to be recognized, those can be added.  We **must** avoid the temptation of *scope creep*.

The only additional command I can imagine is *set PV value* yet that is problematic to execute in a plan since it requires creating the connection to the PV ( a blocking call) within the context of a plan (non-blocking).  If, instead, it were *set signal_name value*, one would expect general users to know the names of signals already configured.  At that point, the goal of simplifying seems to have been lost.

Let's get some better experience with users attempting to write custom plans to do simple (or not so simple) things before we try to further optimize.

prjemian, 2019-05-29:
This is how I intend to implement the parser.  Where possible, I will use the same parser rules with the Excel file handling.

```
    example of reading a list of samples from a text file
    
    view of text file (line numbers shown)::
    
        [1] #List of sample scans to be run              
        [2] # pound sign starts a comment (through end of line)
        [3] 
        [4] ! scan  sx  sy  thickness   sample name
        [5] FlyScan 0   0   0   blank
        [6] FlyScan 5   2   0   blank
    
    The text file is interpreted line-by-line.
    Blank lines are ignored.
    A pound sign (#) marks the rest of that line as a comment.
    An exclamation mark (!) defines the labels to be used with 
    the next command line(s).
    All remaining lines are interpreted as commands with arguments.
    Only recognized commands will be executed.
    Unrecognized commands will be reported as comments.
```

prjemian, 2019-05-29:
Here is an example text file for parsing:

```
#List of sample scans to be run              
# pound sign starts a comment (through end of line)

! action  value
mono_shutter open

! action  x y width height
sample_slits 0 0 0.4 1.2

! action  sx  sy  thickness   sample name
FlyScan 0   0   0   blank
FlyScan 5   2   0   blank

! action  sx  sy  thickness   sample name
SAXS 0 0 0 blank

! action  value
mono_shutter close
```

prjemian, 2019-05-29:
Here is the command list that is generated:

```
In [37]: _text_commands_file_parser("/tmp/actions.txt")
Out[37]: 
[('mono_shutter', OrderedDict([('value', 'open')]), 5, 'mono_shutter open'),
 ('sample_slits',
  OrderedDict([('x', '0'), ('y', '0'), ('width', '0.4'), ('height', '1.2')]),
  8,
  'sample_slits 0 0 0.4 1.2'),
 ('preusaxstune', OrderedDict(), 10, 'preusaxstune'),
 ('FlyScan',
  OrderedDict([('sx', '0'),
               ('sy', '0'),
               ('thickness', '0'),
               ('sample', 'blank')]),
  13,
  'FlyScan 0   0   0   blank'),
 ('FlyScan',
  OrderedDict([('sx', '5'),
               ('sy', '2'),
               ('thickness', '0'),
               ('sample', 'blank')]),
  14,
  'FlyScan 5   2   0   blank'),
 ('SAXS',
  OrderedDict([('sx', '0'),
               ('sy', '0'),
               ('thickness', '0'),
               ('sample', 'blank')]),
  17,
  'SAXS 0 0 0 blank'),
 ('mono_shutter', OrderedDict([('value', 'close')]), 20, 'mono_shutter close')]
```

prjemian, 2019-05-29:
Note that the "sample name" column was parsed into `('sample', 'name')`, two columns.  Need to change that to one word `sample_name` or change the parser to allow one blank and require more than one blank between labels.  Let's try our luck with the first one.

prjemian, 2019-05-29:
revising per LGTM code review recommendations before proceeding to test, will request review from @jilavsky before merge

prjemian, 2019-05-29:
Also, before merge, look at `run_Excel_file()` for possibilities with refactor of common code.

prjemian, 2019-05-29:
@jilavsky: Support is flexible now.  Call one plan to read either Excel or text command file.  Here are examples showing a summary of the two files:

```
In [109]: summarize_command_file("actions.txt")
Command file: actions.txt
====== ============ ==========================================
line # action       parameters                                
====== ============ ==========================================
5      mono_shutter value=open                                
8      sample_slits x=0, y=0, width=0.4, height=1.2           
10     preusaxstune                                           
13     FlyScan      sx=0, sy=0, thickness=0, sample_name=blank
14     FlyScan      sx=5, sy=2, thickness=0, sample_name=blank
15     SAXS         sx=0, sy=0, thickness=0, sample_name=blank
18     mono_shutter value=close                               
====== ============ ==========================================

In [110]: summarize_command_file("actions.xlsx")
Command file: actions.xlsx
====== ============ =========================================================
line # action       parameters                                               
====== ============ =========================================================
1      mono_shutter sx=open                                                  
2      USAXSscan    sx=45.07, sy=98.3, thickness=0.0, sample_name=Water Blank
3      saxsExp      sx=45.07, sy=98.3, thickness=0.0, sample_name=Water Blank
4      waxwsExp     sx=45.07, sy=98.3, thickness=0.0, sample_name=Water Blank
5      USAXSscan    sx=12, sy=12.0, thickness=1.2, sample_name=plastic       
6      USAXSscan    sx=12, sy=37.0, thickness=0.1, sample_name=Al foil       
7      mono_shutter sx=close                                                 
====== ============ =========================================================

In [111]: 
```

commit c96f0df1 (2019-05-29): fixes #216, fixes #218 (once PR #217 is merged)
PR #217 (2019-05-30): samples scanned from text file
fixes #216
closed 2019-05-30 by prjemian