APS-USAXS/usaxs-bluesky-ended-2023 #219: Engineer pre- and post- run code

jilavsky, 2019-05-29:
We need to add code currently residing in CollectData spec macro. This needs to be run in Excel and text run_code before and after execution of user macros. 
Full feature list needs to be extracted from spec, but it opens Monoshutter, syncs order numbers (if requested, which is spec variable at this time), sets a PV telling GUI that data collection is in progress and copies run file for web page/run file logs. After finishing this code lowers "running" PV and does some cleanup.

jilavsky, 2019-05-29:
oops, part of #193 already.

prjemian, 2019-05-29:
> Assigned #219 <https://github.com/APS-USAXS/ipython-usaxs/issues/219> to
> @prjemian <https://github.com/prjemian>.

Check out beforePlan and afterPlan in 50-plans.py

jilavsky, 2019-05-29:
Seems missing "copy command file to web page". Will need BS version of spec command postUSAXSDoFIle2WWW

```
def postUsaxsDofile2WWW '{
  local cmd macFile theTarget date_str  num
  if ($# != 0) {
    print "usage: postUsaxsDofile2WWW"
    exit
  }

  # strip away any file path from DOFILE
  macFile = stripFilePath(DOFILE)
  epics_put ("9idcLAX:USAXS:macroFile",      macFile)
  epics_put ("9idcLAX:USAXS:macroFileTime",  date())
  theTarget = "/tmp/specmacro.txt"
  cmd = sprintf("/bin/cp -f %s %s ", DOFILE, theTarget )
  print cmd
  unix(cmd)

  # Archive each usaxs.mac (or sbusaxs.mac) file.
  #
  # Since the users execute a batch of scans using the
  # usaxs.mac (or sbusaxs.mac) file, they overwrite any
  # previous batch file they used.  This bit of code
  # makes an archive copy of each batch file.
  #
  # make the date _readable_ instead of epoch
  date_str = "date_str is now a string variable"
  unix("/bin/date \"+%Y%m%d-%H%M%S\"", date_str)
  split(date_str, date_arr, "\n")
  fmt = "/bin/cp -f %s /share1/log/macros/%s-%s"
  cmd = sprintf(fmt, DOFILE, date_arr[0], macFile )
  print cmd
  unix(cmd)
}'

``

commit 5c9235a8 (2019-05-30): fixes #219
closed 2019-05-30 by prjemian