APS-USAXS/usaxs-bluesky-ended-2023 #221: livedata not processing bluesky's FlyScan data

prjemian, 2019-05-30:
During today's testing of #217, it was noted that the [livedata WWW page](https://usaxslive.xray.aps.anl.gov/) did not show graphs of the current data.  Also, the [`actions.txt`](https://usaxslive.xray.aps.anl.gov/actions.txt) link generates a 404 error rather than expected content.

prjemian, 2019-05-30:
Likely cause of no plottable data is that the instrument has no beam today.  (APS is operating but the other station has the beam.)  It is noted that any flyscan data collection results in such a small number of reduced data that it fails acceptance testing before plotting.

tl;dr: needs beam

prjemian, 2019-06-04:
Checking again today.

prjemian, 2019-06-04:
Still needs beam and new good data sets to test conclusively.

jilavsky, 2019-06-25:
looks good now during operations

closed 2019-06-25 by jilavsky