APS-USAXS/usaxs-bluesky-ended-2023 #222: add action handlers

prjemian, 2019-05-30:
As [previously noted](https://github.com/APS-USAXS/ipython-usaxs/issues/210#issuecomment-497482238), add these handlers to support reading of text files output from the *wxmtusaxs* tool:

*    CURRENT_EXPERIMENT_NAME
*    saxsExp (same as SAXS?)
*    USAXSscan (same as FlyScan?)
*    waxsExp (same as WAXS?)

prjemian, 2019-05-30:
What is the procedure to handle the `CURRENT_EXPERIMENT_NAME` action?  First guess:

```
        elif action == "current_experiment_name":
            nm = args[0]
            # FIXME: check this!
            yield from bps.mv(user_data.user_name, nm)
```

jilavsky, 2019-05-31:
Yes, these are correct action handlers. 
Note, we have PV which indicates, if USAXS should run step scan or flyscan. Obviously, BS cannot do for now step scans, so USAXSscan = Flyscan, for now.

jilavsky, 2019-05-31:
spec is simply printing in history which tab name in wxmtusaxs tool was used to generate the data. 

# /home/beams/USAXS/spec/macros/local/usaxs_commands.mac
def CURRENT_EXPERIMENT_NAME '{
	tempDummyString = "$1"
	comment "%s"  tempDummyString
}'
Therefore, this command can be ignored or printed in "spec" output file. 
NOTE: do not overwrite user name in metadata, that is obtained from scheduling system.

commit e60fde98 (2019-05-31): fixes #222
closed 2019-05-31 by prjemian