APS-USAXS/usaxs-bluesky-ended-2023 #232: TunableEpicsMotor with EpicsSignal for width

prjemian, 2019-05-31:
As noted in https://github.com/APS-USAXS/ipython-usaxs/issues/230#issuecomment-497853589, to have the tuning width expressed as a PV will require some redesign of our UsaxsMotorTunable.

Look at `apstools.devices.AxisTunerMixin()`

prjemian, 2019-05-31:
@jilavsky: Also, since `dx` & `dy` are also instances of `UsaxsMotorTunable` they should also have width PVs, even if those do not change.

prjemian, 2019-05-31:
The fundamental issue to be resolved is how to keep the float `TuneAxis.width` in sync with the new EpicsSignal for `width`.  

@jilavsky, @kmpeters : It would be **much** easier to interface in bluesky if the new *width* PVs extended the motor PV rather than creating a new PV.

axis | now | suggested new name
-- | -- | --
mr | `9idcLAX:USAXS:tune_mr_range` | `9idcLAX:aero:c3:m1_range`
m2rp | `9idcLAX:USAXS:tune_m2rp_range` | `9idcLAX:pi:c0:m2_range`
msrp | `9idcLAX:USAXS:tune_msrp_range` | `9idcLAX:pi:c0:m3_range`
ar | `9idcLAX:USAXS:tune_ar_range` | `9idcLAX:aero:c0:m1_range`
a2rp | `9idcLAX:USAXS:tune_a2rp_range` | `9idcLAX:pi:c0:m1_range`
asrp | `9idcLAX:USAXS:tune_asrp_range` | `9idcLAX:pi:c0:m4_range`

prjemian, 2019-05-31:
The new names make it easier since we would not need to modify the `__init__()` constructor to handle a separate PV for the width.  We just add a *Component* that uses the `_width` suffix.  That's what makes it easier.  Still, the need to sync the EpicsSignal with the float remains.  Probably handle that by subscribing the EpicsSignal and updating the float whenever the width is changed in EPICS.  We'll ignore the other direction of changes (if the float is changed, EPICS will not be updated).

jilavsky, 2019-06-01:
We should add pvs for the dx and dy widths, that is oversight. 

We can make aliases as needed, I guess. Alternative is if it would make sense to add to motor record fields for “tune width” and “width” where first would be for tuning width and second would be fwhm of the tune result. In long run this could be useful for wide range of motor record users... if it is not already available. This is obviously for longer term and would need some discussion in motor record maintenance group.

prjemian, 2019-06-04:
today, I can see:

```
usaxs@usaxscontrol ~/.ipython $ caget 9idcLAX:aero:c{0,2,3}:m1_range.NAME  9idcLAX:pi:c0:m{1,2,3,4}_range.NAME
9idcLAX:aero:c0:m1_range.NAME  9idcLAX:USAXS:tune_ar_range
9idcLAX:aero:c2:m1_range.NAME  9idcLAX:USAXS:tune_dy_range
9idcLAX:aero:c3:m1_range.NAME  9idcLAX:USAXS:tune_mr_range
9idcLAX:pi:c0:m1_range.NAME    9idcLAX:USAXS:tune_a2rp_range
9idcLAX:pi:c0:m2_range.NAME    9idcLAX:USAXS:tune_m2rp_range
9idcLAX:pi:c0:m3_range.NAME    9idcLAX:USAXS:tune_msrp_range
9idcLAX:pi:c0:m4_range.NAME    9idcLAX:USAXS:tune_asrp_range
```

prjemian, 2020-11-12:
Not convinced this is a high priority item but it certainly is high interest.

prjemian, 2020-11-12:
Perhaps the bidirectional synchronization can be implemented with a @property and a @property.setter?

prjemian, 2020-11-12:
At present:

```
In [36]: object_explorer(axis_tune_range)
==== ============================= =====
name PV reference                  value
==== ============================= =====
a2rp 9idcLAX:USAXS:tune_a2rp_range 7.0  
ar   9idcLAX:USAXS:tune_ar_range   0.002
asrp 9idcLAX:USAXS:tune_asrp_range 3.0  
dx   9idcLAX:USAXS:tune_dx_range   6.0  
dy   9idcLAX:USAXS:tune_dy_range   6.0  
m2rp 9idcLAX:USAXS:tune_m2rp_range 10.0 
mr   9idcLAX:USAXS:tune_mr_range   0.004
msrp 9idcLAX:USAXS:tune_msrp_range 5.0  
==== ============================= =====
```

prjemian, 2020-11-12:
The decorated methods for `width` would be applied to the `TuneAxis()` class.  That's the way forward (standard python technique, too).  Would need to replace all code that defines `.width` attribute directly, such as: https://github.com/APS-USAXS/ipython-usaxs/blob/7270f66eacc172d26c041c66c433a25f2de32abb/profile_bluesky/startup/instrument/devices/axis_tuning.py#L122

by passing a term to the `TuneAxis.__init__()` constructor that identifies the width signal in `axis_tune_range` (in this case, `axis_tune_range.mr`).

prjemian, 2020-11-12:
Work this up in a branch with a PR.

prjemian, 2020-11-13:
https://stackabuse.com/the-python-property-decorator/

jilavsky, 2020-11-17:
is fixed

PR #423 (2020-11-17): make tune width a settable PV
fixes #232
closed 2020-11-17 by jilavsky