APS-USAXS/usaxs-bluesky-ended-2023 #253: guard slit motors sometimes stuck in MOVING state

prjemian, 2019-06-18:
@kmpeters We are still having troubles with the guard slit motors getting stuck in MOVING state.  Sometimes, we find this while running bluesky, sometimes while moving them from caQtDM.  Always, this involves a session where we are using both tools.

Tried adding an explicit `caput MOTOR.STUP 1` for each motor `9idcLAC:max:c0:m{3,4,5,6]` but that did not help us.

kmpeters, 2019-06-19:
Are these the only motors that have this issue?  If so, the problem is likely to be in the 4slit support rather than the motor support.

kmpeters, 2019-06-19:
What is the state of the following PVs when the motors get stuck in the moving state?
```
$ caget 9idcLAX:GSlit1{H,V}:gate{,Open}
9idcLAX:GSlit1H:gate           Closed
9idcLAX:GSlit1H:gateOpen       Closed
9idcLAX:GSlit1V:gate           Closed
9idcLAX:GSlit1V:gateOpen       Closed
```

prjemian, 2019-06-19:
These are the only motors we are having troubles with.  I agree that the problem is in the EPICS IOC layer, not bluesky, since we also see the problem using a GUI.  Your suggestion of the 4slit support seems a good place to look.

It will take a bit to run the diagnostics since users are measuring data now.  We'll get to it but with some delay.

prjemian, 2019-06-21:
If the problem is with the 4slit database, this can be run offline in a softIOC that simulates the guard slits.

prjemian, 2019-06-24:
I've got a softIOC ready to test this: 

`caget gov:Slit1{H,V}{size,center,x{n,p}}` and related PVs

```
gov:Slit1Hsize                 0
gov:Slit1Hcenter               0
gov:Slit1Hxn                   0
gov:Slit1Hxp                   0
gov:Slit1Vsize                 0
gov:Slit1Vcenter               0
gov:Slit1Vxn                   0
gov:Slit1Vxp                   0
```

- EPICS  /APSshare/epics/base-3.14.12.5
- synApps /APSshare/epics/synApps_5_8/support
- OPTICS $(synApps)/optics-2-9-3
- db 2slit_soft.db

---------

Note that there are later updates to this support that might need an IOC update to apply:

- SUPPORT /APSshare/epics/synApps_6_0/support
- EPICS_BASE /APSshare/epics/base-3.15.6
- OPTICS $(synApps)/optics-R2-13-1
- db 2slit_soft.vdb

prjemian, 2019-06-26:
correction:

* synApps 5.7: `/net/s9dserv/xorApps/epics/synApps_5_7/ioc/9idcLAX/iocBoot/ioc9idcLAX/st.cmd`
* 2slit (not soft): 

    dbLoadRecords("$(OPTICS)/opticsApp/Db/2slit.db","P=9idcLAX:,SLIT=GSlit1H:,mXp=mxv:c0:m3,mXn=mxv:c0:m4")
    dbLoadRecords("$(OPTICS)/opticsApp/Db/2slit.db","P=9idcLAX:,SLIT=GSlit1V:,mXp=mxv:c0:m5,mXn=mxv:c0:m6")

* SUPPORT=/APSshare/epics/synApps_5_7/support
* EPICS_BASE=/APSshare/epics/base-3.14.12.3
* OPTICS=$(SUPPORT)/optics-2-9-1
* [`2slit.db`](https://github.com/epics-modules/optics/blob/97af9de5e60ae40211f8dec28d9d14182fdb190c/opticsApp/Db/2slit.db)

prjemian, 2019-09-17:
The softIOC is not ready for testing now due to IT maintenance.  The complete IOC software stack must be rebuilt after the workstation has moved to a new subnet.

kmpeters, 2019-09-18:
What are the symptoms that tell you the complete IOC software stack must be rebuilt?

prjemian, 2019-09-18:
My desktop computer was relocated to the new subnet and none of the files
exist.

prjemian, 2020-08-23:
Regarding the [slow search as noted above](https://github.com/APS-USAXS/ipython-usaxs/issues/253#issuecomment-678786913), this is **much** faster (using the PyMongo API directly) but also might break in future versions of databroker:

```
def _get_run_starts(db, num=1, query=None):
    _query = query or {}
    mdb = db.v2._metadatastore_db
    collection = db.v2._metadatastore_db["run_start"]
    return collection.find(query).sort("time", -1).limit(num)

import datetime
t0 = None
for i, start in enumerate(_get_run_starts(db, 1000)):
    t = start["time"]
    dt = (t0 or t) - t
    print(f"{i}\t{dt:.2f}\t{datetime.datetime.fromtimestamp(t)}")
    t0 = t
```

prjemian, 2020-08-28:
@jilavsky : Can we run this kind of testing on the actual guard slit hardware now?

jilavsky, 2020-09-03:
Sorry for missing this. We can run real hardware now. It is available. Well, motors are probably off at this time and no X-rays.

prjemian, 2020-09-03:
I'll try tomorrow when not in other meetings.  Should I contact you 
first?  If motors are off, I see little problem.

prjemian, 2020-09-03:
test code installed at USAXS instrument (in data directory `2020-09/issue253.py`):

<details>

```python
"""
work on GitHub issue #253

%run -im issue253
uids = RE(issue253_plan(times=10))

* guard slit motors sometimes stuck in MOVING state
* https://github.com/APS-USAXS/ipython-usaxs/issues/253

Scan the guard slit motors repeatedly.
Ignore all the other activities associated when
guard slits are "tuned".  Use the "noisy" detector
from ophyd's simulators.

``
In [4]: wa gslit
gslit
  Positioner                     Value       Low Limit   High Limit  Offset
  guard_slit_bot                 -0.5        -11.3694    12.6306     -7.0664
  guard_slit_inb                 -0.5        -10.3912    13.6088     -33.0632
  guard_slit_outb                0.5         -9.403      14.597      507.34
  guard_slit_top                 0.5         -13.9196    10.0804     -1.8856
  guard_slit_x                   -1.105      -73.88      28.585      -42.65
  guard_slit_y                   14.6773     -17.92      59.00621    10.44885
``
"""

from instrument.session_logs import logger
logger.info(__file__)

from bluesky import plan_stubs as bps
from bluesky import plans as bp
from instrument.devices import guard_slit
from instrument.framework import bec, RE
import matplotlib.pyplot as plt
from ophyd.sim import noisy_det

_total_tunes = 0
_tune_number = 0

def scan_xy(detectors, md=None):
    _md = dict()
    _md.update(md or {})
    yield from bp.rel_scan(detectors, guard_slit.y, .1, -.1, 21, md=_md)
    yield from bp.rel_scan(detectors, guard_slit.x, .1, -.1, 21, md=_md)

def scan_opening(detectors, md=None):
    _md = dict()
    _md.update(md or {})
    yield from bp.rel_scan(detectors, guard_slit.top, .1, -.1, 21, md=_md)
    yield from bp.rel_scan(detectors, guard_slit.bot, -.1, .1, 21, md=_md)
    yield from bp.rel_scan(detectors, guard_slit.outb, .1, -.1, 21, md=_md)
    yield from bp.rel_scan(detectors, guard_slit.inb, -.1, .1, 21, md=_md)

def select_figure(x, y):
    figure_name = f"{y.name} vs {x.name}"
    if figure_name in plt.get_figlabels():
        print(figure_name)
        return plt.figure(figure_name)

def select_live_plot(bec, signal):
    for live_plot_dict in bec._live_plots.values():
        live_plot = live_plot_dict.get(signal.name)
        if live_plot is not None:
            return live_plot

def trim_plot(bec, n, x, y):
    liveplot = select_live_plot(bec, y)
    if liveplot is None:
        return
    fig = select_figure(x, y)
    if fig is None:
        return
    ax = fig.axes[0]
    while len(ax.lines) > n:
        try:
            ax.lines.pop(0).remove()
        except ValueError as exc:
            logger.warning(
                "%s vs %s: mpl remove() error: %s",
                y.name, x.name, str(exc))
    ax.legend()
    liveplot.update_plot()

def trim_all(n=3):
    for axis in "y x top bot outb inb".split():     # in order of tune
        trim_plot(bec, n, getattr(guard_slit, axis), noisy_det)

def scan_guard_slit(md=None):
    global _tune_number, _total_tunes
    _tune_number += 1
    _md = dict(
        purpose="test if guard slit motors sometimes stuck in MOVING state",
        URL="https://github.com/APS-USAXS/ipython-usaxs/issues/253",
        tune_number=_tune_number,
        tune=f"tune_Gslits() {_tune_number} of {_total_tunes}",
    )
    _md.update(md or {})
    logger.info("# tune number: %d", _tune_number)
    detectors = [noisy_det,]
    yield from scan_xy(detectors, md=_md)
    yield from scan_opening(detectors, md=_md)
    yield from bps.sleep(3.0)   # allow all plots to finish drawing
    trim_all()

def issue253_plan(times=1):
    """
    repeat the scans of the guard slit axes

    uids = RE(issue253_plan(times=2))
    """
    global _tune_number, _total_tunes
    logger.info("# ------- Testing for issue #253 with %d iterations", times)
    _tune_number = 0
    _total_tunes = times
    yield from bps.repeat(scan_guard_slit, num=times)
```

</details>

prjemian, 2020-09-03:
short test in progress to verify code operation:

[image]

prjemian, 2020-09-03:
@jilavsky - I'll queue up 1000 repetitions of the six-motor "pseudo-tune" of the guard slits, looking for the original problem where one of the motors gets stuck in MOVING state.  This should be run from the Jump session so you, too, can see what is happening.

prjemian, 2020-09-03:
Test of 1000 iterations started at ~18:28

[image]

prjemian, 2020-09-03:
Note that old lines from each plot are discarded.  At most, 3 or 4 lines on each.

prjemian, 2020-09-04:
172 iterations at 9:45 pm and no stuck MOVING (~52 iterations/hr, 1000 iterations should be complete by ~1:30 pm tomorrow)

prjemian, 2020-09-04:
664 iterations at 7:15 AM and no stuck MOVING

prjemian, 2020-09-04:
RunEngine errored after 732 iterations, 8:34 AM with ValueError

jilavsky, 2020-09-08:
I agree, that we are not seeing motor hanging.  I do not see (or cannot find) the code being run. Tuning GSlits involves multiple steps. Scanning up to 6 motors. Obviously, that works, tested lots of times. Finding center positions (just code, cannot hang motors). And setting motors to proper positions. For x and y this is simple motor move to calculated center. But for slit blades this is using the epics soft slits record or whatever the name is. That last step was what we suspected from failing. Was that tested in these scans? I am not sure how that is used in BS, would need to look at the code.

prjemian, 2020-09-08:
@jilavsky, @kmpeters : Thanks for your reviews. 

The test code is in the **Details** dropdown of this message (above): https://github.com/APS-USAXS/ipython-usaxs/issues/253#issuecomment-686812351

The code was modeled after exactly the operations taken by the Tune_GuardSlits code.  It scans all six motors in the same sequence (y, x, top, bottom, outboard, inboard).  Since the result of each *tune* is unimportant, that was not done in the test code above.  If the tunes had scanned (tuned) either slit's gap or center, that would have used part of the 2slit.db code.  However, operation of *any* of the real motors exercises the 2slit.db code as it evaluates the effects on the virtual motors.

prjemian, 2020-09-09:
Leave this issue open for now.  If problem is not seen again, close automatically at next milestone.  Moving to that milestone now.

commit edafd6ae (2020-09-14): fixes #253 fixes #387
closed 2020-09-14 by prjemian