APS-USAXS/usaxs-bluesky-ended-2023 #255: USAXS instrument is not reset after RE.abort()

jilavsky, 2019-06-19:
When we decide to stop and abort a plan, the instrument is not reset to starting condition.

prjemian, 2019-09-16:
Dan Allan says to do this reset properly, we need to override the device's `stop(success)` method. When the RunEngine is exiting, whether normally or due to an error, it  calls `device.stop(success)` on every device that it has moved. It sounds like in this case, we should reset if `success` is False.

prjemian, 2019-09-17:
But which device if this is a comprehensive instrument reset?  Perhaps related to #254?

prjemian, 2019-10-01:
commit 98bc0b99601f9c385d166ae9df014df998b567db is for #307, not supposed to close this issue

jilavsky, 2019-11-26:
We have spec macro for USAXS/Flyscan reset. This is different than reset when running SAXS or WAXS or any other operations. 
USAXS/FLyscan is in usaxs_uascan.mac:def resetUSAXS 

/home/beams/USAXS/spec/macros/local

In SAXS or WAXS basically, after stopping running mode_SAXS or mode_WAXS should be the reset.

prjemian, 2020-01-14:
There is a reset position decorator from bluesky for the plan that provides this capability for simpler put the motors back to where they were at the plan's start.  For more complex resets, there is a finally wrapper that takes a plan of work to do and runs (in all cases except halt) it.

prjemian, 2020-01-14:
`RE.halt()` means *do not touch the hardware now*.  All the others are more orderly.

prjemian, 2020-01-14:
* https://blueskyproject.io/bluesky/generated/bluesky.preprocessors.finalize_decorator.html?highlight=finalize#bluesky.preprocessors.finalize_decorator
* also `contingency_decorator()`

prjemian, 2020-02-21:
Let's paste a copy of the SPEC `reset()` code here for inspection.

prjemian, 2020-11-12:
@jilavsky Can you paste the SPEC reset macro here?

jilavsky, 2020-11-14:
OK, this will be bit more difficult than I expected. 
Spec macros are below, there is resetUSAXS, it gets run ONLY when running USAXS step scan. We also have resetFlyscan (also below). And we do regular ctrl-C ("stop") at any other time. 
Not sure how this can be implemented in BS, may be there is some way to modify the abort code... 

<resetFlyscan>

```
def resetFlyScan '{
  global SPEC_STD_TITLE
  global useSBUSAXS
  global NOTIFY_ON_RESET
  local _abortstat
   #####################################################
   comment "Resetting Fly USAXS, hang on, this is complex procedure"
   epics_put ("9idcLAX:USAXS:state",     sprintf("Resetting FlyScan"))
   setplot 131
   # close the shutter after each scan to preserve the detector
   closeTiFilterShutter
   # start feedback 
   DCMfeedbackON
   # turn on the AutoCount mode on the scaler
   epics_put(sprintf("%s.CONT",_SCALER_PV), "1") #from jfk
   # turn on the photodiode to auto+bkg mode. I0 and I00 to manual mode
   modeAutoBRange_UPD
   modeManualRange_I0
   modeManualRange_I00
   # UnPause autosave on LAX to prevent delays in PVs processing.
   epics_put ("9idcLAX:SR_disable",0)			## this enables autosave
   # clear the user bit to indicate USAXS scan is not running
   epics_put ("9idcLAX:USAXS:scanning",  0)
   # abort the USAXS fly scan epics code
   epics_put ("9idcLAX:USAXSfly:Abort",  1)
   epics_put ("9idcLAX:traj1:Abort",  1)
   epics_put ("9idcLAX:traj2:Abort",  1)
   sleep(2) 
    while (epics_get("9idcLAX:USAXSfly:Start") != "Done")   {
        comment "Still waiting for FlyScan to abort"
        sleep(1)
    }
    ## stop waiting for writing out data if needed.
    epics_put("9idcLAX:USAXS:FlyScanNotSaved", 0)
    waitmove; get_angles
     A[dy] = DY0
     A[ay] = AY0
     A[ar] = AR_VAL_CENTER
     if(useSBUSAXS){
       A[asrp] = ASRP0
     }
     move_em; waitmove
   #####################################################
   # disable asrp link to ar for 2D USAXS
   FS_disableARSP
   #####################################################
   TITLE = SPEC_STD_TITLE
   epics_put (sprintf("9idcUSX:%s:mode",PDstring), 2)
   epics_put (sprintf("%s.CONT",_SCALER_PV),        1)
   epics_put ("9idcLAX:USAXS:state",           sprintf("FlyScan reset complete"))
   if(NOTIFY_ON_RESET) { sendNotifications("FlyScan has reset","spec has encountered a problem and reset the FlyScan.");}
   chk_beam_off
   rdef _cleanup3 \'\'
   # un set dataCollectionInProgress so GUI (and other tools) know, user is collecting data...  (1-not running, 0 running)
   epics_put("9idcLAX:dataColInProgress", 0) 
}'

```
</resetFlyscan>

<resetUSAXS>

```
def resetUSAXS '{
  global SPEC_STD_TITLE
  global useSBUSAXS
  global NOTIFY_ON_RESET
  comment "Resetting USAXS"
  epics_put ("9idcLAX:USAXS:state",       "resetting motors")
  setplot 131
  # start feedback 
  DCMfeedbackON
  # turn on the AutoCount mode on the scaler
  epics_put(sprintf("%s.CONT",_SCALER_PV), "1") #from jfk
  # turn on the photodiode to auto+bkg mode. I0 and I00 to manual mode
  modeAutoBRange_UPD
  modeManualRange_I0
  modeManualRange_I00
  # close the shutter after each scan to preserve the detector
  closeTiFilterShutter
  # clear the user bit to indicate USAXS scan is not running
  epics_put ("9idcLAX:USAXS:scanning",  0)
     waitmove; get_angles
     A[dy] = DY0
     A[ay] = AY0
     A[ar] = AR_VAL_CENTER
     if(useSBUSAXS){
       A[asrp] = ASRP0
     }
     move_em; waitmove
  TITLE = SPEC_STD_TITLE
  epics_put (sprintf("9idcUSX:%s:mode",PDstring), 2)
  epics_put (sprintf("%s.CONT",_SCALER_PV), 1) #from jfk:vsc:c0 -xjiao
  epics_put ("9idcLAX:USAXS:state",       "USAXS reset complete")
  if(NOTIFY_ON_RESET) { sendNotifications("USAXS has reset","spec has encountered a problem and reset the USAXS.");}
  chk_beam_off
  rdef _cleanup3 \'\'
  # un set dataCollectionInProgress so GUI (and other tools) know, user is collecting data...  (1-not running, 0 running)
  epics_put("9idcLAX:dataColInProgress", 0)
}'

```
</resetUSAXS>

prjemian, 2020-11-18:
Needs a clear discussion on exactly what is to be done and then a way to ensure that happens.  Not happening today so move to next milestone.

prjemian, 2023-04-11:
See https://bcda-aps.github.io/apstools/dev/examples/ho_contingency.html#The-finalize_decorator()

commit c687c460 (2019-10-01): fixes #255