APS-USAXS/usaxs-bluesky-ended-2023 #257: metadata print during scan is not pretty

jilavsky, 2019-06-19:
This console log stream looks ugggly (the `metadata=` line is way too long)

```
Finished SAXS/WAXS data collection in 32.47950220108032 seconds.
file line 33:           preUSAXStune
Moving WAXS out of beam                                                                                                                                    
Removed WAXS from beam position                                                                                                                            
Moving SAXS out of beam                                                                                                                                    
Removed SAXS from beam position
Moving to USAXS mode
USAXS is in position                                                                                                                                       
metadata={'full_filename': '/share1/USAXS_data/2019-06/usaxs.mac', 'filename': 'usaxs.mac', 'line_number': 33, 'action': 'preUSAXStune', 'parameters': [], 'archive': OrderedDict([('text', 'Command file: usaxs.mac\n====== ======================= =======================\nline # action                  parameters             \n====== ======================= =======================\n2      CURRENT_EXPERIMENT_NAME Calibration            \n21     saxsExp                 40, 0, 1, GC_SRM3600   \n23     saxsExp                 20.000, 0.000, 1, Blank\n27     waxsExp                 40, 0, 1, GC_SRM3600   \n29     waxsExp                 20.000, 0.000, 1, Blank\n33     preUSAXStune                                   \n35     USAXSscan               40, 0, 1, GC_SRM3600   \n37     USAXSscan               20.000, 0.000, 1, Blank\n====== ======================= =======================\n'), ('source', '/home/beams11/USAXS/.ipython/profile_bluesky/startup/50-plans.py'), ('is_file', True), ('line', 692), ('caller', 'execute_command_list'), ('caller_code', '    print(text)\n    archive = instrument_archive(text)\n'), ('source_contents', ['print(__file__)\n', '\n', '"""\n', 'Bluesky plans (scans)\n', '\n', 'PLANS\n', '\n', '    afterPlan()\n', '    beforePlan()\n', '    execute_command_list()\n', '    FlyScan()\n', '    preUSAXStune()\n', '    run_command_file()\n', '    SAXS()\n', '    WAXS()\n', '    uascan()\n', '\n', 'UTILITIES\n', '\n', '    command_list_as_table()\n', '    get_command_list()\n', '    makeOrderedDictFromTwoLists()\n', '    parse_Excel_command_file()\n', '    parse_text_command_file()\n', '    postCommandsListfile2WWW()\n', '    split_quoted_line()\n', '    summarize_command_file()\n', '\n', '"""\n', '\n', '\n', 'def uascan():\n', '    """\n', '    USAXS step scan\n', '\n', '    https://github.com/APS-USAXS/ipython-usaxs/issues/8\n', '    """\n', "    # TODO: needs proper args & kwargs matching SPEC's signature\n", '\n', '\n', 'def preUSAXStune(md={}):\n', '    """\n', '    tune the USAXS optics *only* if in USAXS mode\n', '\n', '    USAGE:  ``RE(preUSAXStune())``\n', '    """\n', '    yield from bps.mv(\n', '        monochromator.feedback.on, MONO_FEEDBACK_ON,\n', '        mono_shutter, "open",\n', '        ccd_shutter, "close",\n', '    )\n', '    yield from IfRequestedStopBeforeNextScan()         # stop if user chose to do so.\n', '\n', '    yield from mode_USAXS()\n', '\n', '    if terms.preUSAXStune.use_specific_location.value in (1, "yes"):\n', '        yield from bps.mv(\n', '            s_stage.x, terms.preUSAXStune.sx.value,\n', '            s_stage.y, terms.preUSAXStune.sy.value,\n', '            )\n', '\n', '    yield from bps.mv(\n', '        # ensure diode in place (Radiography puts it elsewhere)\n', '        d_stage.x, terms.USAXS.diode.dx.value,\n', '        d_stage.y, terms.USAXS.diode.dy.value,\n', '\n', '        user_data.time_stamp, str(datetime.datetime.now()),\n', '        user_data.state, "pre-USAXS optics tune",\n', '\n', '        # Is this covered by user_mode, "USAXS"?\n', '        usaxs_slit.v_size,  terms.SAXS.usaxs_v_size.value,\n', '        usaxs_slit.h_size,  terms.SAXS.usaxs_h_size.value,\n', '        guard_slit.v_size,  terms.SAXS.usaxs_guard_v_size.value,\n', '        guard_slit.h_size,  terms.SAXS.usaxs_guard_h_size.value,\n', '\n', '        scaler0.preset_time,  0.1,\n', '    )\n', '    # when all that is complete, then ...\n', '    yield from bps.mv(ti_filter_shutter, "open")\n', '\n', '    # TODO: install suspender using usaxs_CheckBeamStandard.value\n', '\n', '    tuners = OrderedDict()                 # list the axes to tune\n', '    tuners[m_stage.r] = tune_mr            # tune M stage to monochromator\n', '    tuners[m_stage.r2p] = tune_m2rp        # make M stage crystals parallel\n', '    if terms.USAXS.useMSstage.value:\n', '        tuners[ms_stage.rp] = tune_msrp    # align MSR stage with M stage\n', '    if terms.USAXS.useSBUSAXS.value:\n', '        tuners[as_stage.rp] = tune_asrp    # align ASR stage with MSR stage and set ASRP0 value\n', '    tuners[a_stage.r] = tune_ar            # tune A stage to M stage\n', '    tuners[a_stage.r2p] = tune_a2rp        # make A stage crystals parallel\n', '\n', '    # now, tune the desired axes, bail out if a tune fails\n', '    yield from bps.install_suspender(suspend_BeamInHutch)\n', '    for axis, tune in tuners.items():\n', '        yield from bps.mv(ti_filter_shutter, "open")\n', '        yield from tune(md=md)\n', '        if axis.tuner.tune_ok:\n', "            # If we don't wait, the next tune often fails\n", '            # intensity stays flat, statistically\n', '            # We need to wait a short bit to allow EPICS database\n', '            # to complete processing and report back to us.\n', '            yield from bps.sleep(1)\n', '        else:\n', '            print("!!! tune failed for axis {axis.name} !!!")\n', '            break\n', '    yield from bps.remove_suspender(suspend_BeamInHutch)\n', '\n', '    print("USAXS count time: {terms.USAXS.usaxs_time.value} second(s)")\n', '    yield from bps.mv(\n', '        scaler0.preset_time,        terms.USAXS.usaxs_time.value,\n', '        user_data.time_stamp,       str(datetime.datetime.now()),\n', '        user_data.state,            "pre-USAXS optics tuning done",\n', '\n', '        terms.preUSAXStune.num_scans_last_tune, 0,\n', '        terms.preUSAXStune.run_tune_next,       0,\n', '        terms.preUSAXStune.epoch_last_tune,     time.time(),\n', '    )\n', '\n', '\n', 'def Flyscan(pos_X, pos_Y, thickness, scan_title, md={}):\n', '    """\n', '    do one USAXS Fly Scan\n', '    """\n', '    bluesky_runengine_running = RE.state != "idle"\n', '\n', '    yield from IfRequestedStopBeforeNextScan()\n', '\n', '    yield from mode_USAXS()\n', '\n', '    yield from bps.mv(\n', '        usaxs_slit.v_size, terms.SAXS.usaxs_v_size.value,\n', '        usaxs_slit.h_size, terms.SAXS.usaxs_h_size.value,\n', '        guard_slit.v_size, terms.SAXS.usaxs_guard_v_size.value,\n', '        guard_slit.h_size, terms.SAXS.usaxs_guard_h_size.value,\n', '    )\n', '\n', '    if terms.preUSAXStune.needed:\n', '        # tune at previous sample position\n', "        # don't overexpose the new sample position\n", '        yield from tune_usaxs_optics(md=md)\n', '\n', '    yield from bps.mv(\n', '        s_stage.x, pos_X,\n', '        s_stage.y, pos_Y,\n', '    )\n', '\n', '    scan_title_clean = cleanupText(scan_title)\n', '\n', '    # SPEC-compatibility symbols\n', '    SCAN_N = RE.md["scan_id"]+1     # the next scan number (user-controllable)\n', '    # use our specwriter to get a pseudo-SPEC file name\n', '    DATAFILE = os.path.split(specwriter.spec_filename)[-1]\n', '\n', '    # directory is pwd + DATAFILE + "_usaxs"\n', '    flyscan_path = os.path.join(os.getcwd(), os.path.splitext(DATAFILE)[0] + "_usaxs")\n', '    if not os.path.exists(flyscan_path) and bluesky_runengine_running:\n', '        # must create this directory if not exists\n', '        os.mkdir(flyscan_path)\n', '    flyscan_file_name = "%s_%04d.h5" % (scan_title_clean, terms.FlyScan.order_number.value)\n', '\n', '    usaxs_flyscan.saveFlyData_HDF5_dir = flyscan_path\n', '    usaxs_flyscan.saveFlyData_HDF5_file = flyscan_file_name\n', '\n', '    ts = str(datetime.datetime.now())\n', '    yield from bps.mv(\n', '        user_data.sample_title, scan_title,\n', '        user_data.state, "starting USAXS Flyscan",\n', '        user_data.sample_thickness, thickness,\n', '        user_data.user_name, USERNAME,\n', '        user_data.spec_scan, str(SCAN_N),\n', '        # or terms.FlyScan.order_number.value\n', '  [... cut ...]
|        18 | 11:19:19.3 | 11.0227056 |      52562 |                                                                                                       
|        19 | 11:19:19.7 | 11.0228056 |      52521 |                                                                                                       
|        20 | 11:19:20.1 | 11.0229056 |      51674 |                                                                                                       
|        21 | 11:19:20.5 | 11.0230056 |      48747 |                                                                                                       
```

prjemian, 2019-06-19:
This was printed during `tune_mr()`

```
In [5]: metadata["plan_name"]
Out[5]: 'tune_mr'
```

prjemian, 2019-06-19:
This metadata contains a dict within a dict and that leads to too much data to print nicely.  While pyrestTable can handle nested tables, the `metadata["archive"]` contains a lot of content, too much to print here.  Only `tune_mr()` has this.  Pound it out for now.

commit 9c336dc8 (2019-06-19): fixes #257
commit c8bcd7db (2019-06-19): Merge pull request #260 from APS-USAXS/metadata-print-257

fixes #257
closed 2019-06-19 by prjemian