APS-USAXS/usaxs-bluesky-ended-2023 #265: BS hangs when mode_Radiography is used with door opened

jilavsky, 2019-06-19:
Did RE(mode_Radiography()) while doors are opened, so it vcannot open the shutters. 

BS hangs:

```
USAXS is in position                                                                                                                                       
mono_shutter [In progress. No progress bar available.]                                                                                                     
Exception in thread Thread-4407:                                                                                                                           
Traceback (most recent call last):
  File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/apstools/devices.py", line 413, in move_it
    self.open()
  File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/apstools/devices.py", line 762, in open
    timeout=timeout)
  File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/10-devices.py", line 171, in wait_for_state
    raise TimeoutError(msg)
TimeoutError: Timeout (10 s) waiting for shutter state to reach a value in [1, 'O', 'N']

```

jilavsky, 2019-06-19:
We'd like to make messages such as this (not just this PV timeout but others, too) much nicer, without the stack trace.

jilavsky, 2019-06-19:
Could the suspender also watch the station door and act if opened?

tacaswell, 2019-06-28:
There is a hook in IPython to control how tracebacks are printed which you can use to mostly suppress them, however it makes debugging way harder!

Yes, suspenders can watch any bool signal.

prjemian, 2019-06-28:
@tacaswell : This magic? https://ipython.readthedocs.io/en/stable/interactive/magics.html#magic-xmode  Thanks for the tip.  We'll try it.

prjemian, 2019-06-28:
Wonder if I can have my cake and eat it too?  Terse report to command shell, verbose report to logger file.

prjemian, 2019-06-28:
ref: https://stackoverflow.com/questions/3702675/how-to-print-the-full-traceback-without-halting-the-program?rq=1

tacaswell, 2019-07-02:
@prjemian Yes, and what ever hook that is setting for you behind the scenes.

jilavsky, 2019-11-26:
This works fine when using simulated shutters.

PR #304 (2019-11-26): open shutter only if allowed in mode_Radiography()
fixes #265
closed 2019-11-26 by jilavsky