APS-USAXS/usaxs-bluesky-ended-2023 #266: BS throws error when APS dumps the ring

jilavsky, 2019-06-19:
BS seems to be annoyed at beam dump. This is not suitable to be seen by users. 
```
Out[5]: ()

/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/ophyd/ophydobj.py:269: RuntimeWarning: coroutine 'Event.wait' was never awaited
  cb(*args, **kwargs)
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
In [6]:                                                                   
```

jilavsky, 2019-06-19:
This is not specific to a beam dump.  We've seen this at other times.  Likely caused by something we've done but the call and report is so low-level it is not a valuable diagnostic for us.

@tacaswell, @mrakitin, @danielballan : Might y'all have any insight what actions of ours have triggered this message?

danielballan, 2019-06-21:
Thanks for the report @jilavsky. This is helpful. What version of ophyd are you running? `ophyd.__version__` is a good way to check.

prjemian, 2019-06-21:
ophyd      1.3.3

```
(/share1/bluesky/conda_envs/bluesky) usaxs@usaxscontrol /share1/USAXS_data $ ipython --profile=bluesky
Python 3.7.2 (default, Dec 29 2018, 06:19:36)
Type 'copyright', 'credits' or 'license' for more information
IPython 7.2.0 -- An enhanced Interactive Python. Type '?' for help.

IPython profile: bluesky
/home/beams11/USAXS/.ipython/profile_bluesky/startup/00-0-checks.py
BlueSky version: 1.5.3
Ophyd version: 1.3.3
/home/beams11/USAXS/.ipython/profile_bluesky/startup/00-startup.py
Loading metadata history from /home/beams/USAXS/.config/bluesky/bluesky_history.db
/home/beams11/USAXS/.ipython/profile_bluesky/startup/01-databroker.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/02-pyepics.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/09-imports.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/10-devices.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/11-motors.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/19-constants.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/20-detectors.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/21-aps.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/21-filters_shutters.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/21-signals.py
Waiting 1s for mono shutter PV to connect
Installing suspender for low APS current.
mono shutter connected = True
Defining suspend_BeamInHutch.  Install/remove in scan plans as desired.
/home/beams11/USAXS/.ipython/profile_bluesky/startup/22-amplifiers.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/24-detector_data_file_names.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/24-overrides_hdf5.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/25-alta.py
Timeout connecting with Alta (9idalta:)
/home/beams11/USAXS/.ipython/profile_bluesky/startup/25-blackfly.py
Timeout connecting with PointGrey BlackFly (9idFLY1:)
/home/beams11/USAXS/.ipython/profile_bluesky/startup/25-pilatus.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/25-simdetector.py
Timeout connecting with SimDetector (9idcSIM1:)
/home/beams11/USAXS/.ipython/profile_bluesky/startup/28-protection_plc.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/29-axis_tuning.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/32-usaxs_flyscan.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/41-commands.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/43-procedures.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/50-plans.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/51-tune_guard_slits.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/59-metadata-support.py
/home/beams11/USAXS/.ipython/profile_bluesky/startup/60-metadata.py
RunEngine metadata dictionary:
======================== ===================================
key                      value
======================== ===================================
EPICS_CA_MAX_ARRAY_BYTES 1280000
EPICS_HOST_ARCH          linux-x86_64
beamline_id              APS USAXS 9-ID-C
login_id                 usaxs@usaxscontrol.xray.aps.anl.gov
pid                      89021
proposal_id              testing Bluesky installation
scan_id                  0
versions                 ========== ======
                         key        value
                         ========== ======
                         apstools   1.1.6
                         bluesky    1.5.3
                         databroker 0.13.0
                         epics      3.3.3
                         ophyd      1.3.3
                         ========== ======
======================== ===================================

/home/beams11/USAXS/.ipython/profile_bluesky/startup/80-callbacks.py

writing to SPEC file: 20190621-094503.dat
>>>>   Using default SPEC file name   <<<<
file will be created when bluesky ends its next scan
to change SPEC file, use command: newSpecFile('title')

In [1]:
```

prjemian, 2019-06-21:
Looking through the databroker, these scans had long times until the next scan started: 

* eb4b1b17 (SAXS)
* 4d2f8406 (FlyScan)
* 28e04e42 (tune_a2rp)
* 4393e366 (FlyScan)
* bdd6f0bc (FlyScan)

Perhaps other metadata, such as in baseline, could tell us more such as which APS fill number.  We set that up here: https://github.com/APS-USAXS/ipython-usaxs/blob/559aea84fc59c53d802f0016751ae2e5f9c55d3c/profile_bluesky/startup/21-aps.py#L9-L10

That is in `aps.fill_number`: https://github.com/BCDA-APS/apstools/blob/eeb2e4e8e06fd67bf7aa225ab193a4f4e341934a/apstools/devices.py#L224

We need to look for `aps_fill_number` in the baseline.

prjemian, 2019-06-21:
Perhaps this search with databroker:

```
for h in db(since="2019-06-19", until="2019-06-20"): 
    el = list(h.events(stream_name="baseline", fields=["aps_fill_number", ]))
    if len(el) == 0: continue
    ev = el[0]
    print(datetime.datetime.fromtimestamp(ev["time"]), ev["uid"][:8], ev["data"]["aps_fill_number"]) 
```

which produces this output:

```
...
2019-06-19 14:34:40.611965 e7c64b90 16.0
2019-06-19 14:34:21.078551 11ad89c1 16.0
2019-06-19 14:34:13.508029 fb7b119e 16.0
2019-06-19 14:33:55.803763 43e378a6 16.0
2019-06-19 13:59:17.517182 e1c5703a 15.0
2019-06-19 13:59:07.345766 d6ac6354 15.0
2019-06-19 13:58:36.120748 47a04a44 15.0
2019-06-19 13:58:16.696620 13970e8c 15.0
2019-06-19 13:58:08.899830 38c18de2 15.0
...
```

so it must be this data: `2019-06-19 13:59:07.345766 d6ac6354 ` when the beam dumped

prjemian, 2019-06-21:
further:

```
In [48]: h = db["eb4b1b17"]                                                                                                                                                                      

In [49]: h.stop                                                                                                                                                                                  
Out[49]: 
{'run_start': 'eb4b1b17-f923-4390-813c-85dd377ed7bd',
 'time': 1560970788.2068527,
 'uid': '58b880a0-d193-48ce-8dc8-d731036d9c29',
 'exit_status': 'success',
 'num_events': {'baseline': 2, 'primary': 1}}

In [50]: h.start["plan_name"]                                                                                                                                                                    
Out[50]: 'SAXS'
```

prjemian, 2019-06-21:
@danielballan : Confirming my [earlier report](https://github.com/APS-USAXS/ipython-usaxs/issues/266#issuecomment-504452484), I've mined that info from the databroker:

```
In [48]: h = db["eb4b1b17"]                                                                                                                                                                      

In [49]: h.stop                                                                                                                                                                                  
Out[49]: 
{'run_start': 'eb4b1b17-f923-4390-813c-85dd377ed7bd',
 'time': 1560970788.2068527,
 'uid': '58b880a0-d193-48ce-8dc8-d731036d9c29',
 'exit_status': 'success',
 'num_events': {'baseline': 2, 'primary': 1}}

In [50]: h.start["plan_name"]                                                                                                                                                                    
Out[50]: 'SAXS'

In [51]: h.start["versions"]                                                                                                                                                                     
Out[51]: 
{'bluesky': '1.5.3',
 'ophyd': '1.3.3',
 'databroker': '0.13.0',
 'apstools': '1.1.6',
 'epics': '3.3.3'}
```

danielballan, 2019-06-21:
Perfect, thanks. As an aside, did you see that your suggestion of recording version information automatically has finally [made its way into the upcoming bluesky release](https://blueskyproject.io/bluesky/api_changes.html#features)?

prjemian, 2019-06-21:
@danielballan Yes, am overjoyed.  Brought my implementation in line with what is coming.  The [subdictionary](https://github.com/APS-USAXS/ipython-usaxs/issues/266#issuecomment-504452484) implementation looks better anyway.

tacaswell, 2019-06-24:
This is a warning (in that prints something) not an error (as it does not take down the process).

Internally, when you do a possibly 'slow' process (such as `abs_set` and `trigger`) a `asyncio.Event.wait` co-routine is created and is what we use to act as a bridge between the hardware an the RE (basically in the RE we do `await ev.wait()` and in a callback on the epics PV we do `ev.set()` with a few extra layers of machinery).  This warning is coming out of core python (see https://docs.python.org/3/library/asyncio-dev.html#detect-never-awaited-coroutines).

I _think_ this is an artifact of a subtle bug we had in how we were using the events (we were caching the coroutine from `fut = ev.wait()` which leads to both this issue and things not resuming correctly if we pause while we are waiting for a `set` to complete) which I think is fixed in https://github.com/bluesky/bluesky/pull/1125.

prjemian, 2019-06-24:
@tacaswell Thanks!  We'll look for that in a future release.  Until then, we'll ignore the warning.  That PR was merged about 3 weeks ago while we're using a release from a week before that.  Assume it is available in tag [v1.6.0rc1](https://github.com/bluesky/bluesky/releases/tag/v1.6.0rc1)?

prjemian, 2019-06-24:
The latest development version of bluesky is [1.2.0.post51](https://anaconda.org/lightsource2-dev/bluesky).  We'll have to test that tag by installing from the github repository.

tacaswell, 2019-06-24:
`pip install --pre bluesky` (after `conda uninstall bluesky`) should also do the trick.