APS-USAXS/usaxs-bluesky-ended-2023 #270: RuntimeError: FAILED TO FIND CORRECT GAIN IN 9 AUTOSCALE ITERATIONS

jilavsky, 2019-06-20:
Tried to run data collection, this error came up:

```
final position: 8.8855708
tuning axis:  a_stage_r2p                                                                                                                                  
converged=[True, False, True, True, True, True]                                                                                                            
[E 08:10:27.107 run_engine:1261] Run aborted
    Traceback (most recent call last):
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/run_engine.py", line 1150, in _run
        msg = self._plan_stack[-1].send(resp)
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 1294, in __call__
        return (yield from plan)
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 1149, in baseline_wrapper
        return (yield from plan_mutator(plan, insert_baseline))
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 169, in plan_mutator
        raise ex
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 122, in plan_mutator
        msg = plan_stack[-1].send(ret)
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 790, in monitor_during_wrapper
        return (yield from plan2)
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 169, in plan_mutator
        raise ex
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 122, in plan_mutator
        msg = plan_stack[-1].send(ret)
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 169, in plan_mutator
        raise ex
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 122, in plan_mutator
        msg = plan_stack[-1].send(ret)
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 848, in fly_during_wrapper
        return (yield from plan2)
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 169, in plan_mutator
        raise ex
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 122, in plan_mutator
        msg = plan_stack[-1].send(ret)
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 169, in plan_mutator
        raise ex
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 122, in plan_mutator
        msg = plan_stack[-1].send(ret)
      File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/50-plans.py", line 378, in run_command_file
        yield from execute_command_list(filename, commands)
      File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/50-plans.py", line 425, in execute_command_list
        yield from before_command_list(md=md, commands=commands)
      File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/50-plans.py", line 106, in before_command_list
        yield from preUSAXStune(md=md)
      File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/50-plans.py", line 543, in preUSAXStune
        yield from tune(md=md)
      File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/29-axis_tuning.py", line 375, in tune_a2rp
        yield from _tune_base_(a_stage.r2p, md=md)
      File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/29-axis_tuning.py", line 135, in _tune_base_
        yield from autoscale_amplifiers([upd_controls, I0_controls, I00_controls])
      File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/22-amplifiers.py", line 502, in autoscale_amplifiers
        max_iterations=max_iterations)
      File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/22-amplifiers.py", line 478, in _scaler_autoscale_
        raise RuntimeError(msg)
    RuntimeError: FAILED TO FIND CORRECT GAIN IN 9 AUTOSCALE ITERATIONS
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-3-07ecb13f554f> in <module>
----> 1 RE(run_command_file('usaxs.mac'))

/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    751                     # it (unless it is a canceled error)
    752                     if exc is not None:
--> 753                         raise exc
    754 
    755             if self._interrupted:

/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/run_engine.py in _run(self)
   1260             self._reason = str(err)
   1261             self.log.exception("Run aborted")
[... 141 lines cut ...]

/home/beams11/USAXS/.ipython/profile_bluesky/startup/29-axis_tuning.py in _tune_base_(axis, md)
    133         ti_filter_shutter, "open",
    134     )
--> 135     yield from autoscale_amplifiers([upd_controls, I0_controls, I00_controls])
    136     yield from axis.tune(md=md)
    137     yield from bps.mv(

/home/beams11/USAXS/.ipython/profile_bluesky/startup/22-amplifiers.py in autoscale_amplifiers(controls, shutter, count_time, max_iterations)
    500                 control_list,
    501                 count_time=count_time,
--> 502                 max_iterations=max_iterations)
    503 
    504 

/home/beams11/USAXS/.ipython/profile_bluesky/startup/22-amplifiers.py in _scaler_autoscale_(controls, count_time, max_iterations)
    476         msg = f"FAILED TO FIND CORRECT GAIN IN {max_iterations} AUTOSCALE ITERATIONS"
    477         if RE.state != "idle":      # don't raise if in summarize_plan()
--> 478             raise RuntimeError(msg)
    479 
    480 

RuntimeError: FAILED TO FIND CORRECT GAIN IN 9 AUTOSCALE ITERATIONS

```

jilavsky, 2019-06-20:
This was in preUSAXstune, after ar tune and before a2rp tune. 
This is known issue which needs to be handles somehow - if amplifier is in overload state, it takes long time for it to recover from it. I believe spec simply gives up, but need to check the code.

prjemian, 2019-06-20:
How long?

jilavsky, 2019-06-20:
It happened again. I suspect this may need some sleep after opening the Ti shutter. May be the shutter is slow and so the code operates while shutter is opening. That would cause it to fail. 
In any case, if it fails it should still continue. Last thing we want is glitch to stop overnight runs. It may cause a bad data set, but if we have 12 hours of experiments, loss of single data set is not fatal - code must continue.
E.G.: I stepped out now and the thing stopped minute after I went for coffee. Lost time needlessly ;-)

jilavsky, 2019-06-20:
Add may be 50 ms after opening?

prjemian, 2019-06-20:
How long to wait?

prjemian, 2019-06-21:
Let's try a combination of adding a bit of sleep (~50 ms) after opening the Ti shutter *and* not failing if it can't autoscale.  The latter will involve adding a custom exception that we can trap.

prjemian, 2019-06-21:
I believe the correct place to add the wait is not after opening the shutter but just before running the scaler each iteration through the loop.  Just before the `trigger()` here before line 342: https://github.com/APS-USAXS/ipython-usaxs/blob/559aea84fc59c53d802f0016751ae2e5f9c55d3c/profile_bluesky/startup/22-amplifiers.py#L341-L343

prjemian, 2019-06-21:
change 19d0ce2b7e342425cb167834107df1de7bc7c9f7 belongs here

prjemian, 2019-06-21:
Could make the pre-trigger delay programmable rather than fixed.

PR #273 (2019-06-25): fix autoscale failure
fixes #270
closed 2019-06-25 by prjemian