APS-USAXS/usaxs-bluesky-ended-2023 #271: Instrument hanged no obviosu reason

jilavsky, 2019-06-20:
End of flyscan, this is state of command line for about 2 minutes now:
```
55.08        8.6940511    -0.73094     9.81212      2761         49.85      
60.09        8.6157983    -1.03008     8.57445      3339         54.87      
65.10        8.5203318    -1.38956     7.05361      3928         59.87      
70.11        8.3999173    -1.84546     5.18682      4543         64.88      
75.12        8.2583856    -2.38111     2.96575      5168         69.90      
80.13        8.0851718    -3.03404     0.22322      5806         74.92      
85.13        7.8866914    -3.79200     -2.85676     6449         79.90      
90.14        7.6493788    -4.68339     -6.59899     7103         84.92      
95.15        7.3752949    -5.69415     -10.83771    7762         89.92      
writing fly scan HDF5 file: /share1/USAXS_data/2019-06/06_20_KoLan_usaxs/K_30_15_55_14d_1_0064.h5
99.01        7.2277710    -6.22300     -12.95576    7999         93.75      
```
Not obvious what to do.
Flyscan tool is Done, flyscan motors are at the end of travel. 

Will cancel and run samples again in spec. I have also two failed WAXS data sets (images with wrong intensity).

jilavsky, 2019-06-20:
This may help?
```
In [19]: exit                                                                                                                                              
set_and_wait('user_data_state', writing fly scan HDF5 file) failed
Traceback (most recent call last):
  File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/ophyd/signal.py", line 161, in set_thread
    rtol=self.rtolerance)
  File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/ophyd/utils/epics_pvs.py", line 261, in set_and_wait
    current_value = signal.get()
  File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/ophyd/signal.py", line 556, in get
    ret = self._read_pv.get(as_string=as_string, **kwargs)
  File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/epics/pv.py", line 325, in get
    use_monitor=use_monitor)
  File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/epics/pv.py", line 398, in get_with_metadata
    if ca.get_cache(self.pvname)['value'] is not None:
  File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/epics/ca.py", line 310, in get_cache
    return _cache[current_context()].get(pvname, None)
KeyError: 93938412730880
(/share1/bluesky/conda_envs/bluesky) usaxs@usaxscontrol .../USAXS_data/2019-06 $ 

```

prjemian, 2019-06-21:
The KeyError is trying hard to tell us something we cannot understand.  During a `set_and_wait()` for an EPICS PV, the process failed with the KeyError.  The report does not indicate which PV.

Will continue to add deeper diagnostics to identify what is the real problem here.

prjemian, 2019-06-21:
This bit `set_and_wait('user_data_state', writing fly scan HDF5 file) failed` is informative.  The faliure is on this line: https://github.com/APS-USAXS/ipython-usaxs/blob/559aea84fc59c53d802f0016751ae2e5f9c55d3c/profile_bluesky/startup/32-usaxs_flyscan.py#L177

closed 2019-06-25 by prjemian

tacaswell, 2019-06-28:
This also smells like something that might be fixed by pyepics 3.4.0.

PR #272 (2019-06-25): do not fail the scan just because of updating program state
fixes #271

**Do not merge until tested**