APS-USAXS/usaxs-bluesky-ended-2023 #276: Error messages from BS while running Flyscans

jilavsky, 2019-06-25:
This comes out randomly, does not see to be fatal:
```
110.13       7.9087411    -3.47045     -2.10814     6378         104.88     
/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/epics/ca.py:1378: UserWarning: ca.get('9idcLAX:3820:ElapsedReal') timed out after 1.00 seconds.
  warnings.warn(msg % (name(chid), timeout))
115.14       7.7486192    -4.02960     -4.58843   
/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/epics/ca.py:1378: UserWarning: ca.get('9idcLAX:3820:CurrentChannel') timed out after 1.00 seconds.
  warnings.warn(msg % (name(chid), timeout))
120.15       7.5827820    -4.62318     -7.15742     109.90     

```

jilavsky, 2019-06-25:
Not errors.  `UserWarning` that comes from PyEpics (`python3.7/site-packages/epics/ca.py:1378`).

Might be related to a comment from @tacaswell [yesterday](https://github.com/APS-USAXS/ipython-usaxs/issues/266#issuecomment-505044264) .  But, since this is from PyEpics, that is not certain.  We'll try using [bluesky v1.6.0rc1](https://github.com/bluesky/bluesky/releases/tag/v1.6.0rc1) and see if this relaxes these warnings.

jilavsky, 2019-06-25:
What is happening is that we are getting timeouts while reading some PVs from the Struck MCA.  These timeouts are also evident during the scan since they result in `None` values.  The console output looks like this:

```
flying, s    ar, deg      ay, mm       dy, mm       channel      elapsed, s 
5.01         8.8888013    0.00999      13.10342     0            0.00       
10.01        8.8872282    0.00416      13.07734     55           4.77       
15.01        8.8860581    0.00021      13.06080     98           9.73       
20.02        8.8854952    -0.00173     13.05197     119          14.77      
25.02        8.8840115    -0.00720     13.02782     174          19.77      
30.02        8.8808224    -0.01877     12.97771     277          24.78      
35.03        8.8752529    -0.03864     12.88988     434          29.78      
40.04        8.8668494    -0.06899     12.75818     639          34.78      
45.04        8.8543194    -0.11382     12.56233     900          39.80      
50.05        8.8370340    -0.17456     12.29687     1194         44.80      
55.06        8.8147853    -0.25556     11.94595     1527         49.82      
60.07        8.7833627    -0.36676     11.45653     1928         54.82      
65.07        8.7491553    -0.48603     10.92458   
70.07        8.7111963    -0.62686     10.33204     59.82      
75.07        8.6710712    -0.76409     9.71145    
80.08        8.6310114    -0.90456     9.08940      2679         69.83      
85.09        8.5602785    -1.16561     7.98808    
90.09        8.4880206    -1.41188     6.88798      79.83      
95.09        8.3887965    -1.76792     5.33723    
100.09       8.3151073    -2.02925     4.18928      4443         89.85      
105.10       8.1657763    -2.55902     1.87472    
110.11       8.0138411    -3.09725     -0.47236     99.87      
115.11       7.8772738    -3.58338     -2.60424   
120.27       7.7316906    -4.09716     -4.79332     6353         110.07     
125.27       7.4908947    -4.95304     -8.60098   
writing fly scan HDF5 file: /share1/USAXS_data/2019-06/06_25_GegSmith1_usaxs/Sm_7476_3_0016.h5
129.38       7.2318500    -5.83331     -12.43934    7999         120.07     
finished writing fly scan HDF5 file: /share1/USAXS_data/2019-06/06_25_GegSmith1_usaxs/Sm_7476_3_0016.h5
```

The *shorter* lines during the scan (such as at 115.11 seconds) are due to this code: https://github.com/APS-USAXS/ipython-usaxs/blob/559aea84fc59c53d802f0016751ae2e5f9c55d3c/profile_bluesky/startup/32-usaxs_flyscan.py#L61-L64

We can add text here to show that these numbers are missing instead of leaving the text empty.  @jilavsky: Your choice on this.

jilavsky, 2019-06-25:
@tacaswell:  No point in us trying bluesky v1.6.0rc1  since that only has changes in a file for sphinx and we are using the previous tag 1.5.3.  Will wait for the changes to appear in a tag.  We have some adverse experience when we try to run from the master branch.

jilavsky, 2019-06-25:
Observe that these messages tend to appear and appear more the longer we operate.  From a fresh start, none or not so many.

prjemian, 2019-06-25:
Suggest this text for *missing* values: `-missing-`

jilavsky, 2019-06-25:
Got another one with problems:

```
Transient Scan ID: 130     Time: 2019-06-25 14:32:40                                                                                                       
Persistent Unique Scan ID: '64b30643-32f2-492e-86a9-45e726501aa7'                                                                                          
New stream: 'baseline'                                                                                                                                     
New stream: 'aps_current_monitor'
HDF5 config: /share1/AreaDetectorConfig/FlyScan_config/saveFlyData.xml
HDF5 output: /share1/USAXS_data/2019-06/06_25_Byorn2_usaxs/S2A_1_Top_v_0017.h5
flying, s    ar, deg      ay, mm       dy, mm       channel      elapsed, s 
5.00         8.8887172    0.01016      13.10415     0            0.00       
10.01        8.8867283    0.00279      13.07217     69           4.77       
15.02        8.8857074    -0.00050     13.05753     108          9.78       
20.03        8.8841195    -0.00657     13.03109     167          14.78      
25.03        8.8798206    -0.02225     12.96259     305          19.78      
30.03        8.8710068    -0.05418     12.82248     541          24.78      
35.03        8.8563321    -0.10776     12.59153     859          29.78      
40.05        8.8331571    -0.19054     12.21883     1261         34.83      
45.06        8.8003381    -0.31049     11.71320     1717         39.82      
50.07        8.7553454    -0.47157     10.99317     2219         44.82      
55.07        8.6909673    -0.69400     10.01984     2785         49.83      
60.08        8.6056593    -1.00228     8.67125      3415         54.83      
65.08        8.5128622    -1.34144     7.22568      3970         59.83      
70.09        8.4039514    -1.73062     5.52303      -missing-    -missing-  
75.10        8.2891478    -2.12603     3.76814      -missing-    64.85      
80.10        8.1605604    -2.58491     1.76411      -missing-    -missing-  
85.11        8.0491745    -2.98056     0.03503      5164         74.85      
90.12        7.7965957    -3.88051     -3.81664     -missing-    -missing-  
95.13        7.5828150    -4.63712     -7.20497     -missing-    84.95      
100.14       7.3114944    -5.61040     -11.47511    -missing-    -missing-  
writing fly scan HDF5 file: /share1/USAXS_data/2019-06/06_25_Byorn2_usaxs/S2A_1_Top_v_0017.h5
102.37       7.2318137    -5.84673     -12.47481    7784         93.65      
```

And, the function that writes the fly scan's HDF5 file has not returned.  Probably related to #271.  Still do not see root cause.

jilavsky, 2019-06-25:
Again, this did not happen right away.  After ten or so flyscans, then we start to see this problem.

jilavsky, 2019-06-25:
Stack trace

<details>
<summary>As if this will help:</summary>
<p>

```
90.12        7.7965957    -3.88051     -3.81664     -missing-    -missing-  
95.13        7.5828150    -4.63712     -7.20497     -missing-    84.95      
100.14       7.3114944    -5.61040     -11.47511    -missing-    -missing-  
writing fly scan HDF5 file: /share1/USAXS_data/2019-06/06_25_Byorn2_usaxs/S2A_1_Top_v_0017.h5
102.37       7.2318137    -5.84673     -12.47481    7784         93.65      
^CA 'deferred pause' has been requested. The RunEngine will pause at the next checkpoint. To pause immediately, hit Ctrl+C again in the next 10 seconds.
Deferred pause acknowledged. Continuing to checkpoint.
^CPausing...
---------------------------------------------------------------------------
RunEngineInterrupted                      Traceback (most recent call last)
<ipython-input-2-07ecb13f554f> in <module>
----> 1 RE(run_command_file('usaxs.mac'))

/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    754 
    755             if self._interrupted:
--> 756                 raise RunEngineInterrupted(self.pause_msg) from None
    757 
    758         return tuple(self._run_start_uids)

RunEngineInterrupted: 
Your RunEngine is entering a paused state. These are your options for changing
the state of the RunEngine:

RE.resume()    Resume the plan.
RE.abort()     Perform cleanup, then kill plan. Mark exit_stats='aborted'.
RE.stop()      Perform cleanup, then kill plan. Mark exit_status='success'.
RE.halt()      Emergency Stop: Do not perform cleanup --- just stop.

In [3]: RE.abort()                                                                                                                                         
Aborting: running cleanup and marking exit_status as 'abort'...
Error: writing fly scan HDF5 file: /share1/USAXS_data/2019-06/06_25_Byorn2_usaxs/S2A_1_Top_v_0017.h5 - 
finished writing fly scan HDF5 file: /share1/USAXS_data/2019-06/06_25_Byorn2_usaxs/S2A_1_Top_v_0017.h5
HDF5 output complete: /share1/USAXS_data/2019-06/06_25_Byorn2_usaxs/S2A_1_Top_v_0017.h5
New stream: 'mca'                                                                                                                                          
[E 14:41:38.855 run_engine:1261] Run aborted
    Traceback (most recent call last):
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/run_engine.py", line 1127, in _run
        self._exception or resp)
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 1294, in __call__
        return (yield from plan)
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 1149, in baseline_wrapper
        return (yield from plan_mutator(plan, insert_baseline))
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 77, in plan_mutator
        msg = plan_stack[-1].throw(exception)
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 790, in monitor_during_wrapper
        return (yield from plan2)
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 77, in plan_mutator
        msg = plan_stack[-1].throw(exception)
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 77, in plan_mutator
        msg = plan_stack[-1].throw(exception)
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 848, in fly_during_wrapper
        return (yield from plan2)
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 77, in plan_mutator
        msg = plan_stack[-1].throw(exception)
      File "/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 77, in plan_mutator
        msg = plan_stack[-1].throw(exception)
      File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/50-plans.py", line 378, in run_command_file
        yield from execute_command_list(filename, commands)
      File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/50-plans.py", line 450, in execute_command_list
        yield from Flyscan(sx, sy, sth, snm, md=_md)
      File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/50-plans.py", line 728, in Flyscan
        yield from usaxs_flyscan.plan(md=_md)        # DO THE FLY SCAN
[... 218 lines cut ...]
-> 1955         ret = msg.obj.set(*msg.args, **kwargs)
   1956         p_event = asyncio.Event(loop=self.loop)
   1957         pardon_failures = self._pardon_failures

/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/ophyd/signal.py in set(self, value, timeout, settle_time)
    939         '''
    940         if not self._put_complete:
--> 941             return super().set(value, timeout=timeout, settle_time=settle_time)
    942 
    943         # using put completion:

/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/ophyd/signal.py in set(self, value, timeout, settle_time)
    183 
    184         if self._set_thread is not None:
--> 185             raise RuntimeError('Another set() call is still in progress')
    186 
    187         st = Status(self)

RuntimeError: Another set() call is still in progress

In [4]:     
```

</p>
</details>

prjemian, 2019-06-25:
In this code: https://github.com/APS-USAXS/ipython-usaxs/blob/e3d597d20b22cc43ef101407ac3a75ccde3d9b9f/profile_bluesky/startup/50-plans.py#L736-L743

we used `.get()` because `.value` was not working.  Probably another indication of this same problem.

jilavsky, 2019-06-25:
changeset e7d51758cb2546f07cab901ba9bf7236aaa530ae is related to this issue

jilavsky, 2019-06-25:
Wonder if `Another set() call is still in progress` is related to `/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/epics/ca.py:1378: UserWarning: ca.get('9idcLAX:3820:ElapsedReal') timed out after 1.00 seconds.
  warnings.warn(msg % (name(chid), timeout))`

Need some diagnostics here to further examine...

jilavsky, 2019-06-25:
When these -missing- messages are reported top shows, that iPython is taking 100+ (103-106) % cpu load. This happens towards the end of Flyscan when struck is collecting high number of points/second.  
Generally, cpu load increases as we collect more and more Flyscans. Need to do more quantification on this tomorrow. 
After ~12 Flyscans collected we reach 100% cpu load around second 55-60 into the Flyscan. 
After 16 Flyscans we reach 100% cpu around 45seconds into the FS.

prjemian, 2019-06-26:
Let's add to the flyscan progress report  about CPU load and possibly heap size.  I'm suspicious that the saveFlyData code (same as we have used with SPEC) may be involved here.  Ultimately, that code needs to be refactored into bluesky.  See #19 and also #113.

prjemian, 2019-06-26:
memory profile:

* https://www.quora.com/How-can-one-print-memory-usage-in-Python-program
* https://stackoverflow.com/questions/552744/how-do-i-profile-memory-usage-in-python
* https://docs.python.org/3/library/tracemalloc.html
  https://stackoverflow.com/a/45679009/1046449

SO showed a truly simple approach:

```
import resource
def using(point=""):
    usage=resource.getrusage(resource.RUSAGE_SELF)
    return '''%s: usertime=%s systime=%s mem=%s usage,kB
           '''%(point,usage[0],usage[1], usage[2])```

objects:

* https://mg.pov.lt/objgraph/

prjemian, 2019-06-26:
https://stackoverflow.com/questions/276052/how-to-get-current-cpu-and-ram-usage-in-python

prjemian, 2019-06-26:
Also, the saveFlyData.saveFile method uses logger.debug, need to turn on the logger.  That is in 00-startup.py and is turned off by default.  Need to start routine logging to a file (see #109).

jilavsky, 2019-06-26:
This is the first Flyscan after fresh start:

```
Transient Scan ID: 20     Time: 2019-06-26 13:02:52                                                                                                        
Persistent Unique Scan ID: '07059d30-d34c-4c66-bc57-496ce729c02a'                                                                                          
New stream: 'baseline'                                                                                                                                     
New stream: 'aps_current_monitor'
HDF5 config: /share1/AreaDetectorConfig/FlyScan_config/saveFlyData.xml
HDF5 output: /share1/USAXS_data/2019-06/06_26_NXSchool0626_usaxs/AirBlank_0287.h5
before SaveFlyScan(): user:72.566s sys:10.916s mem:280.17MB cpu:18.7%
before saveFlyData.preliminaryWriteFile(): user:72.685s sys:10.953s mem:283.58MB cpu:43.8%
flying, s    ar, deg      ay, mm       dy, mm       channel      elapsed, s 
5.00         8.8885352    0.01016      13.10415     0            0.00         Flyscan: user:74.507s sys:11.291s mem:286.80MB cpu:25.5%
after saveFlyData.preliminaryWriteFile(): user:75.051s sys:11.393s mem:287.60MB cpu:36.5%
10.01        8.8885352    0.01016      13.10415     0            0.00         Flyscan: user:75.107s sys:11.403s mem:287.76MB cpu:22.6%
15.02        8.8885352    0.01016      13.10415     0            0.00         Flyscan: user:75.195s sys:11.419s mem:287.76MB cpu:17.6%
20.02        8.8885352    0.01016      13.10415     0            0.00         Flyscan: user:75.280s sys:11.438s mem:287.76MB cpu:17.3%
25.02        8.8885352    0.01016      13.10415     0            0.00         Flyscan: user:75.362s sys:11.460s mem:287.76MB cpu:18.1%
30.02        8.8885352    0.01016      13.10415     0            0.00         Flyscan: user:75.451s sys:11.474s mem:287.76MB cpu:19.8%
35.03        8.8885352    0.01016      13.10415     0            0.00         Flyscan: user:75.530s sys:11.498s mem:287.76MB cpu:21.3%
40.03        8.8885352    0.01016      13.10415     0            0.00         Flyscan: user:75.613s sys:11.517s mem:287.76MB cpu:21.8%
45.04        8.8885352    0.01016      13.10415     0            0.00         Flyscan: user:75.693s sys:11.539s mem:287.76MB cpu:21.0%
50.04        8.8885352    0.01016      13.10415     0            0.00         Flyscan: user:75.781s sys:11.557s mem:287.76MB cpu:20.0%
55.05        8.8885352    0.01016      13.10415     0            0.63         Flyscan: user:75.881s sys:11.581s mem:287.76MB cpu:18.7%
60.06        8.8862255    0.00185      13.06745     59           5.65         Flyscan: user:76.141s sys:11.608s mem:287.76MB cpu:20.2%
65.07        8.8853789    -0.00107     13.05516     90           10.65        Flyscan: user:76.465s sys:11.653s mem:287.76MB cpu:20.5%
70.08        8.8834542    -0.00827     13.02323     164          15.67        Flyscan: user:76.858s sys:11.700s mem:288.29MB cpu:21.4%
75.08        8.8784512    -0.02673     12.94428     323          20.67        Flyscan: user:77.343s sys:11.748s mem:288.43MB cpu:21.7%
80.09        8.8687858    -0.06153     12.79064     576          25.67        Flyscan: user:77.884s sys:11.798s mem:288.43MB cpu:23.1%
85.09        8.8526051    -0.12018     12.53402     915          30.68        Flyscan: user:78.509s sys:11.853s mem:288.70MB cpu:23.2%
90.10        8.8276484    -0.20888     12.15383     1326         35.68        Flyscan: user:79.214s sys:11.908s mem:288.98MB cpu:24.5%
95.11        8.7924306    -0.33422     11.60225     1796         40.70        Flyscan: user:79.956s sys:11.968s mem:289.53MB cpu:25.1%
100.11       8.7452101    -0.50396     10.85010     2310         45.70        Flyscan: user:80.736s sys:12.021s mem:289.54MB cpu:24.7%
105.12       8.6814853    -0.72958     9.87321      2858         50.70        Flyscan: user:81.543s sys:12.088s mem:289.83MB cpu:23.5%
110.12       8.6025204    -1.01584     8.62338      3434         55.70        Flyscan: user:82.420s sys:12.142s mem:289.83MB cpu:22.6%
115.12       8.5013508    -1.37390     7.06958      4031         60.70        Flyscan: user:83.367s sys:12.210s mem:290.39MB cpu:21.3%
120.12       8.3799430    -1.80280     5.17041      4646         65.70        Flyscan: user:84.308s sys:12.277s mem:290.46MB cpu:21.9%
125.13       8.2307548    -2.33486     2.86094      5273         70.72        Flyscan: user:85.314s sys:12.345s mem:290.76MB cpu:21.9%
130.13       8.0549499    -2.95697     0.13222      5911         75.72        Flyscan: user:86.325s sys:12.408s mem:291.30MB cpu:27.5%
135.14       7.8471582    -3.69504     -3.09442     6557         80.72        Flyscan: user:87.362s sys:12.451s mem:291.34MB cpu:47.4%
140.14       7.6103228    -4.53490     -6.76842     7211         85.72        Flyscan: user:88.475s sys:12.496s mem:291.89MB cpu:25.7%
145.14       7.3302235    -5.53191     -11.09703    7870         90.72        Flyscan: user:89.593s sys:12.561s mem:291.92MB cpu:24.9%
writing fly scan HDF5 file: /share1/USAXS_data/2019-06/06_26_NXSchool0626_usaxs/AirBlank_0287.h5
before saveFlyData.finish_HDF5_file(): user:90.078s sys:12.592s mem:291.92MB cpu:22.7%
after saveFlyData.finish_HDF5_file(): user:90.078s sys:12.593s mem:291.92MB cpu:0.0%
finished writing fly scan HDF5 file: /share1/USAXS_data/2019-06/06_26_NXSchool0626_usaxs/AirBlank_0287.h5
148.28       7.2277350    -5.84673     -12.49798    7999         93.82        Flyscan: user:90.083s sys:12.594s mem:291.92MB cpu:33.3%
[... cut ...]                                                                               
HDF5 output complete: /share1/USAXS_data/2019-06/06_26_NXSchool0626_usaxs/AirBlank_0287.h5
/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/bluesky/run_engine.py:482: RuntimeWarning: coroutine 'Event.wait' was never awaited
  self._groups.clear()
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
exit status: success
# descriptor(s): 3
# event(s): 299
Out[8]:                   
```

jilavsky, 2019-06-26:
This is later scan:

```
Transient Scan ID: 52     Time: 2019-06-26 13:26:34                                                                                                        
Persistent Unique Scan ID: '7e3e869f-f449-4910-91fb-baef004c8456'                                                                                          
New stream: 'baseline'                                                                                                                                     
New stream: 'aps_current_monitor'
HDF5 config: /share1/AreaDetectorConfig/FlyScan_config/saveFlyData.xml
HDF5 output: /share1/USAXS_data/2019-06/06_26_NXSchool0626_usaxs/AirBlank_0297.h5
before SaveFlyScan(): user:572.589s sys:45.916s mem:358.37MB cpu:28.4%
before saveFlyData.preliminaryWriteFile(): user:572.698s sys:45.948s mem:360.13MB cpu:59.2%
flying, s    ar, deg      ay, mm       dy, mm       channel      elapsed, s 
after saveFlyData.preliminaryWriteFile(): user:573.421s sys:46.049s mem:362.81MB cpu:45.4%
5.01         8.8884555    0.00972      13.10280     0            0.00         Flyscan: user:573.494s sys:46.066s mem:363.01MB cpu:20.0%
10.02        8.8865344    0.00279      13.07193     48           4.80         Flyscan: user:574.355s sys:46.180s mem:363.06MB cpu:24.8%
15.02        8.8855252    -0.00056     13.05725     85           9.82         Flyscan: user:575.919s sys:46.331s mem:363.06MB cpu:26.4%
20.03        8.8839383    -0.00661     13.03074     146          14.82        Flyscan: user:577.914s sys:46.484s mem:363.06MB cpu:27.6%
25.05        8.8795167    -0.02271     12.96008     290          19.83        Flyscan: user:580.282s sys:46.627s mem:363.32MB cpu:28.1%
30.05        8.8707291    -0.05419     12.82346     528          24.83        Flyscan: user:583.003s sys:46.823s mem:363.32MB cpu:28.4%
35.06        8.8558557    -0.10842     12.58873     853          29.85        Flyscan: user:586.036s sys:47.015s mem:363.32MB cpu:27.8%
40.07        8.8328404    -0.18991     12.22691     1253         34.85        Flyscan: user:589.326s sys:47.202s mem:363.42MB cpu:28.0%
45.08        8.7995733    -0.31103     11.70444     1714         39.87        Flyscan: user:592.820s sys:47.395s mem:363.42MB cpu:27.2%
50.08        8.7542134    -0.47474     10.98484     2221         44.87        Flyscan: user:596.512s sys:47.581s mem:363.42MB cpu:28.2%
55.09        8.6939111    -0.68919     10.06478     2764         49.87        Flyscan: user:600.409s sys:47.757s mem:363.42MB cpu:28.5%
60.09        8.6164083    -0.96589     8.84617      3338         54.88        Flyscan: user:604.741s sys:47.940s mem:363.43MB cpu:29.9%
65.10        8.5172235    -1.31214     7.30757      3939         59.90        Flyscan: user:609.173s sys:48.101s mem:363.43MB cpu:30.2%
70.11        8.4003831    -1.73406     5.47316      4548         64.90        Flyscan: user:613.454s sys:48.259s mem:363.43MB cpu:31.5%
75.11        8.2538613    -2.25042     3.27180      5171         69.90        Flyscan: user:618.093s sys:48.435s mem:363.44MB cpu:31.9%
80.11        8.0835071    -2.84650     0.62490      5811         74.90        Flyscan: user:622.775s sys:48.612s mem:363.44MB cpu:32.5%
85.11        7.8804358    -3.57833     -2.58081     6465         79.90        Flyscan: user:627.871s sys:48.803s mem:363.44MB cpu:31.9%
90.12        7.6504537    -4.38236     -6.10946     7107         84.90        Flyscan: user:632.550s sys:48.986s mem:363.44MB cpu:32.5%
95.12        7.3452096    -5.47056     -10.88488    7843         89.90        Flyscan: user:638.693s sys:49.212s mem:363.44MB cpu:30.5%
writing fly scan HDF5 file: /share1/USAXS_data/2019-06/06_26_NXSchool0626_usaxs/AirBlank_0297.h5
99.04        7.2297674    -5.84673     -12.49798    7999         93.75        Flyscan: user:641.050s sys:49.288s mem:363.70MB cpu:28.5%
before saveFlyData.finish_HDF5_file(): user:641.134s sys:49.290s mem:363.77MB cpu:44.6%
after saveFlyData.finish_HDF5_file(): user:641.175s sys:49.291s mem:363.77MB cpu:27.8%
finished writing fly scan HDF5 file: /share1/USAXS_data/2019-06/06_26_NXSchool0626_usaxs/AirBlank_0297.h5
HDF5 output complete: /share1/USAXS_data/2019-06/06_26_NXSchool0626_usaxs/AirBlank_0297.h5
New stream: 'mca'                                                                                                                                          
exit status: success
# descriptor(s): 3
# event(s): 203

```

prjemian, 2019-06-26:
@jilavsky : Thanks.  Turns out the [console logging](https://github.com/APS-USAXS/ipython-usaxs/issues/284#issuecomment-505980868) only captures part of the output.  Improvement will come eventually.

prjemian, 2019-06-26:
Interesting to see if that `mem` number continues to increase.  Could be a memory leak then.

jilavsky, 2019-06-26:
After about 20 Flyscans - seems better...
```
Transient Scan ID: 150     Time: 2019-06-26 15:46:28                                                                                                       
Persistent Unique Scan ID: '1ce38025-bafc-40f2-9802-7c90940a0ae8'                                                                                          
New stream: 'baseline'                                                                                                                                     
New stream: 'aps_current_monitor'
HDF5 config: /share1/AreaDetectorConfig/FlyScan_config/saveFlyData.xml
HDF5 output: /share1/USAXS_data/2019-06/06_26_NXSchool0626_usaxs/White_before_0311.h5
before SaveFlyScan(): user:2049.012s sys:167.297s mem:503.04MB cpu:27.7%
before saveFlyData.preliminaryWriteFile(): user:2049.128s sys:167.329s mem:503.04MB cpu:50.4%
flying, s    ar, deg      ay, mm       dy, mm       channel      elapsed, s 
after saveFlyData.preliminaryWriteFile(): user:2050.008s sys:167.426s mem:503.04MB cpu:31.7%
5.00         8.8884902    0.00972      13.10280     0            0.00         Flyscan: user:2050.104s sys:167.443s mem:503.04MB cpu:16.2%
10.01        8.8866103    0.00291      13.07246     48           4.78         Flyscan: user:2051.738s sys:167.640s mem:503.04MB cpu:23.1%
15.02        8.8855670    -0.00053     13.05743     85           9.80         Flyscan: user:2054.445s sys:167.861s mem:503.04MB cpu:24.7%
20.02        8.8839624    -0.00661     13.03103     147          14.80        Flyscan: user:2057.990s sys:168.112s mem:503.04MB cpu:28.7%
25.02        8.8796486    -0.02232     12.96254     289          19.80        Flyscan: user:2062.076s sys:168.388s mem:503.04MB cpu:28.6%
30.03        8.8705081    -0.05573     12.81813     534          24.80        Flyscan: user:2066.866s sys:168.674s mem:503.04MB cpu:28.3%
35.04        8.8548179    -0.11118     12.57096     870          29.82        Flyscan: user:2072.357s sys:168.996s mem:503.04MB cpu:28.0%
40.04        8.8291908    -0.20165     12.16938     -missing-    34.83        Flyscan: user:2078.441s sys:169.394s mem:503.04MB cpu:27.5%
45.05        8.8078661    -0.27740     11.84365     -missing-    -missing-    Flyscan: user:2082.500s sys:169.665s mem:503.04MB cpu:27.2%
50.05        8.7814980    -0.37157     11.42871     1329         39.83        Flyscan: user:2086.465s sys:169.857s mem:503.04MB cpu:27.3%
55.06        8.7257627    -0.57611     10.54233     -missing-    -missing-    Flyscan: user:2092.458s sys:170.176s mem:503.04MB cpu:27.3%
60.07        8.6614118    -0.80341     9.54831      -missing-    49.83        Flyscan: user:2097.458s sys:170.442s mem:503.04MB cpu:27.7%
65.08        8.5843384    -1.07706     8.35916      -missing-    -missing-    Flyscan: user:2102.409s sys:170.742s mem:503.04MB cpu:29.2%
70.09        8.4899087    -1.41280     6.88466      3346         59.92        Flyscan: user:2106.422s sys:170.933s mem:503.04MB cpu:29.6%
75.09        8.3338973    -1.96749     4.46683      -missing-    -missing-    Flyscan: user:2112.429s sys:171.287s mem:503.04MB cpu:30.4%
80.09        8.1807701    -2.52198     2.04603      -missing-    69.97        Flyscan: user:2117.403s sys:171.588s mem:503.04MB cpu:30.5%
85.10        7.9966228    -3.16593     -0.81735     -missing-    -missing-    Flyscan: user:2122.396s sys:171.867s mem:503.04MB cpu:30.3%
90.11        7.8151125    -3.81033     -3.62889     5814         79.88        Flyscan: user:2126.381s sys:172.087s mem:503.04MB cpu:28.5%
95.12        7.5110204    -4.89026     -8.31027     -missing-    -missing-    Flyscan: user:2132.407s sys:172.386s mem:503.04MB cpu:26.5%
writing fly scan HDF5 file: /share1/USAXS_data/2019-06/06_26_NXSchool0626_usaxs/White_before_0311.h5
100.12       7.2330312    -5.84673     -12.47663    7999         89.95        Flyscan: user:2137.063s sys:172.617s mem:503.04MB cpu:25.6%
100.80       7.2304481    -5.84673     -12.47663    7999         93.63        Flyscan: user:2137.100s sys:172.623s mem:503.04MB cpu:50.0%
before saveFlyData.finish_HDF5_file(): user:2137.203s sys:172.627s mem:503.04MB cpu:40.3%
after saveFlyData.finish_HDF5_file(): user:2137.214s sys:172.627s mem:503.04MB cpu:38.5%
finished writing fly scan HDF5 file: /share1/USAXS_data/2019-06/06_26_NXSchool0626_usaxs/White_before_0311.h5
HDF5 output complete: /share1/USAXS_data/2019-06/06_26_NXSchool0626_usaxs/White_before_0311.h5                                                             
New stream: 'mca'
exit status: success
# descriptor(s): 3
# event(s): 207

```

jilavsky, 2019-06-26:
Will try again tomorrow during NX school (after 1pm). Will also try in the morning (after 9am). I will restart before each session to make sure any fixes/changes are being used.

jilavsky, 2019-06-27:
Note: values of cpu load reported by Top for the pid process iPython and values reported by added prints in the console do not agree by vast amount. Diagnostics reports 26-30% cpu while top shows iPython sing over 100% cpu at the same time.

prjemian, 2019-06-27:
see:

* https://stackoverflow.com/questions/8491353/uptime-vs-top-cpu-usage-what-should-i-believe-why-this-difference
* https://duckduckgo.com/?q=uptime+cpu+load+vs.+top+cpu+load&t=ffsb&ia=web

prjemian, 2019-06-27:
Also:

* https://stackoverflow.com/questions/27267159/pythons-psutils-will-it-reflect-cpu-load-similar-to-top-monit
* https://duckduckgo.com/?q=psutils+cpu+load+vs.+top+cpu+load&t=ffsb&ia=web

prjemian, 2019-06-27:
A couple blogs were they investigated resource problems:

* https://benbernardblog.com/tracking-down-a-freaky-python-memory-leak/
* https://chase-seibert.github.io/blog/2013/08/03/diagnosing-memory-leaks-python.html

prjemian, 2019-06-28:
`conda install -c rmg guppy` : http://guppy-pe.sourceforge.net/
`pip install objgraph` : https://mg.pov.lt/objgraph/

tacaswell, 2019-06-28:
> Not errors. UserWarning that comes from PyEpics (python3.7/site-packages/epics/ca.py:1378).

These are warnings, but they probably should be fatal exceptions. Can you try updating to the latest pyepics (3.4.0)?  Ken cleaned up a number of weird race conditions / improved thread safety through out.

> No point in us trying bluesky v1.6.0rc1 since that only has changes in a file for sphinx and we are using the previous tag 1.5.3

I suspect that you were looking at just the last commit. The difference between 1.5.3 and 1.6.0rc1 is https://github.com/bluesky/bluesky/compare/v1.5.3...v1.6.0rc1

Might be worth also trying the ohpyd rc (1.4.0rc1) as well?

From the description and skimming the code, my guess is that someplace subscriptions are being made to PVs that are not being cleared.  

Digging back to https://github.com/APS-USAXS/ipython-usaxs/blob/c8bcd7db18b768cefdd50cd37993f126e8cc4870/profile_bluesky/startup/usaxs_support/saveFlyData.py#L209 it looks like you are using `pv.PV` directly which leads to weird resource leaks (because it used to cache it's self in `__init__`), I suggest turning all of those into `pv.get_pv` (or making that class an ophyd object ;) ).  By default pyepics auto monitors (to update it's internal cache) "small" PVs and you are creating new `saveFlyData` on each run so you may be leaking many busy PVs which will leak many `pv.PV` objects (that are kept alive via the monitor callback) that you can not get to.  If they are getting lots of updates that could be the source of your resource contention (the biggest problem is likely the callback threads fighting for the GIL).

I caveat the above with the note that I only quickly skimmed the code and that was the first thing that jumped out at me.

prjemian, 2019-06-28:
> between 1.5.3 and 1.6.0rc1 is bluesky/bluesky@v1.5.3...v1.6.0rc1

Tom

Thanks. Yes, the saveFlyData code is probably the weak link here.  It was
modified only slightly from the code we called from SPEC.  Changing it to
proper ophyd/bluesky is the next step.

prjemian, 2019-07-29:
There are a set PVs to be added to each fly scan as metadata.  The set is described in an XML file.  The file is parsed at the start of each fly scan, the PVs are connected while the scan is progressing, then the set is written at the end.  Likely, the biggest system load comes from the PV connection phase.

As an alternative, parse that XML file only once per batch of scans (which includes a fly scan).  Develop the PV list and use [`APS_utils.connect_pvlist()`](https://apstools.readthedocs.io/en/latest/source/_utils.html?highlight=connect_pvlist#apstools.utils.connect_pvlist) to connect.  This function returns a dictionary of `EpicsSignal` objects.  Refactor `saveFlyData()` accordingly.

prjemian, 2019-07-29:
Let's describe with more details how HDF5 files are created during the process described above: https://github.com/APS-USAXS/ipython-usaxs/issues/276#issuecomment-515991309

1. `run_command_file()` plan (`50-plans.py`) is started
1. `before_command_list()` plan (`50-plans.py`) is called
1. the command list is parsed; a command is executed if a handler is defined
1. `after_command_list()` (`50-plans.py`) plan is called

`FlyScan` commands are handled:

1. `Flyscan()` plan (`50-plans.py`) is started
1. `before_plan()` (`50-plans.py`) plan is called
1. HDF5 file name and path are determined
1. transmission is measured
1. amplifiers are autoscaled
1. plan-specific RunEngine metadata is added
1. `usaxs_flyscan.plan()` (`32-usaxs_flyscan.py`) is called (HDF5 file saved in this step)
1. `after_plan()` (`50-plans.py`) plan is called

When `usaxs_flyscan.plan()` (`32-usaxs_flyscan.py`) is called, these steps:

1. `bp.open_run()` (`bluesky.plans`) is called
1. `prepare_HDF5_file()` (`32-usaxs_flyscan.py`) is started in a thread
1. `progress_reporting()` (`32-usaxs_flyscan.py`) is started in a thread
1. EPICS is told to run the fly scan
1. `usaxs_flyscan.plan()` (`32-usaxs_flyscan.py`) waits until EPICS reports the fly scan is done or timeout
1. `finish_HDF5_file()` (`32-usaxs_flyscan.py`) is started in a thread
1. MCA data signals are added (to bluesky documents) as data stream `mca`
1. `bp.close_run()` (`bluesky.plans`) is called

`prepare_HDF5_file()` (`32-usaxs_flyscan.py`) steps:

1. if directory does not exist, use fallback directory, log this decision
1. if filename exists, use fallback name, log this decision
1. `SaveFlyScan()` object (`usaxs_support.saveFlyData.py`) is created (reads config file, connects with PVs, creates HDF5 file structure)
1. `preliminaryWriteFile()` (`usaxs_support.saveFlyData.py`) is called (writes initial data to HDF5 file)

`finish_HDF5_file()` (`32-usaxs_flyscan.py`) steps:

1. raise `RuntimeError` is `prepare_HDF5_file()` not called first
1. `saveFile()` (`usaxs_support.saveFlyData.py`) is called (writes final data to HDF5 file)

prjemian, 2019-07-29:
One obvious step is to minimize the number of times the config file is parsed and PVs connected.  Perhaps this can all be contained within the `usaxs_support.saveFlyData.py` module.  The PV configuration could be a singleton.  If that object is `None`, the next FlyScan would need to create an instance of the PV configuration class (TBA).  The caller would then only need to `reset` this singleton back to `None` to force the configuration to be read on the next fly scan.

All this code could be contained with the `usaxs_support.saveFlyData.py` module, making it easier to maintain.

prjemian, 2019-07-29:
These existing objects could be gathered up into this new structure:

```
field_registry = {}    # key: node/@label,        value: Field_Specification object
group_registry = {}    # key: HDF5 absolute path, value: Group_Specification object
link_registry = {}     # key: node/@label,        value: Link_Specification object
pv_registry = {}       # key: node/@label,        value: PV_Specification object
```

PR #289 (2019-08-01): refactor flyscan pv connections
fixes #276
closed 2019-08-01 by prjemian