APS-USAXS/usaxs-bluesky-ended-2023 #285: tune_ar does nto seem to set proper plot center

jilavsky, 2019-06-26:
using BS the plots center seems to be off. need to check why tune_ar in spec sets center for plotting while tune_ar in BS doe snot. Something is missing. Research needed.

prjemian, 2019-06-27:
notes: your report was timestamped 2019-06-26 16:07
I'll look for tune_AXIS just prior to that time and see what the metadata can tell us.

prjemian, 2019-06-27:
Running bluesky for this file: `/share1/USAXS_data/2019-06/06_26_NXSchool0626.dat`, SPEC afterwards

prjemian, 2019-06-27:
UID of tune_ar() scans within ~ one hour of the report:

* #C Wed Jun 26 15:02:16 2019.  uid = d768c3a2-c56e-4ce5-8256-d218cbd3f954
* #C Wed Jun 26 15:11:48 2019.  uid = 92b92ed5-641a-4d73-9c5d-becdd28726d6
* #C Wed Jun 26 15:16:02 2019.  uid = 91f17635-8754-46e9-af36-7ad06998bafe
* #C Wed Jun 26 15:25:21 2019.  uid = 01502285-3867-447b-b707-adec2bb2ef04
* #C Wed Jun 26 15:34:30 2019.  uid = ba304f97-2247-4177-929e-3b6d508821f8
* #C Wed Jun 26 15:43:40 2019.  uid = ce429062-462e-4a1e-a501-7b96d406df9a

prjemian, 2019-06-27:
@jilavsky - Your report is not so clear.  Does tune_ar fail to find a center or does it find a center in the wrong place?  If the wrong place, how far off is it?  Could it be choice of centroid instead of mid-point of FWHM or peak position?  We've got some choices and can make this selection for AR independent of the choices for other tuned axes.

jilavsky, 2019-06-27:
spec TUNE_AR

```
# /home/beams/USAXS/spec/macros/local/usaxs_tuning.mac
def tune_ar '{
  local __range
  local sbMessage1
  __range = USAXS_tune_ar_range/2
  if( $# == 1 ) { __range = $1 }
 
  sbMessage1 = sprintf("Tunning motor Ar, starting condition is %g", A[ar])
  comment "%s" sbMessage1
  epics_put("9idcLAX:vsc:c0.DLY", 0.02)
  tune_usaxs_motor ar __range -1*__range 30 0.1
  epics_put("9idcLAX:vsc:c0.DLY", 0.05)
  # we are now at the top of the AR rocking curve
  # let us define the beam center as such
  waitmove; get_angles;
  sbMessage1 = sprintf("Tunning motor Ar, ending condition is %g", A[ar])
  comment "%s" sbMessage1
  # store AR center
  set_AR_VAL_CENTER A[ar]
  __ArValCeterUSAXS = A[ar]
  # remember the Q calculation needs a new 2theta0
  # use the current AR encoder position
  epics_put("9idcLAX:USAXS:Q.B", epics_get("9idcLAX:USAXS:Q.A"))
}'

479.USAXS> prdef set_AR_VAL_CENTER

# /home/beams/USAXS/spec/macros/local/usaxs_commands.mac
def set_AR_VAL_CENTER 'epics_put("9idcLAX:USAXS:ARcenter", $1)'
```

jilavsky, 2019-06-27:
SPEC tune_mr

```
480.USAXS> prdef tune_mr

# /home/beams/USAXS/spec/macros/local/usaxs_tuning.mac
def tune_mr '
     global useSBUSAXS
     global useMSstage
     global tune_ok
     if(useSBUSAXS==1||useMSstage==1){counters cnt_num(I00) cnt_num(I00)} else {counters cnt_num(I0) cnt_num(I0)}
     if(useSBUSAXS==1||useMSstage==1){plotselect I00} else {plotselect I0}
  	 epics_put("9idcLAX:vsc:c0.DLY", 0.02)
     tune_usaxs_motor mr -1*USAXS_tune_mr_range/2 USAXS_tune_mr_range/2 30 0.1
     #tune_usaxs_motorFly mr -1*USAXS_tune_mr_range USAXS_tune_mr_range 30 5
	 epics_put("9idcLAX:vsc:c0.DLY", 0.05)
     set_MR_VAL_CENTER A[mr]
     plotselect upd2
     counters cnt_num(I0) cnt_num(upd2)
'

481.USAXS> prdef set_MR_VAL_CENTER

# /home/beams/USAXS/spec/macros/local/usaxs_commands.mac
def set_MR_VAL_CENTER 'epics_put("9idcLAX:USAXS:MRcenter", $1)'
```

jilavsky, 2019-06-27:
also tune_{m,a}r2p and also side stages

jilavsky, 2019-06-27:
The tune is OK but EPICS PVs must be updated for other software to use.

jilavsky, 2019-06-27:
Are we writing `9idcLAX:USAXS:Q.B` from tune_ar?

Looks like it:

```
bash-4.2$ grep "\.B" *
20-detectors.py:scaler2_I000_cps = EpicsSignalRO("9idcLAX:vsc:c2_cts1.B", name="scaler2_I000_counts")
29-axis_tuning.py:            usaxs_q_calc.channels.B.value, terms.USAXS.ar_val_center.value,
50-plans.py:        usaxs_q_calc.channels.B.value, terms.USAXS.ar_val_center.value,
```

prjemian, 2019-07-29:
There are a set PVs to be added to each fly scan as metadata.  The set is described in an XML file.  The file is parsed at the start of each fly scan, the PVs are connected while the scan is progressing, then the set is written at the end.  Likely, the biggest system load comes from the PV connection phase.

As an alternative, parse that XML file only once per batch of scans (which includes a fly scan).  Develop the PV list and use [`APS_utils.connect_pvlist()`](https://apstools.readthedocs.io/en/latest/source/_utils.html?highlight=connect_pvlist#apstools.utils.connect_pvlist) to connect.  This function returns a dictionary of `EpicsSignal` objects.  Refactor `saveFlyData()` accordingly.

prjemian, 2019-08-01:
With the new `APS_utils.replay()` command, we can find a scan and replay it's tunes to examine the decision process.

file: 08_01_ABrady_usaxs.dat
scan: Blank__0016
center should be: 8.85516 (according to Igor)

Find out what BS determined.

prjemian, 2019-08-01:
We should identify which parameter is used from the HDF5 file for generating the livedata plot and verify that it will be correct.

prjemian, 2019-08-01:
We can do this later, not during beam time.

prjemian, 2019-08-01:
Regarding the [scan to find](https://github.com/APS-USAXS/ipython-usaxs/issues/285#issuecomment-517377542): that's actually scan 22 in file: `/share1/USAXS_data/2019-08/08_01_ABrady.dat`

```
#S 22  Flyscan(pos_X=50.4, pos_Y=-30.0, thickness=1.0, scan_title=Blank)
#D Thu Aug 01 12:05:14 2019
#C Thu Aug 01 12:05:14 2019.  plan_type = generator
#C Thu Aug 01 12:05:14 2019.  uid = 93e41540-b4fc-4dc6-8d8f-134d85271c45
#C Thu Aug 01 12:05:16 2019.  start USAXS Fly scan
#C Thu Aug 01 12:05:16 2019.  HDF5 configuration file: /share1/AreaDetectorConfig/FlyScan_config/saveFlyData.xml
#MD uid = 93e41540-b4fc-4dc6-8d8f-134d85271c45
#MD EPICS_CA_MAX_ARRAY_BYTES = 1280000
#MD EPICS_HOST_ARCH = linux-x86_64
#MD action = USAXSscan
... something really too big
#MD beamline_id = APS USAXS 9-ID-C
#MD datetime = 2019-08-01 11:58:24.599616
#MD filename = usaxs.mac
#MD fly_scan_time = 90.0
#MD full_filename = /share1/USAXS_data/2019-08/usaxs.mac
#MD hdf5_file = Blank_0016.h5
#MD hdf5_path = /share1/USAXS_data/2019-08/08_01_ABrady_usaxs
#MD line_number = 43
#MD login_id = usaxs@usaxscontrol.xray.aps.anl.gov
#MD parameters = ['50.400', '-30.000', '1', 'Blank']
#MD pid = 53167
#MD proposal_id = testing Bluesky installation
#MD purpose = tuner
#MD versions = {'bluesky': '1.5.3', 'ophyd': '1.3.3', 'databroker': '0.13.0', 'apstools': '1.1.11', 'epics': '3.3.3'}
#N 0
#C no data column labels identified
#C Thu Aug 01 12:06:55 2019.  fly scan completed in 98.98128247261047 s
#C Thu Aug 01 12:06:55 2019.  finished writing fly scan HDF5 file: /share1/USAXS_data/2019-08/08_01_ABrady_usaxs/Blank_0016.h5
#C Thu Aug 01 12:06:56 2019.  num_events_baseline = 2
#C Thu Aug 01 12:06:56 2019.  num_events_mca = 1
#C Thu Aug 01 12:06:56 2019.  exit_status = success
```

prjemian, 2019-08-01:
From this command:

```
In [10]: APS_utils.list_recent_scans(num=200, show_command=True)                                                                                                  
========= ========================== ======= ========================================
short_uid date/time                  scan_id command                                 
========= ========================== ======= ========================================
...
93e4154   2019-08-01 12:05:14.818483 22      Flyscan(pos_X=50.4, pos_Y=-30.0, thi ...
a571cc7   2019-08-01 12:05:05.719313 21      measure_USAXS_Transmission(detectors ...
c90ba3b   2019-08-01 12:04:48.971826 20      tune_a2rp()                             
c8429eb   2019-08-01 12:04:29.696432 19      tune_ar()                               
c4de10b   2019-08-01 12:04:22.198886 18      tune_m2rp()                             
9735b7a   2019-08-01 12:04:04.607618 17      tune_mr()                               
```

So let's look at c8429eb.

prjemian, 2019-08-01:
[image]

Nice but not what we need. We want to see the output of PeakStats.

prjemian, 2019-08-01:
Almost there:

```
In [13]: h = db["c8429eb"]                                                                                                                                        

In [14]: h.stop                                                                                                                                                   
Out[14]: 
{'run_start': 'c8429ebd-4b79-4a04-b9c4-4431a701230f',
 'time': 1564679086.7192347,
 'uid': 'bdf3b616-d2d6-4b34-a4a4-2ce230c8d889',
 'exit_status': 'success',
 'num_events': {'baseline': 2, 'primary': 35, 'PeakStats': 1}}

In [15]: h.table("PeakStats")                                                                                                                                     
Out[15]: 
                                 time  PeakStats_tune_ok                  ...                          PeakStats_max                   PeakStats_crossings
seq_num                                                                   ...                                                                             
1       2019-08-01 12:04:46.072799444               True                  ...                   [8.8854597, 23468.0]  [8.885210736671368, 8.8857416194051]

[1 rows x 13 columns]
```

prjemian, 2019-08-01:
After a bit of command line work, this is what we seek:

```
In [28]: hh = list(h.documents("PeakStats"))

In [29]: hh[2][1]["data"]                                                                                                                                         
Out[29]: 
{'PeakStats_tune_ok': True,
 'PeakStats_initial_position': 8.8854597,
 'PeakStats_final_position': 8.885470486471954,
 'PeakStats_center': 8.885470486471954,
 'PeakStats_x': 'a_stage_r',
 'PeakStats_y': 'PD_USAXS',
 'PeakStats_cen': 8.885476178038235,
 'PeakStats_com': 8.885470486471954,
 'PeakStats_fwhm': 0.000530882733732696,
 'PeakStats_min': array([   8.8849597, 4178.       ]),
 'PeakStats_max': array([8.8854597e+00, 2.3468000e+04]),
 'PeakStats_crossings': array([8.88521074, 8.88574162])}

```

prjemian, 2019-08-01:
Our code has picked the COM value (compare PeakStats_final_position with PeakStats_com), same as SPEC.

Looking for 8.85516, as determined by Igor, not finding that here:

| source | value |
| --- | --- |
| max | 8.8854597 |
| COM | 8.885470486471954 |
| CEN | 8.885476178038235 |
| final | 8.885470486471954 |
| IgorPro | **8.85516** |

prjemian, 2019-08-01:
After the `tune_ar()` scan: this procedure is called to set the EPICS PV for beam center: https://github.com/APS-USAXS/ipython-usaxs/blob/7b80eeda2c904a19ce5a52a69c5f736c96bf9238/profile_bluesky/startup/29-axis_tuning.py#L258-L270

Line 263 sets the signal (PV=`9idcLAX:USAXS:ARcenter`), line 267 set the calculation to use the same number.

note: 

```
In [30]: terms.USAXS.ar_val_center.pvname                                                                                                                         
Out[30]: '9idcLAX:USAXS:ARcenter'
```

prjemian, 2019-08-01:
That value is also written to the HDF file using this configuration: https://github.com/APS-USAXS/ipython-usaxs/blob/fe6c3757b137c45ec30b36d617f918562f1128b9/profile_bluesky/startup/usaxs_support/saveFlyData.xml#L282

prjemian, 2019-08-01:
The livedata page (in ` livedata/reduceFlyData.py`) picks up a value from the HDF5 file: 

    ar_center  = float(hdf['/entry/metadata/AR_center'][0])

and passes that to the *R(Q)* calculator:

        full = calc.calc_R_Q(wavelength,            # wavelength
                             raw_ar,                # ar
                             channel_time_s,        # seconds
                             raw_upd,               # pd
                             upd_dark,              # pd_bkg
                             upd_gain,              # pd_gain
                             raw_I0,                # I0
                             I0_gain=I0_amp_gain,
                             ar_center=ar_center,
                             )

where that value is used for the beam center of the USAXS fly scans:

    if ar_center is None:       # compute ar_center from rVec and ar
        ar_center = centroid(ar, rVec)      # centroid of central peak

prjemian, 2019-08-01:
If, instead, we were to ignore the `tune_ar()` final position and pass `None` to the *R(Q)* calculator, then it would compute the centroid using the scipy library.  Might be interesting to try.

Note that Igor adds an additional filter to the centroid computation data:

1. find the maximum intensity
1. verify it is comparable to its adjacent data (not a noise value)
1. walk down both sides and find the 40% level
1. mask off and values below that cutoff (to avoid bias by long scan on one side of peak)
1. evaluate centroid

This can be tested numerically on the above data.

prjemian, 2019-08-02:
It's worthwhile to see this data plotted as *R* vs. *ar*:
[image]

@jilavsky: No way that Igor can find the center of this peak at 8.85516.  That's off the chart.  What scan was Igor describing?

Using Igor's algorithm with the python centroid method above, python computes the centroid as **8.885845099822449** using points 63-83.  My eye would tell me different, at: 8.8860 or so.

So, with revision, Python could do a better job finding the centroid and FWHM than by using the previous tune_ar value or by using its current method (biased by the long tails).

prjemian, 2019-08-02:
Python code:

```

"""

compute centroid of USAXS fly scan data

"""

import h5py

import numpy

import os

import sys

import reduceFlyData

import calc

TESTFILE = "Blank_0016.h5"

def centroid(x, y):

    '''compute centroid of y(x)'''

    import scipy.integrate

    weight = y*y

    top    = scipy.integrate.simps(x*weight, x)

    bottom = scipy.integrate.simps(weight, x)

    center = top/bottom

    return center

def flyscan_centroid(x, y, peak_index, cutoff):

    # walk down each side from the peak

    n = len(x)

    pLo = peak_index

    while pLo >= 0 and y[pLo] > cutoff:

        pLo -= 1

    pHi = peak_index + 1

    while pHi < n and y[pHi] > cutoff:

        pHi += 1

    

    print pLo, pHi

    ar = x[pLo:pHi]

    R = y[pLo:pHi]

    cen = centroid(ar, R)

    return cen, ar, R

    

def main():

    fly = reduceFlyData.UsaxsFlyScan(TESTFILE)

    fly.reduce()

    full = fly.reduced["full"]

    print full.keys()

    print "ar[:5]", full['ar'][:5]

    for k in "R_max r_peak ar_r_peak".split():

        print k, full[k]

    

    print "peak index:", list(full["ar"]).index(full["ar_r_peak"])

    

    cen, ar, R = flyscan_centroid(

        full["ar"],

        full["R"],

        list(full["ar"]).index(full["ar_r_peak"]),

        full["R_max"] * 0.4,

    )

    print "centroid:", cen

    

    with h5py.File("reduced.h5") as h5:

        h5.create_dataset("ar", data=ar)

        h5.create_dataset("R", data=R)

        h5.create_dataset("cen", data=cen)

if __name__ == "__main__":

    main()

```

prjemian, 2019-08-02:
output from that code:

```
(py27) mintadmin@mint-vm:~/.../eclipse/livedata$ /bin/rm reduced.h5 
(py27) mintadmin@mint-vm:~/.../eclipse/livedata$ python ./compare_centers.py 
/home/mintadmin/Apps/anaconda/envs/py27/lib/python2.7/site-packages/h5py/_hl/dataset.py:313: H5pyDeprecationWarning: dataset.value has been deprecated. Use dataset[()] instead.
  "Use dataset[()] instead.", H5pyDeprecationWarning)
['ar_0', 'r0', 'r_peak', 'Q', 'ar_r_peak', 'r', 'R_max', 'R', 'ar']
ar[:5] [8.88814779 8.88811789 8.88808802 8.8880582  8.88802841]
R_max 1487.0982317117118
r_peak 1487.0982317117118
ar_r_peak 8.885993218921367
peak index: 74
62 84
centroid: 8.886022840783497
(py27) mintadmin@mint-vm:~/.../eclipse/livedata$ 
```

prjemian, 2019-08-02:
Issue has been created to apply this new computation.  Will close this issue once the computation is revised.

prjemian, 2019-08-02:
@jilavsky : Test the [livedata PR](https://github.com/APS-USAXS/livedata/pull/36) on Monday.

prjemian, 2019-08-03:
The python code needs to determine that the maximum R is not a noise point.  One of the scans today shows a great example for testing.

| index | title | type | scan | file | started | ended |
| -- | -- | -- | -- | -- | -- | --
| 23360 | blank | FlyScan | 1255 | /share1/USAXS_data/2019-08/08_02_LLNL_Plates.dat | 2019-08-03 16:18:47 | 2019-08-03 16:19:05 |

prjemian, 2019-08-03:
That is, emphasize item 2 of https://github.com/APS-USAXS/ipython-usaxs/issues/285#issuecomment-517480219, above.

prjemian, 2019-08-03:
From the SPEC file:

```
#C Sat Aug 03 15:16:13 2019.  FlyScan #1240 started.
#C Sat Aug 03 15:16:13 2019.  FlyScan file name = ./08_02_LLNL_Plates_usaxs/S6_r1SOTy2_0235.h5.
```

prjemian, 2019-08-03:
One effective test if a given point is a *zinger* is to test with its neighboring values.  With *i* being the index of *R_max*:

    v = average(R_[i-1], R_[i], R_[i+1]) / average(R_[i-1], R_[i+1])

*v* should fluctuate,within a few percent, about 1.0 at the peak.  If *v* is greater than about 1.1, it's probably a zinger.  In the case above, *v ~ 14*, clearly a zinger.

So, what is the threshold *v* for a zinger?  With the above data, and the zinger removed, the threshold is clearly between 1.1 and 1.2.

*R(ar)* with zinger removed:
[image]

*v(ar)* with zinger removed:
[image]

prjemian, 2019-08-03:
Conclusion:  to test if reported *R_max* is a zinger, if *v > 2*, remove the point, determine a new *R_max* and test again.

Another test is that determined center with or without *R_max* should be about the same if *R_max* is not a zinger.

prjemian, 2019-08-03:
Adding this test (and revision of *R(ar)* peak data), the center determined for the above data is 8.87407689229.  *Plausible.*

prjemian, 2019-08-05:
Fix installed into [livedata](https://github.com/APS-USAXS/livedata/issues/35)

closed 2019-08-05 by prjemian