APS-USAXS/usaxs-bluesky-ended-2023 #306: Tune Guard slits is not working

jilavsky, 2019-10-01:
Tune Guard slits is not working, causes error. Needs to be tested in December with the beam. 

```
 2019-10-01 10:09:18.966 - Returning the guard slit motors to original (pre-tune) positions
[E 10:09:22.491 run_engine:1261] Run aborted                                                                                                                          
    Traceback (most recent call last):                                                                                                                                
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/run_engine.py", line 1150, in _run                                                        
        msg = self._plan_stack[-1].send(resp)                                                                                                                         
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 1294, in __call__
        return (yield from plan)
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 1149, in baseline_wrapper
        return (yield from plan_mutator(plan, insert_baseline))
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 169, in plan_mutator
        raise ex
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 122, in plan_mutator
        msg = plan_stack[-1].send(ret)
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 790, in monitor_during_wrapper
        return (yield from plan2)
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 169, in plan_mutator
        raise ex
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 122, in plan_mutator
        msg = plan_stack[-1].send(ret)
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 169, in plan_mutator
        raise ex
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 122, in plan_mutator
        msg = plan_stack[-1].send(ret)
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 848, in fly_during_wrapper
        return (yield from plan2)
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 169, in plan_mutator
        raise ex
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 122, in plan_mutator
        msg = plan_stack[-1].send(ret)
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 169, in plan_mutator
        raise ex
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 122, in plan_mutator
        msg = plan_stack[-1].send(ret)
      File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/51-tune_guard_slits.py", line 399, in tune_Gslits
        yield from tune_GslitsSize()
      File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/51-tune_guard_slits.py", line 384, in tune_GslitsSize
        yield from _USAXS_tune_guardSlits()
      File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/51-tune_guard_slits.py", line 294, in _USAXS_tune_guardSlits
        tunes["top"])
      File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/51-tune_guard_slits.py", line 269, in tune_blade_edge
        yield from cleanup(msg)
      File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/51-tune_guard_slits.py", line 228, in cleanup
        raise GuardSlitTuneError(msg)
    GuardSlitTuneError: guard_slit_top: Computed tune position 0.06131566457805631 < 0.07520000000000004.  Not tuning this axis.
---------------------------------------------------------------------------
GuardSlitTuneError                        Traceback (most recent call last)
<ipython-input-11-21e823a1a57b> in <module>
----> 1 RE(tune_Gslits())

/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    751                     # it (unless it is a canceled error)
    752                     if exc is not None:
--> 753                         raise exc
    754 
    755             if self._interrupted:

/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/run_engine.py in _run(self)
   1260             self._reason = str(err)
   1261             self.log.exception("Run aborted")
-> 1262             raise err
   1263         finally:
   1264             # Some done_callbacks may still be alive in other threads.

/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/run_engine.py in _run(self)
   1148                     else:
   1149                         try:
-> 1150                             msg = self._plan_stack[-1].send(resp)
[... 109 lines cut ...]
    386         ti_filter_shutter, "close",

/home/beams11/USAXS/.ipython/profile_bluesky/startup/51-tune_guard_slits.py in _USAXS_tune_guardSlits()
    292         60,
    293         0.25,
--> 294         tunes["top"])
    295 
    296     logger.info("*** 2. tune bottom guard slits")

/home/beams11/USAXS/.ipython/profile_bluesky/startup/51-tune_guard_slits.py in tune_blade_edge(axis, start, end, steps, ct_time, results)
    267             msg = f"{axis.name}: Computed tune position {position} < {min(start, end)}."
    268             msg += "  Not tuning this axis."
--> 269             yield from cleanup(msg)
    270         if position > max(start, end):
    271             msg = f"{axis.name}: Computed tune position {position} > {max(start, end)}."

/home/beams11/USAXS/.ipython/profile_bluesky/startup/51-tune_guard_slits.py in cleanup(msg)
    226             guard_slit.inb, original_position["inb"],
    227             )
--> 228         raise GuardSlitTuneError(msg)
    229 
    230     logger.info("And now we can tune all of the guard slits, blade-by-blade")

GuardSlitTuneError: guard_slit_top: Computed tune position 0.06131566457805631 < 0.07520000000000004.  Not tuning this axis.
```

jilavsky, 2019-10-01:
This is old problem of BlueSky with the Blades motors. Left in moving state. Forgot about it...

prjemian, 2020-08-28:
Closing now as duplicate of #253.

closed 2020-08-28 by prjemian