APS-USAXS/usaxs-bluesky-ended-2023 #307: Flyscan does not save data

jilavsky, 2019-10-01:
Tried to run Flyscan, does not seem to be able to save data. No data are saved in the folder, but folder was cretaed. 
Cannot continue testing without ability to save data. 

```
D 2019-10-01 11:11:01.750 - 120.09       8.2527267    -2.34604     3.09280      4912         64.60        user:216.180s sys:31.762s mem:349.84MB cpu:11.6%
D 2019-10-01 11:11:06.758 - 125.09       8.1104161    -2.86013     0.93059      5470         69.60        user:216.314s sys:31.782s mem:349.84MB cpu:14.7%
D 2019-10-01 11:11:11.766 - 130.10       7.9549591    -3.44737     -1.58743     6036         74.60        user:216.452s sys:31.809s mem:349.84MB cpu:13.7%
D 2019-10-01 11:11:16.775 - 135.11       7.7753808    -4.09100     -4.32842     6606         79.62        user:216.587s sys:31.831s mem:349.84MB cpu:13.8%
D 2019-10-01 11:11:21.825 - 140.12       7.5825860    -4.81016     -7.34582     7180         84.62        user:216.735s sys:31.859s mem:349.84MB cpu:11.5%
D 2019-10-01 11:11:26.799 - 145.12       7.3657525    -5.61081     -10.73358    7762         89.62        user:216.865s sys:31.887s mem:349.84MB cpu:14.3%
D 2019-10-01 11:11:30.897 - writing fly scan HDF5 file: /share1/USAXS_data/2019-10/10_01_setupBS_usaxs/AirBlank_0046.h5
I 2019-10-01 11:11:30.914 - 149.24       7.2384970    -6.04166     -12.55675    7999         93.70        user:216.980s sys:31.908s mem:349.84MB cpu:9.4%
D 2019-10-01 11:11:30.914 - before saveFlyData.finish_HDF5_file(): user:216.980s sys:31.908s mem:349.84MB cpu:9.4%
D 2019-10-01 11:11:30.915 - 149.2528419494629s - progress_reporting is done
D 2019-10-01 11:11:30.915 - after saveFlyData.finish_HDF5_file(): user:216.981s sys:31.910s mem:349.84MB cpu:0.0%
I 2019-10-01 11:11:30.916 - finished writing fly scan HDF5 file: /share1/USAXS_data/2019-10/10_01_setupBS_usaxs/AirBlank_0046.h5
Exception in thread Thread-8651:
Traceback (most recent call last):
  File "/APSshare/anaconda3/BlueSky/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/APSshare/anaconda3/BlueSky/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/32-usaxs_flyscan.py", line 134, in finish_HDF5_file
    raise RuntimeError("Must first call prepare_HDF5_file()")
RuntimeError: Must first call prepare_HDF5_file()

New stream: 'mca'                                                                                                                                                     
[E 11:11:31.195 run_engine:1261] Run aborted
    Traceback (most recent call last):
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/run_engine.py", line 1127, in _run
        self._exception or resp)
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 1294, in __call__
        return (yield from plan)
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 1149, in baseline_wrapper
        return (yield from plan_mutator(plan, insert_baseline))
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 77, in plan_mutator
        msg = plan_stack[-1].throw(exception)
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 790, in monitor_during_wrapper
        return (yield from plan2)
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 77, in plan_mutator
        msg = plan_stack[-1].throw(exception)
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 77, in plan_mutator
        msg = plan_stack[-1].throw(exception)
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 848, in fly_during_wrapper
        return (yield from plan2)
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 77, in plan_mutator
        msg = plan_stack[-1].throw(exception)
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 77, in plan_mutator
        msg = plan_stack[-1].throw(exception)
      File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/50-plans.py", line 387, in run_command_file
        yield from execute_command_list(filename, commands)
      File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/50-plans.py", line 461, in execute_command_list
        yield from Flyscan(sx, sy, sth, snm, md=_md)
      File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/50-plans.py", line 753, in Flyscan
        yield from usaxs_flyscan.plan(md=_md)        # DO THE FLY SCAN
      File "/home/beams11/USAXS/.ipython/profile_bluesky/startup/32-usaxs_flyscan.py", line 205, in plan
        [struck.mca1, struck.mca2, struck.mca3], "mca")
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/apstools/plans.py", line 81, in addDeviceDataAsStream
        yield from bps.save()
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/plan_stubs.py", line 57, in save
        return (yield Msg('save'))
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 194, in plan_mutator
        inner_ret = yield msg
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 194, in plan_mutator
        inner_ret = yield msg
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 194, in plan_mutator
        inner_ret = yield msg
      [Previous line repeated 2 more times]
      File "/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/run_engine.py", line 1208, in _run
        new_response = yield from coro(msg)
[... 166 lines cut ...]
/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
    192         try:
    193             # yield out the 'current message' and collect the return
--> 194             inner_ret = yield msg
    195         except GeneratorExit:
    196             # special case GeneratorExit.  We must clean up all of our plans

/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/run_engine.py in _run(self)
   1206                         # exceptions (coming in via throw) can be
   1207                         # raised
-> 1208                         new_response = yield from coro(msg)
   1209 
   1210                     # special case `CancelledError` and let the outer

/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/bluesky/run_engine.py in _save(self, msg)
   1623         """
   1624         if not self._bundling:
-> 1625             raise IllegalMessageSequence("A 'create' message must be sent, to "
   1626                                          "open an event bundle, before that "
   1627                                          "bundle can be saved with 'save'.")

IllegalMessageSequence: A 'create' message must be sent, to open an event bundle, before that bundle can be saved with 'save'.

In [21]:      
```

jilavsky, 2019-10-01:
Another try. Note, this run for 150seconds while PV for Flyscan time was 90 seconds... Hm... 

Then it crashed.
```
 2019-10-01 14:43:03.707 - gain=10000.0  rate: 773400.0  max: 950000  converged=[True, True]
D 2019-10-01 14:43:03.709 - gain=10000000.0  rate: 500200.0  max: 950000  converged=[True, True, True, True]
D 2019-10-01 14:43:03.711 - gain=1000000000.0  rate: 1800.0  max: 950000  converged=[True, True, True, True, True, True]
Transient Scan ID: 53     Time: 2019-10-01 14:43:04                                                                                                                   
Persistent Unique Scan ID: '8c84fc65-0f54-4cb0-b0ac-d7a5790fc933'                                                                                                     
New stream: 'baseline'                                                                                                                                                
New stream: 'aps_current_monitor'
I 2019-10-01 14:43:05.665 - HDF5 config: /share1/AreaDetectorConfig/FlyScan_config/saveFlyData.xml
I 2019-10-01 14:43:05.665 - HDF5 output: /share1/USAXS_data/2019-10/10_01_setupBS_usaxs/AirBlank_0097.h5
D 2019-10-01 14:43:05.666 - progress_reporting has arrived
D 2019-10-01 14:43:05.668 - before SaveFlyScan(): user:384.897s sys:65.698s mem:366.35MB cpu:15.7%
/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/epics/pv.py:110: UserWarning: The `context` kwarg for epics.get_pv() is deprecated. New PVs will _not_ be created in the requested context.
  'The `context` kwarg for epics.get_pv() is deprecated. New PVs '
I 2019-10-01 14:43:06.281 - flying, s    ar, deg      ay, mm       dy, mm       channel      elapsed, s 
D 2019-10-01 14:43:06.895 - before saveFlyData.preliminaryWriteFile(): user:385.480s sys:65.849s mem:370.44MB cpu:21.4%
D 2019-10-01 14:43:07.411 - after saveFlyData.preliminaryWriteFile(): user:385.850s sys:65.887s mem:373.34MB cpu:23.2%
D 2019-10-01 14:43:10.674 - 5.01         8.8888271    0.01085      13.09436     0            0.00         user:385.903s sys:65.899s mem:373.43MB cpu:5.3%
D 2019-10-01 14:43:15.680 - 10.01        8.8888271    0.01085      13.09436     0            0.00         user:385.984s sys:65.921s mem:373.43MB cpu:5.8%
D 2019-10-01 14:43:20.684 - 15.02        8.8888271    0.01085      13.09436     0            0.00         user:386.062s sys:65.940s mem:373.43MB cpu:6.6%
D 2019-10-01 14:43:25.685 - 20.02        8.8888271    0.01085      13.09436     0            0.00         user:386.141s sys:65.961s mem:373.43MB cpu:8.0%
D 2019-10-01 14:43:30.690 - 25.02        8.8888271    0.01085      13.09436     0            0.00         user:386.220s sys:65.981s mem:373.43MB cpu:8.5%
D 2019-10-01 14:43:35.694 - 30.03        8.8888271    0.01085      13.09436     0            0.00         user:386.302s sys:65.999s mem:373.43MB cpu:9.2%
D 2019-10-01 14:43:40.702 - 35.04        8.8888271    0.01085      13.09436     0            0.00         user:386.386s sys:66.014s mem:373.43MB cpu:8.3%
D 2019-10-01 14:43:45.708 - 40.04        8.8888271    0.01085      13.09436     0            0.00         user:386.468s sys:66.034s mem:373.43MB cpu:7.0%
D 2019-10-01 14:43:50.714 - 45.05        8.8888271    0.01085      13.09436     0            0.00         user:386.546s sys:66.059s mem:373.43MB cpu:5.4%
D 2019-10-01 14:43:55.715 - 50.05        8.8888271    0.01085      13.09436     0            0.00         user:386.628s sys:66.075s mem:373.43MB cpu:5.4%
D 2019-10-01 14:44:00.724 - 55.06        8.8888271    0.01085      13.09436     0            0.00         user:386.714s sys:66.090s mem:373.43MB cpu:4.1%
D 2019-10-01 14:44:05.729 - 60.06        8.8863317    0.00140      13.05429     66           4.57         user:386.855s sys:66.118s mem:373.43MB cpu:8.8%
D 2019-10-01 14:44:10.735 - 65.07        8.8847293    -0.00468     13.02868     126          9.58         user:386.998s sys:66.143s mem:373.43MB cpu:8.0%
D 2019-10-01 14:44:15.745 - 70.08        8.8788392    -0.02710     12.93298     296          14.58        user:387.134s sys:66.172s mem:373.43MB cpu:8.7%
D 2019-10-01 14:44:20.755 - 75.09        8.8664258    -0.07390     12.72986     575          19.60        user:387.275s sys:66.199s mem:373.43MB cpu:10.7%
D 2019-10-01 14:44:25.760 - 80.09        8.8454798    -0.15276     12.40065     930          24.60        user:387.416s sys:66.231s mem:373.43MB cpu:10.4%
D 2019-10-01 14:44:30.768 - 85.10        8.8153885    -0.26440     11.93106     1342         29.62        user:387.562s sys:66.256s mem:373.43MB cpu:11.0%
D 2019-10-01 14:44:35.775 - 90.11        8.7738725    -0.41677     11.28148     1791         34.62        user:387.702s sys:66.288s mem:373.43MB cpu:10.4%
D 2019-10-01 14:44:40.780 - 95.11        8.7207214    -0.61463     10.44279     2270         39.62        user:387.843s sys:66.315s mem:373.43MB cpu:9.2%
D 2019-10-01 14:44:45.824 - 100.12       8.6562229    -0.85327     9.43191      2771         44.62        user:387.986s sys:66.348s mem:373.43MB cpu:7.2%
D 2019-10-01 14:44:50.800 - 105.12       8.5762366    -1.14867     8.18882      3292         49.62        user:388.130s sys:66.379s mem:373.63MB cpu:8.5%
D 2019-10-01 14:44:55.826 - 110.13       8.4842248    -1.49454     6.73959      3820         54.62        user:388.281s sys:66.409s mem:373.64MB cpu:6.3%
D 2019-10-01 14:45:00.808 - 115.13       8.3743410    -1.89356     5.02186      4366         59.62        user:388.424s sys:66.438s mem:373.64MB cpu:7.1%
[... cut ...]
   1626                                          "open an event bundle, before that "
   1627                                          "bundle can be saved with 'save'.")

IllegalMessageSequence: A 'create' message must be sent, to open an event bundle, before that bundle can be saved with 'save'.

/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/ophyd/ophydobj.py:269: RuntimeWarning: coroutine 'Event.wait' was never awaited
  cb(*args, **kwargs)
RuntimeWarning: Enable tracemalloc to get the object allocation traceback

```

prjemian, 2019-10-01:
Aha!  I see it.  It's an indentation error in the code when it saves the raw MCA channels as Bluesky event data:

```
/APSshare/anaconda3/BlueSky/lib/python3.7/site-packages/apstools/plans.py in addDeviceDataAsStream(devices, label)
     79     for d in devices:
     80         yield from bps.read(d)
---> 81         yield from bps.save()
     82 
     83 
```

written without line numbers:

```
     for d in devices:
         yield from bps.read(d)
         yield from bps.save()
```

Should be:

```
     for d in devices:
         yield from bps.read(d)
     yield from bps.save()
```

The `bps.save()` should be called *once and only once* for every `bps.create()`.  Here, it is called each time through the loop and that's just an indentation typo.

prjemian, 2019-10-01:
Since that comes from [`apstools.plans`](https://github.com/BCDA-APS/apstools/blob/master/apstools/plans.py), we'll need to replace that locally until `apstools` is updated.

prjemian, 2019-10-01:
@jilavsky : Thanks for the complete report!

prjemian, 2019-10-01:
fixed with commit 98bc0b99601f9c385d166ae9df014df998b567db (mislabeled to #255)

closed 2019-10-01 by prjemian