APS-USAXS/usaxs-bluesky-ended-2023 #309: Matplotlib fails after multiple tuning runs

jilavsky, 2019-10-02:
When graphs are opened for too many scans, matPlotLib used to generate these has bug. Graphs become useless and there does not seem obvious way to fix this. 
I suspect this may have no solution, but it is ugly and makes plots useless. Having these plots over time actually has serious value for me. 

[image]

jilavsky, 2019-10-02:
While we are on these graphs, could these graphs be in tabbed interface? It is really difficult to layout 4 or more graphs on screen... If they were all docked in single window with tabs, we may have chance to utilize them better... Just food for though ;-)

jilavsky, 2019-10-02:
The size can be fixed from graph controls, but next time a new data set is appended, it is screwed up again. The legend is causing these issues as it is too large. Need some smarts to keep legend under control, as Irena has for its graphs.

jilavsky, 2019-11-30:
This seems to start impacting the shape with around 18-20 lines. There is no hard limit as it depends on physical size of image also. In any case, we should limit display to last 15 scans, if possible.

jilavsky, 2019-12-02:
@prjemian is mocking this up offline

prjemian, 2019-12-02:
the offline failed after 187 scans, the failure happened while returning the motor to the peak center

prjemian, 2019-12-02:
subsequent tests with the offline simulator failed after 16, 16, 1, 1, and 3 runs -- **ALWAYS** when waiting for the motor move to complete, *never* from the plotting module.  Here is the console output from the last scan, with exception trace:

```
In [97]: RE(my_test(), purpose="testing LivePlot failure mode")                                                                                                           
                                                                                                                                                                          
I Mon-12:54:10 - new settings for simulated peak #1:
============== ====================
parameter      value               
============== ====================
peak #         1                   
center         -0.5760758688075183 
width          0.01928846101415571 
scale          96420.60323210615   
noise          0.032283943915875045
motor position 0.0                 
============== ====================
Transient Scan ID: 716     Time: 2019-12-02 12:54:10
Persistent Unique Scan ID: '1f99dfbf-fcf4-4764-9d43-a36b3d62145e'
New stream: 'primary'                                                                                                                                                     
+-----------+------------+------------+------------+
|   seq_num |       time |         m1 |      noisy |
+-----------+------------+------------+------------+
|         1 | 12:54:13.1 |   -2.00000 |   17.31375 |
|         2 | 12:54:13.4 |   -1.90000 |   19.45359 |                                                                                                                      
|         3 | 12:54:13.7 |   -1.80000 |   23.18015 |                                                                                                                      
|         4 | 12:54:14.0 |   -1.70000 |   27.53876 |                                                                                                                      
|         5 | 12:54:14.3 |   -1.60000 |   33.32324 |                                                                                                                      
|         6 | 12:54:14.6 |   -1.50000 |   40.85971 |                                                                                                                      
|         7 | 12:54:14.9 |   -1.40000 |   51.34895 |                                                                                                                      
|         8 | 12:54:15.2 |   -1.30000 |   66.95457 |                                                                                                                      
|         9 | 12:54:15.5 |   -1.20000 |   89.42539 |                                                                                                                      
|        10 | 12:54:15.8 |   -1.10000 |  126.26293 |                                                                                                                      
|        11 | 12:54:16.1 |   -1.00000 |  190.02627 |                                                                                                                      
|        12 | 12:54:16.4 |   -0.90000 |  327.60697 |                                                                                                                      
|        13 | 12:54:16.7 |   -0.80000 |  693.54093 |                                                                                                                      
|        14 | 12:54:17.0 |   -0.70000 | 2170.21097 |                                                                                                                      
|        15 | 12:54:17.3 |   -0.60000 | 37238.91411 |                                                                                                                     
|        16 | 12:54:17.6 |   -0.50000 | 5704.33050 |                                                                                                                      
|        17 | 12:54:17.9 |   -0.40000 | 1093.32286 |                                                                                                                      
|        18 | 12:54:18.2 |   -0.30000 |  452.41494 |                                                                                                                      
|        19 | 12:54:18.5 |   -0.20000 |  248.27339 |                                                                                                                      
|        20 | 12:54:18.8 |   -0.10000 |  151.08782 |                                                                                                                      
|        21 | 12:54:19.1 |    0.00000 |  104.21783 |                                                                                                                      
|        22 | 12:54:19.4 |    0.10000 |   76.75076 |                                                                                                                      
|        23 | 12:54:19.7 |    0.20000 |   58.08063 |                                                                                                                      
|        24 | 12:54:20.0 |    0.30000 |   45.26278 |                                                                                                                      
[... cut ...]
      File "/home/mintadmin/Apps/anaconda/envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 194, in plan_mutator
        inner_ret = yield msg
      File "/home/mintadmin/Apps/anaconda/envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 194, in plan_mutator
        inner_ret = yield msg
      File "/home/mintadmin/Apps/anaconda/envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 194, in plan_mutator
        inner_ret = yield msg
      [Previous line repeated 1 more time]
    bluesky.utils.RequestAbort
Out[98]: ['1f99dfbf-fcf4-4764-9d43-a36b3d62145e']
```

prjemian, 2019-12-02:
Here's the last plot:  
[image]

prjemian, 2019-12-02:
@jilavsky - two questions

1. What are the last few lines of output on the console before you press `^C^C`?
1. What is the terminal window output after `RE.abort()`?

I suspect BS may be waiting for the motor move to complete.

prjemian, 2019-12-02:
Offline simulator here: https://github.com/prjemian/ipython-vm7/commit/48d925cba480bcdcfa3051b33ffad71b412e196e

prjemian, 2019-12-02:
Repeated, again the RunEngine is stuck while waiting for the motor to move (or complete moving).  This time after 8 scans.

We need to see what is reported from the instrument when this fails.

jilavsky, 2019-12-02:
Motor motion is not the problem at USAXS.  Here, the bluesky plans continue but the graphs are useless.  The simulator is different.

prjemian, 2019-12-02:
Looking at how to limit the number of plots to just the *n* most recent (FIFO).  This sequence of commands in ipython dropped the older scans from the plot, leaving just the 3 most recent:

```
In [65]: bec._live_plots                                                                                                                                                  
Out[65]: {'e794e29d-2498-4861-9886-cc54a6607d69': {'noisy': <bluesky.callbacks.best_effort.LivePlotPlusPeaks at 0x7efe64e55350>}}

In [66]: bec._live_plots["e794e29d-2498-4861-9886-cc54a6607d69"]["noisy"]                                                                                                 
Out[66]: <bluesky.callbacks.best_effort.LivePlotPlusPeaks at 0x7efe64e55350>

In [67]: lp = bec._live_plots["e794e29d-2498-4861-9886-cc54a6607d69"]["noisy"]                                                                                            

In [68]: lp.lines                                                                                                                                                         
Out[68]: [<matplotlib.lines.Line2D at 0x7efe64e55450>]

In [69]: lp.ax                                                                                                                                                            
Out[69]: <matplotlib.axes._subplots.AxesSubplot at 0x7efe5ea88210>

In [70]: lp.ax.get_children()                                                                                                                                             
Out[70]: 
[<matplotlib.lines.Line2D at 0x7efe5ebb7950>,
 <matplotlib.lines.Line2D at 0x7efe5ec3dbd0>,
 <matplotlib.lines.Line2D at 0x7efe64f69ed0>,
 <matplotlib.lines.Line2D at 0x7efe65077690>,
 <matplotlib.lines.Line2D at 0x7efe5ec4b0d0>,
 <matplotlib.lines.Line2D at 0x7efe648f8210>,
 <matplotlib.lines.Line2D at 0x7efe64ec1a10>,
 <matplotlib.lines.Line2D at 0x7efe64bc5690>,
 <matplotlib.lines.Line2D at 0x7efe64989b10>,
 <matplotlib.lines.Line2D at 0x7efe64e55450>,
 <matplotlib.lines.Line2D at 0x7efe648cc250>,
 <matplotlib.lines.Line2D at 0x7efe64f42890>,
 <matplotlib.spines.Spine at 0x7efe5e9dda10>,
 <matplotlib.spines.Spine at 0x7efe5e9ddb50>,
 <matplotlib.spines.Spine at 0x7efe5e9dd410>,
 <matplotlib.spines.Spine at 0x7efe5e9dd650>,
 <matplotlib.axis.XAxis at 0x7efe5e9dd9d0>,
 <matplotlib.axis.YAxis at 0x7efe5e9d6110>,
 Text(0.5, 1, ''),
 Text(0.0, 1, ''),
 Text(1.0, 1, ''),
 <matplotlib.legend.Legend at 0x7efe6499c990>,
 <matplotlib.patches.Rectangle at 0x7efe5eb945d0>]

In [71]: lp.ax.get_lines()                                                                                                                                                
Out[71]: <a list of 12 Line2D objects>

In [72]: list(lp.ax.get_lines())                                                                                                                                          
Out[72]: 
[<matplotlib.lines.Line2D at 0x7efe5ebb7950>,
 <matplotlib.lines.Line2D at 0x7efe5ec3dbd0>,
 <matplotlib.lines.Line2D at 0x7efe64f69ed0>,
 <matplotlib.lines.Line2D at 0x7efe65077690>,
 <matplotlib.lines.Line2D at 0x7efe5ec4b0d0>,
 <matplotlib.lines.Line2D at 0x7efe648f8210>,
 <matplotlib.lines.Line2D at 0x7efe64ec1a10>,
 <matplotlib.lines.Line2D at 0x7efe64bc5690>,
 <matplotlib.lines.Line2D at 0x7efe64989b10>,
 <matplotlib.lines.Line2D at 0x7efe64e55450>,
 <matplotlib.lines.Line2D at 0x7efe648cc250>,
 <matplotlib.lines.Line2D at 0x7efe64f42890>]

In [73]: lp.ax.get_lines??                                                                                                                                                
Signature: lp.ax.get_lines()
Source:   
    def get_lines(self):
        """Return a list of lines contained by the Axes"""
        return cbook.silent_list('Line2D', self.lines)
File:      ~/Apps/anaconda/envs/bluesky/lib/python3.7/site-packages/matplotlib/axes/_base.py
Type:      method

In [74]: lp.ax.lines                                                                                                                                                      
Out[74]: 
[<matplotlib.lines.Line2D at 0x7efe5ebb7950>,
 <matplotlib.lines.Line2D at 0x7efe5ec3dbd0>,
 <matplotlib.lines.Line2D at 0x7efe64f69ed0>,
 <matplotlib.lines.Line2D at 0x7efe65077690>,
 <matplotlib.lines.Line2D at 0x7efe5ec4b0d0>,
 <matplotlib.lines.Line2D at 0x7efe648f8210>,
 <matplotlib.lines.Line2D at 0x7efe64ec1a10>,
 <matplotlib.lines.Line2D at 0x7efe64bc5690>,
 <matplotlib.lines.Line2D at 0x7efe64989b10>,
 <matplotlib.lines.Line2D at 0x7efe64e55450>,
 <matplotlib.lines.Line2D at 0x7efe648cc250>,
 <matplotlib.lines.Line2D at 0x7efe64f42890>]

In [75]: lp.ax.lines = lp.ax.lines[-3:]                                                                                                                                   

In [76]: lp.update_plot()                                                                                                                                                 
```

However, it did not change the plot legend.  After typing `P` in the plot window, the legend redrew itself.
[image]

prjemian, 2019-12-02:
1. remove the plot window
1. create new plot window with 4 new scans, zooming into the apparent peak
1. get UUID of plots
1. get last plot object
1. keep only the last two scans
1. scan one more into the same window

```
# 2.
RE(bp.scan([noisy], m1, -2, 2, 21))
RE(bp.scan([noisy], m1, 0.5, 1, 21))
RE(bp.scan([noisy], m1, 0.8, 0.95, 21))
RE(bp.scan([noisy], m1, 0.86, 0.89, 21))
```

LivePlot at this point:
[image]

```
# 3.
k = list(bec._live_plots.keys())[0]

# 4.
lp = bec._live_plots[list(bec._live_plots.keys())[0]]["noisy"]

# 5.
n = 2
lp.ax.lines = lp.ax.lines[-n:]

#6.
RE(bp.scan([noisy], m1, 0, 2, 21))
```

LivePlot at this point: 
[image]

prjemian, 2019-12-02:
Procedure described in https://github.com/APS-USAXS/ipython-usaxs/issues/309#issuecomment-560700459 seems easy to follow except for the step `#3` for the case when more than one plot window exists on screen.  Querying the values of `lp.x` and `lp.y` are necessary to identify uniquely the plot with the correct axes:

```
In [99]: lp.x                                                                                                                                                             
Out[99]: 'm1'

In [100]: lp.y                                                                                                                                                            
Out[100]: 'noisy'
```

prjemian, 2019-12-02:
Here's a routine that finds the *right* plot (same x & y labels) and prunes it to the last *n* scans.

```
def plot_prune_fifo(n, y, x):
    """
    find the plot with axes x and y and replot with only the last *n* lines
    """
    for liveplot in bec._live_plots.values():
        lp = liveplot.get(y)
        if y is None:
            continue
        if lp.x != x or lp.y != y:
            continue
        # print(lp.x, lp.y)
        if len(lp.ax.lines) > n:
            lp.ax.lines = lp.ax.lines[-n:]
            lp.update_plot()

plot_prune_fifo(2, "noisy", "m1")
```

prjemian, 2019-12-02:
Refactor that to use the ophyd object rather than the objects' names.

prjemian, 2019-12-02:
Like this:

```
def plot_prune_fifo(n, y, x):
    """
    find the plot with axes x and y and replot with only the last *n* lines

    PARAMETERS
    
    n : int
        number of plots to keep
    """
    for liveplot in bec._live_plots.values():
        lp = liveplot.get(y.name)
        if lp is None:
            continue
        if lp.x != x.name or lp.y != y.name:
            continue
        # print(lp.x, lp.y)
        if len(lp.ax.lines) > n:
            lp.ax.lines = lp.ax.lines[-n:]
            lp.update_plot()

plot_prune_fifo(2, noisy, m1)
RE(bp.scan([noisy], m1, 0.75, 1.0, 31))
plot_prune_fifo(2, noisy, m1)
RE(bp.scan([noisy], m1, 0.85, 0.91, 31))

```

prjemian, 2019-12-02:
Note:  The [BestEffortCallback](https://blueskyproject.io/bluesky/callbacks.html?highlight=besteffortcallback#best-effort-callback) object `bec` has two methods to enable and disable plotting (`bec.enable_plots()` & `bec.disable_plots()`) that are attractive.  However, when plots are disabled, data collection for the peak statistics (used to determine the peak center and width for tuning) is also turned off.

tl;dr: Don't use `bec.disable_plots()`!  You won't get the peak center and width.

prjemian, 2019-12-02:
This plan shows how to display the plot for the previous and current scan:

```
def showme():
    for i in range(3):
        plot_prune_fifo(1, noisy, m1)
        yield from bp.scan([noisy], m1, 0.85, 0.91, 31)

        plot_prune_fifo(1, noisy, m1)
        yield from bp.scan([noisy], m1, 0.75, 1.0, 31)

        plot_prune_fifo(1, noisy, m1)
        yield from bp.scan([noisy], m1, 0.5, 1.5, 31)

RE(showme())
```

prjemian, 2019-12-02:
Using a tabbed interface is a separate issue (wish list item) and has been brought up in bluesky (https://github.com/bluesky/bluesky/issues/1269).  Proceding with FIFO-style implementation.  From the report above, no more than about 10 lines per graph seems OK.  A more practical case might be no more than 2 or three scans per graph.  (*n* is 1 or 2)

prjemian, 2019-12-02:
The result of this issue might be a good addition to apstools.

commit 2875eb86 (2019-12-02): fixes #309
closed 2019-12-02 by prjemian