APS-USAXS/usaxs-bluesky-ended-2023 #310: Add new Pvs to Nexus files

jilavsky, 2019-11-06:
I added new Sample PVs based on Nexus definition. These need to be added to Nexus files stored by USAXS instrument (when running spec or BS). Also, BS will need to get support for these as metadata. Need to discuss. 

This is code used to clear them, so this contain all needed information

'''
variables:
caput 9idcSample:Thickness      1		2>&1 > /dev/null
caput 9idcSample:Temperature   25		2>&1 > /dev/null
caput 9idcSample:Concentration  1		2>&1 > /dev/null
caput 9idcSample:VolumeFraction 1		2>&1 > /dev/null
caput 9idcSample:ScatteringLengthDensity 1	2>&1 > /dev/null
caput 9idcSample:MagneticField  0		2>&1 > /dev/null
caput 9idcSample:StressField    0		2>&1 > /dev/null
caput 9idcSample:ElectricField  0		2>&1 > /dev/null
caput 9idcSample:XTranslation   0		2>&1 > /dev/null
caput 9idcSample:RotationAngle  0		2>&1 > /dev/null

Strings with x/y/z values (and associated numerical values 0/1/2):
caput 9idcSample:MagneticFieldDir  0		2>&1 > /dev/null
caput 9idcSample:StressFieldDir    0		2>&1 > /dev/null
caput 9idcSample:ElectricFieldDir  0		2>&1 > /dev/null

Strings:
caput -S  9idcSample:Name 		""		2>&1 > /dev/null
caput -S  9idcSample:ShortTitle 	""		2>&1 > /dev/null
caput -S  9idcSample:Description 	""		2>&1 > /dev/null
caput -S  9idcSample:ChemicalFormula ""	2>&1 > /dev/null
'''

prjemian, 2019-11-19:
bluesky:

```
In [4]: sample_data.magnetic_field_dir.enum_strs
Out[4]: ('X', 'Y', 'Z')

In [5]: sample_data.stress_field_dir.enum_strs
Out[5]: ('X', 'Y', 'Z')

In [6]: sample_data.electric_field_dir.enum_strs
Out[6]: ('X', 'Y', 'Z')

In [7]: mov sample_data.stress_field_dir "Y"

In [8]: sample_data.stress_field_dir.value
Out[8]: 'Y'

In [9]: mov sample_data.stress_field_dir 0

In [10]: sample_data.stress_field_dir.value
Out[10]: 'X'

In [14]: print(APS_utils.device_read2table(sample_data))
===================================== ===== ==========================
name                                  value timestamp
===================================== ===== ==========================
sample_data_temperature               25.0  2019-11-11 14:03:37.820430
sample_data_concentration             1.0   2019-11-11 14:03:37.857035
sample_data_volume_fraction           1.0   2019-11-11 14:03:37.928887
sample_data_scattering_length_density 1.0   2019-11-11 14:03:37.964712
sample_data_magnetic_field            0.0   2019-11-11 14:03:38.000694
sample_data_stress_field              0.0   2019-11-11 14:03:38.036502
sample_data_electric_field            0.0   2019-11-11 14:03:38.072280
sample_data_x_translation             0.0   2019-11-11 14:03:38.216859
sample_data_rotation_angle            0.0   2019-11-11 14:03:38.253041
sample_data_magnetic_field_dir        X     2019-11-11 14:03:38.108230
sample_data_stress_field_dir          X     2019-11-19 15:09:02.882901
sample_data_electric_field_dir        X     2019-11-11 14:03:38.180695
sample_data_description                     2019-11-11 14:03:38.289089
sample_data_chemical_formula                2019-11-11 14:03:38.325328
===================================== ===== ==========================

```

prjemian, 2019-11-19:
Example changing the description (can't use `mov` here for some reason):

```
In [16]: sample_data.description.put("this is a test")
```

Display the sample data:

```
In [17]: print(APS_utils.device_read2table(sample_data))
===================================== ============== ==========================
name                                  value          timestamp
===================================== ============== ==========================
sample_data_temperature               25.0           2019-11-11 14:03:37.820430
sample_data_concentration             1.0            2019-11-11 14:03:37.857035
sample_data_volume_fraction           1.0            2019-11-11 14:03:37.928887
sample_data_scattering_length_density 1.0            2019-11-11 14:03:37.964712
sample_data_magnetic_field            0.0            2019-11-11 14:03:38.000694
sample_data_stress_field              0.0            2019-11-11 14:03:38.036502
sample_data_electric_field            0.0            2019-11-11 14:03:38.072280
sample_data_x_translation             0.0            2019-11-11 14:03:38.216859
sample_data_rotation_angle            0.0            2019-11-11 14:03:38.253041
sample_data_magnetic_field_dir        X              2019-11-11 14:03:38.108230
sample_data_stress_field_dir          X              2019-11-19 15:09:02.882901
sample_data_electric_field_dir        X              2019-11-11 14:03:38.180695
sample_data_description               this is a test 2019-11-19 15:11:30.978114
sample_data_chemical_formula                         2019-11-11 14:03:38.325328
===================================== ============== ==========================
```

*resetAll()* (clear) : reset all PVs back to preset values

```
In [18]: RE(sample_data.resetAll())
Out[18]: ()
```

Display again:

```
In [19]: print(APS_utils.device_read2table(sample_data))
===================================== ===== ==========================
name                                  value timestamp
===================================== ===== ==========================
sample_data_temperature               25.0  2019-11-19 15:11:50.854071
sample_data_concentration             1.0   2019-11-19 15:11:50.852394
sample_data_volume_fraction           1.0   2019-11-19 15:11:50.853157
sample_data_scattering_length_density 1.0   2019-11-19 15:11:50.856137
sample_data_magnetic_field            0.0   2019-11-19 15:11:50.851244
sample_data_stress_field              0.0   2019-11-19 15:11:50.855555
sample_data_electric_field            0.0   2019-11-19 15:11:50.854066
sample_data_x_translation             0.0   2019-11-19 15:11:50.862302
sample_data_rotation_angle            0.0   2019-11-19 15:11:50.865528
sample_data_magnetic_field_dir        X     2019-11-19 15:11:50.864339
sample_data_stress_field_dir          X     2019-11-19 15:11:50.870363
sample_data_electric_field_dir        X     2019-11-19 15:11:50.875617
sample_data_description                     2019-11-19 15:11:50.871322
sample_data_chemical_formula                2019-11-19 15:11:50.879635
===================================== ===== ==========================
```

prjemian, 2019-11-19:
@jilavsky : Units (`.EGU` where appropriate), too?

prjemian, 2019-11-19:
These were added as regular bluesky data, baseline measurements (recorded at beginning and end of every bluesky scan).  Need them added to fly scan HDF5 file.

prjemian, 2019-11-19:
These all go into various field of the NeXus [NXsample](http://download.nexusformat.org/doc/html/classes/base_classes/NXsample.html#nxsample) group.

prjemian, 2019-11-20:
Added identical changes (as 915aad1) to file used by SPEC: `/home/beams/USAXS/Documents/eclipse/USAXS/tools/saveFlyData.xml`.

closed 2019-11-20 by prjemian