APS-USAXS/usaxs-bluesky-ended-2023 #314: Post properly command file for web page

jilavsky, 2019-11-30:
This may be due to change in web page. Currently, web page is looking at https://usaxslive.xray.aps.anl.gov/specmacro.txt to show currently running command file. spec posts this correctly, BS not.

prjemian, 2019-11-30:
Time to rename it on the web page to `command_file.txt` which makes it generic for how it got there.  With this change, the name accurately represents the intent of this feature pf the livedata page.

HISTORICAL NOTE

This file was originally named `specmacro.txt` and the content of the file that was named in SPEC with the `qdo` command was copied to this file.  To avoid filling the livedata web page directory with a history of user macro file names, the macro file content was copied into the file `specmacro.txt` that is described as the command file most recently started.

prjemian, 2020-02-01:
Since a [rename on the web page](https://github.com/APS-USAXS/ipython-usaxs/issues/314#issuecomment-560019334) involves other software, first fix the original problem within bluesky.

prjemian, 2020-02-01:
investigating:

```
usaxs@usaxscontrol ~/.ipython $ git grep specmacro
profile_bluesky/startup/ref/macros/local/usaxs_commands.mac:  theTarget = "/tmp/specmacro.txt"
```

That directory is for reference only, not used by bluesky.  (Might be time to delete it since it was for development of the bluesky code base here.)

Might not need a code fix in the bluesky software.  Look to the livedata software.  Perhaps move the issue there?