APS-USAXS/usaxs-bluesky-ended-2023 #316: After BS is quit, command line is unresponsive and behaves weird

jilavsky, 2019-11-30:
Quit BlueSky, command line does not echo keyboard input, but seems to properly execute commands. Something is weird on GUI in terminal?
below I typed lots of ls:
```
(/share1/bluesky/conda_envs/bluesky) usaxs@usaxscontrol .../USAXS_data/2019-12 $ 
```
but they were not echoed - yet were execute when I typed enter:
```
(/share1/bluesky/conda_envs/bluesky) usaxs@usaxscontrol .../USAXS_data/2019-12 $ bash: llllll: command not found
```

jilavsky, 2019-12-02:
Some program (possibly bluesky session) has sent the magic (non-displayable) characters to the console, telling it to not echo what the user types and also not to print the line feed to the terminal.  Just awful display.  Applying the terminal's *Reset and clear* command does not help.

This problem has been seen in other linux or unix versions from other software that long predates bluesky.  Needs some in-depth search on stackoverflow or somewhere.

prjemian, 2020-05-22:
We're seeing this same behavior (command line unresponsive) at 8-ID-I when startup fails with an error from Qt: https://github.com/aps-8id-dys/ipython-8idiuser/issues/156

Actually, the command line executes commands that are typed but does not show keyboard input.  Limited to the bash shell.  Exit the shell or start ipython and keyboard input shows up again.

prjemian, 2020-05-22:
I believe there is nothing we can do about this in our bluesky installation or configuration.  Rather, it is a problem for IT to resolve.

closed 2020-05-22 by prjemian