APS-USAXS/usaxs-bluesky-ended-2023 #317: pre tune logic for S/WAXS v all of it

jilavsky, 2019-12-02:
visit the logic of when to call tune SWAXS v. tune all of it

Jan has done some work in the past few days.  Organize that into this property: https://github.com/APS-USAXS/ipython-usaxs/blob/15320b452d0777abd1898615a01415e382f0a2e0/profile_bluesky/startup/10-devices.py#L358-L376

jilavsky, 2019-12-02:
Jan's code was added to `50-plans.before_plan()` as:

```
-        yield from preUSAXStune(md=md)
+        # select between positions 
+        mode_now = terms.SAXS.UsaxsSaxsMode.get(as_string=True)
+        if mode_now == "USAXS in beam" : 
+            yield from preUSAXStune(md=md)
+        else:
+            yield from preSWAXStune(md=md)
+
```

jilavsky, 2019-12-02:
`preUSAXStune()` must be run if the checkbox for `9idcLAX:USAXS:RunPreUSAXStuneNext` has been checked (thus the PV is 1).

jilavsky, 2019-12-02:
Look at both tune plans and make sure they are tidy as we want

prjemian, 2020-02-01:
note:  Running Fly, SAXS, WAXS today and periodic tunes seem appropriate.  Periodic tune during SAXS was just M stage, update frequency was as expected, periodic tune during Fly USAXS was as expected.

We'll check this [feature](https://github.com/APS-USAXS/ipython-usaxs/issues/317#issuecomment-560492851) (`9idcLAX:USAXS:RunPreUSAXStuneNext `) today.

prjemian, 2020-02-01:
The tune next for SAXS/WAXS works.  Will test the one for USAXS soon...

prjemian, 2020-02-03:
@jilavsky reports this works for USAXS.

closed 2020-02-03 by prjemian