APS-USAXS/usaxs-bluesky-ended-2023 #320: Linkam support needs to be consistent

jilavsky, 2019-12-02:
We have two Linkam controllers. One has set_rate the other has ramp_rate as input values to set heating/cooling rate. We need this consistent.

prjemian, 2019-12-03:
@kmpeters :

* CI94 has both `9idcLAX:ci94:setRate` and `9idcLAX:ci94:setSpeed` PVs, these look like they would set the same physical term.  What's the difference?
* T96 has `9idcLINKAM:tc1:rampRate` which sets the comparable value for which of the above two PVs?

@jilavsky :

* 'rampRate` is redundant - can I name this signal (in bluesky) as `rate`?  For all temperature controllers?

jilavsky, 2019-12-03:
We should simply use "rate" in BlueSky. No need to use PV names, confuses the situation here. 

speed-vs-rate: 
ci94 "speed" is speed (in rpm) of the LN2 pump, different value than rate (which is change in temperature per minute).  Speed of 0 means automatic control. 
T96 rampRate sets "rate" - temperature change per minute.

prjemian, 2019-12-03:
Good.  Agree with using `rate` as the bluesky name.  So for the ci94, the speed is a pump speed.

prjemian, 2019-12-31:
note:

```
(base) mintadmin@mint-vm:~/.../eclipse/ipython-usaxs$ git grep set_rate
examples/custom_loop.md:    yield from bps.mv(linkam.set_rate, 100)                     # degrees C/minute
examples/use_linkam.md:    yield from bps.mv(linkam_tc1.set_rate, 100)            # degrees C/minute
profile_bluesky/startup/10-devices.py:    set_rate = Component(EpicsSignal, "setRate", kind="omitted")
user/linkam.py:    yield from bps.mv(linkam.set_rate, rate1)          #sets the rate of next ramp
user/linkam.py:    yield from bps.mv(linkam.set_rate, rate2)          #sets the rate of next ramp
(base) mintadmin@mint-vm:~/.../eclipse/ipython-usaxs$ git grep ramp_rate
profile_bluesky/startup/10-devices.py:    ramp_rate = Component(EpicsSignalWithRBV, "rampRate", kind="omitted")
```

prjemian, 2019-12-31:
>     * For all temperature controllers?

We can make this happen for all *USAXS* temperature controllers in `UsaxsProcessController`, but we must establish and adhere to a pattern.  Here is the *family tree*:

* `ophyd.Device`
  * `apstools.devices.ProcessController`
    * `10-devices.UsaxsProcessController`
      * `10-devices.Linkam_CI94`
      * `10-devices.Linkam_T96`

We'll add an `ophyd.Signal` to `UsaxsProcessController` for both `rate` and `speed`, along with usage comments.  These should be replaced in subclasses (`Linkam_CI94` and `Linkam_T96`) with `EpicsSignal` as appropiate.

commit 8517c59c (2019-12-31): fixes #320
closed 2019-12-31 by prjemian