APS-USAXS/usaxs-bluesky-ended-2023 #335: review this test, it is a duplciate

prjemian, 2020-03-09:
https://github.com/APS-USAXS/ipython-usaxs/blob/a92e08c587110dcf2c566b3baac750851531177e/profile_bluesky/startup/instrument/devices/diagnostics.py#L49

jilavsky, 2020-05-28:
Did we wanted to check also on chain_B here? In which case the line should be changed  to self.c_shutter_closed_chain_B.get() == "OFF" so we are checking on both chains.

prjemian, 2020-05-28:
You are correct.  That's a copy/paste error.

prjemian, 2020-05-28:
Fix it here on Github (easy for me now) or at the beam line (you fix it)?

prjemian, 2020-05-28:
I'll fix it at beam line now.

```
        chain_A = self.c_shutter_closed_chain_A
        chain_B = self.c_shutter_closed_chain_B
        enabled = chain_A.get() == "OFF" or chain_B.get() == "OFF"
```

The `chain_A.get()` is done within the implied `if` so that if the first test succeeds, the second test is not used and it is not necessary to call `chain_B.get()`

commit 2b73267f (2020-05-28): fixes #335
closed 2020-05-28 by prjemian