APS-USAXS/usaxs-bluesky-ended-2023 #337: When perUSXStune fails to find peak, it stops

jilavsky, 2020-06-01:
While testing fix for bug in code without beam I run preUSAXStune. This is 3 tune macros sequentially. 

Interesting observation. 
1.	I run preUSAXStune. 
2.	tune_mr failed. (no X-rays)
preUSAXStune stopped, did not continue. 

This is different behavior that in spec. Is this BlueSky behavior or did we built it in? We need to rethink the proper procedure here, since preUSAXStune failing on one or more tunes is relatively common (multiple scattering, tune_ar) and stopping data collection would be disatster. I believ eit needs to continue. 
However, argumentatively, we should have bit more smarts here and try to make do some more action. 

Also, failed tune on spec sends e-mail. I did nto get anything from BlueSky.

prjemian, 2020-06-02:
An error dump would be most informative to point to the exact piece of code that triggered this exception.  I thought we had an option for this, to ignore the error (just report it to the console) or stop the scan.

prjemian, 2020-06-02:
Since we have no beam, this should be reproducible, same error.

jilavsky, 2020-06-02:
There is no error message, it gave !!! tune failed for axis (axis_name} !!! which suggest this si planned behavior.

prjemian, 2020-06-02:
```

In [1]: RE(preUSAXStune())                                                                                                                                    
                                                                                                                                                              
I Tue-15:42:25 - Preparing for USAXS mode ... please wait ...
                                                                                                                                                              
I Tue-15:42:26 - tuning axis: m_stage_r
                                                                                                                                                              
D Tue-15:42:26 - gain=1000000000000.0  rate: 20.0  max: 950000  converged=[True, True]
D Tue-15:42:26 - gain=1000000000.0  rate: 1960.0  max: 950000  converged=[True, True, True, True]
D Tue-15:42:26 - gain=1000000000.0  rate: 3420.0  max: 950000  converged=[True, True, True, True, True, True]
D Tue-15:42:26 - converged: [True, True, True, True, True, True]
I Tue-15:42:26 - Tuning axis m_stage_r, current position is 9.051535

Transient Scan ID: 8     Time: 2020-06-02 15:42:27
Persistent Unique Scan ID: 'b824957b-4dc7-43b9-ba94-47cfa56656d2'
New stream: 'baseline'
New stream: 'primary'                                                                                                                                         
+-----------+------------+------------+------------+                                                                                                          
|   seq_num |       time |  m_stage_r |   I0_USAXS |
+-----------+------------+------------+------------+
|         1 | 15:42:28.0 |  9.0502850 |        195 |
|         2 | 15:42:28.6 |  9.0503680 |        193 |
|         3 | 15:42:29.0 |  9.0504510 |        194 |                                                                                                          
|         4 | 15:42:29.5 |  9.0505350 |        194 |                                                                                                          
|         5 | 15:42:29.9 |  9.0506180 |        194 |                                                                                                          
|         6 | 15:42:30.3 |  9.0507010 |        195 |                                                                                                          
|         7 | 15:42:30.7 |  9.0507850 |        194 |                                                                                                          
|         8 | 15:42:31.1 |  9.0508680 |        194 |                                                                                                          
|         9 | 15:42:31.5 |  9.0509510 |        194 |                                                                                                          
|        10 | 15:42:31.9 |  9.0510350 |        192 |                                                                                                          
|        11 | 15:42:32.3 |  9.0511180 |        194 |                                                                                                          
|        12 | 15:42:32.8 |  9.0512010 |        194 |                                                                                                          
|        13 | 15:42:33.2 |  9.0512850 |        195 |                                                                                                          
|        14 | 15:42:33.6 |  9.0513680 |        195 |                                                                                                          
|        15 | 15:42:34.0 |  9.0514510 |        195 |                                                                                                          
|        16 | 15:42:34.4 |  9.0515350 |        194 |                                                                                                          
|        17 | 15:42:34.8 |  9.0516180 |        194 |                                                                                                          
|        18 | 15:42:35.3 |  9.0517010 |        195 |                                                                                                          
|        19 | 15:42:35.7 |  9.0517850 |        195 |                                                                                                          
|        20 | 15:42:36.1 |  9.0518680 |        194 |                                                                                                          
|        21 | 15:42:36.5 |  9.0519510 |        193 |                                                                                                          
|        22 | 15:42:36.9 |  9.0520350 |        194 |                                                                                                          
|        23 | 15:42:37.3 |  9.0521180 |        194 |                                                                                                          
|        24 | 15:42:37.7 |  9.0522010 |        193 |                                                                                                          
|        25 | 15:42:38.2 |  9.0522850 |        194 |                                                                                                          
[... cut ...]
                                                                                                                                                              
I Tue-15:42:41 - ymax/yman not big enough: is it a peak?
I Tue-15:42:41 - axis: m_stage_r
I Tue-15:42:41 - starting position: 9.051535
I Tue-15:42:41 - peak detected: False
I Tue-15:42:41 - final position: 9.051535
W Tue-15:42:41 - !!! tune failed for axis {axis.name} !!!
Out[1]: ('b824957b-4dc7-43b9-ba94-47cfa56656d2',)saxs_time.get()} second(s)

```

commit 40a90916 (2020-06-02): fixes #337
closed 2020-06-02 by prjemian