APS-USAXS/usaxs-bluesky-ended-2023 #346: extend the command_list a capability to run a config file

prjemian, 2020-06-18:
add to the command_list support so that it can run a config file (that installs a temporary set of changes -- from a defined list of things allowed to be changed) for the duration of the command list

prjemian, 2020-09-10:
@jilavsky 
Can you provide an example of a list of activities that a config file would describe?  I'll use that as a starting point here.

jilavsky, 2020-09-10:
Look at USAXS.conf and USAXS_user_macros.mac files in spec. There are two main types of changes which are set there:
1. User Macro definitions. This is something we talked about when we need to add user defined functions, there is solution which I recall we looked at. This is needed when user is developing their data collection plan (e.g., Linkam) and need to change existing one.  
2. Parameters definitions which are normally defined with their default values, but may need to change. Below are examples from spec, which are currently written in BS code. We should be able to overwrite individually, if for some reason necessary. The most typical would be the ar_tune_range which I noticed was aliased to some motor width value now. Note the general split personality here - some exist in epics, but their default values are set in spec to guarantee they get reset (on restart) for users if they change them.  

  SAXS_Tr_Time            =    3    # how long to measure transmission for
  SAXS_PD_Yoffset         =    11    # measured on 6/13/2020 JIL; 
  #USAXS transmission
  set_USAXS_MEASURE_PIN_TRANS    1     # default is no = 0
  set_USAXSPinT_MeasurementTime  3     # seconds, default time
  __tmpCalcValue67 =  (AY0  + 8 + 12*sin(AR_VAL_CENTER * 3.1415/180))
  set_USAXSPinT_AyPosition  __tmpCalcValue67
  set_USAXS_tune_dx_range 6
  set_USAXS_tune_dy_range 6
  # these numbers may change. set based on energies
  __CurrentSystemEn=A[en]

  set_USAXS_tune_mr_range   0.004    # range for tune mr for Si 220 crystals, this is energy independent
  if(__CurrentSystemEn<10.99){  
    set_USAXS_tune_ar_range    0.002    # range for tune ar for about 10keV 
    set_USAXS_tune_m2rp_range  10        # range for tune m2rp for about 10keV
    set_USAXS_tune_a2rp_range  7        # range for tune a2rp for about 10keV
    set_USAXS_tune_msrp_range   5        # range for tune msr for about 10keV
    set_USAXS_tune_asrp_range   3        # range for tune asr for about 10keV 
    set_USAXS_MINSTEP 0.000045
  }
  if(__CurrentSystemEn>10.99 && __CurrentSystemEn<12.99){  
 	### these are for ~12keV
     set_USAXS_tune_mr_range    0.004    # range for tune mr for Si 220 crystals
     set_USAXS_tune_ar_range    0.0020   # range for tune ar for about 12keV 
     set_USAXS_tune_m2rp_range  9        # range for tune m2rp for about 12keV
     set_USAXS_tune_a2rp_range  5        # range for tune a2rp for about 12keV - Prague
     set_USAXS_tune_msrp_range   3        # range for tune msr for about 12keV
     set_USAXS_tune_asrp_range   3      # range for tune asr for about 12keV 
     set_USAXS_MINSTEP 0.000038
  }
  if(__CurrentSystemEn>13 && __CurrentSystemEn<18.099){  
	### these are for 16-18keV
    set_USAXS_tune_mr_range    0.003    # range for tune mr for Si 220 crystals
    set_USAXS_tune_ar_range    0.0018   # range for tune ar for about 17keV 
    set_USAXS_tune_m2rp_range  8        # range for tune m2rp for about 17keV
    set_USAXS_tune_a2rp_range  5        # range for tune a2rp for about 17keV - Prague
    set_USAXS_tune_msrp_range   3        # range for tune msr for about 17keV
    set_USAXS_tune_asrp_range   3        # range for tune asr for about 17keV 
    set_USAXS_MINSTEP 0.000038
 }  
  if(__CurrentSystemEn>18.1 && __CurrentSystemEn<20.799){  
	### these are for 18-21keV, assume 220
    set_USAXS_tune_mr_range     0.0025    # range for tune mr for Si 220 crystals
    set_USAXS_tune_ar_range     0.0016     # range for tune ar for about 21keV 
    set_USAXS_tune_m2rp_range   8        # range for tune m2rp for about 21keV
    set_USAXS_tune_a2rp_range   5        # range for tune a2rp for about 21keV - Prague
    set_USAXS_tune_msrp_range   3        # range for tune msr for about 21keV
    set_USAXS_tune_asrp_range   3        # range for tune asr for about 21keV 
    set_USAXS_MINSTEP 0.000036
 }
  if(__CurrentSystemEn>20.8){  
	### these are for 21 and higherkeV, assume 220
    set_USAXS_tune_mr_range    0.0025    # range for tune mr for Si 220 crystals
    set_USAXS_tune_ar_range    0.0013     # range for tune ar for about 21keV 
    set_USAXS_tune_m2rp_range  9        # range for tune m2rp for about 21keV
    set_USAXS_tune_a2rp_range  5        # range for tune a2rp for about 21keV - Prague
    set_USAXS_tune_msrp_range   3        # range for tune msr for about 21keV
    set_USAXS_tune_asrp_range   3        # range for tune asr for about 21keV 
    set_USAXS_MINSTEP 0.000034
 }

prjemian, 2020-11-18:
This could include the configuration from #428.

prjemian, 2020-11-23:
We can take advantage of the code that can load a local python code file.  The `%run` (an IPython magic command) can be run by:

```python
from IPython import get_ipython
get_ipython().run_line_magic('run', '-i ./local_file.py')
```

The *config file* would be a local python file that has whatever configuration commands desired, written as non-plans.

Yet to be proven this works.

prjemian, 2020-11-23:
Implementing the `%run` feature is justified since one requirement is to modify existing plans or add new ones.

PR #441 (2021-01-26): run python file from command_list
fixes #346

Needs testing!
closed 2021-01-26 by jilavsky