APS-USAXS/usaxs-bluesky-ended-2023 #352: usaxs Nexus files for flyscan & stepscan

jilavsky, 2020-07-05:
@prjemian 
Need to create easy to handle system of Nexus files for USAXS. 
That bluesky Nexus writer does really good job in saving nearly excessive amount of metadata. Even though, files are 2x larger.  Still, very nice. Working on Igor support. 

### Few requests/notes:
1. Change extension of saved file to h5 same as existing flyscan from spec. Can we add writing Nexus files with correct name to correct location for BS step scan and test it July 27 or 28? 

2. To distinguish between existing files, need easy way to identify. Is this correct observation:
BS step scan:
**entry/plan_name = uascan
entry/program_name = bluesky**
BS flyscan:
**entry/plan_name - does not exist
entry/program_name = saveFlyData.py**
spec flyscan:
**entry/plan_name - does not exist
entry/program_name = saveFlyData.py**
Seems in this case both spec and BS use same writer, in which case this is correct.  For completeness, can we add 
**entry/plan_name = flyscan**
to existing Nexus files in saveFlyData.py, please? 

3. Flyscan data file could improve by adding more metadata in line with what step scan in BS now has.  In which case suggest to use:
BS flyscan:
**entry/plan_name = flyscan
entry/program_name = bluesky**
so I can distinguish in Igor code. 

Or suggest better way. Do we want to add version number there?

jilavsky, 2020-07-06:
**Metadata which are missing and are useful or needed:**
"SPEC_data_file" file name.	
"sample name" in entry/sample. Should contain same thing as entry:title. 
"sample thickness", critically needed. 

**all transmission measurements PVs, recorded in flyscan here :** 
:entry:metadata:trans_I0_counts
:entry:metadata:trans_I0_gain
:entry:metadata:trans_pin_aypos
:entry:metadata:trans_pin_counts
:entry:metadata:trans_pin_gain
 :entry:metadata:trans_pin_time

**all of the diode values:**
"upd size", needed.
:entry:metadata:upd_gain0
:entry:metadata:upd_gain1
:entry:metadata:upd_gain2
:entry:metadata:upd_gain3
:entry:metadata:upd_gain4	
:entry:metadata:upd_bkg0
:entry:metadata:upd_bkg1
:entry:metadata:upd_bkg2
:entry:metadata:upd_bkg3
:entry:metadata:upd_bkg4	
:entry:metadata:upd_bkg_err0
:entry:metadata:upd_bkgErr1
:entry:metadata:upd_bkgErr2
:entry:metadata:upd_bkgErr3
:entry:metadata:upd_bkgErr4

prjemian, 2020-07-06:
I will divide this issue into separate issues (referencing this one).  Much easier to determine when a specific need has been satisfied that way.

Once all new issues are created, I'll close this one.

jilavsky, 2020-07-06:
@prjemian Can you eventually regenerate the nexus files so I can finish the reader, please? 
20200610-135050-S00049-78b4801.hdf
20200618-105527-S00029-355d8b5.hdf
20200618-121245-S00020-e904886.hdf

should be enough.

prjemian, 2020-07-06:
All requests converted to separate issues.  Closing this one.

closed 2020-07-06 by prjemian