APS-USAXS/usaxs-bluesky-ended-2023 #356: additional metadata

prjemian, 2020-07-06:
as noted in #352:

Metadata which are missing and are useful or needed:

* [ ] `SPEC_data_file` file name.
* [ ] `sample name` in `/entry/sample`. Should contain same thing as `/entry/title`.
* [ ] `sample thickness` -- critically needed.

jilavsky, 2020-07-06:
Please add attribute config_version to entry/program_name
We can start with version 1.0
reminder, saveFlyData.py is currently at version 1.2. This is important for future updates to bluesky as program_name.

prjemian, 2020-07-24:
File **creator version** currently written as file root attribute `creator_version`:  https://github.com/APS-USAXS/ipython-usaxs/blob/8c9ab660bbcf99eea7021c2b1035964eabdd15de/profile_bluesky/startup/usaxs_support/saveFlyData.py#L48 https://github.com/APS-USAXS/ipython-usaxs/blob/8c9ab660bbcf99eea7021c2b1035964eabdd15de/profile_bluesky/startup/usaxs_support/saveFlyData.py#L206

prjemian, 2020-07-24:
https://github.com/APS-USAXS/ipython-usaxs/blob/8c9ab660bbcf99eea7021c2b1035964eabdd15de/profile_bluesky/startup/usaxs_support/saveFlyData.xml#L46-L50

prjemian, 2020-07-24:
But note this interesting comment: https://github.com/APS-USAXS/ipython-usaxs/blob/8c9ab660bbcf99eea7021c2b1035964eabdd15de/profile_bluesky/startup/usaxs_support/saveFlyData.xml#L4-L9

prjemian, 2020-07-25:
@jilavsky -- What does it mean to add `SPEC_data_file` to a NeXus file for data collected by Bluesky?

prjemian, 2020-07-25:
```
    sample:NXsample
      name --> /entry/instrument/bluesky/streams/baseline/user_data_sample_title/value_start
      thickness --> /entry/instrument/bluesky/streams/baseline/user_data_sample_thickness/value_start
```

`/entry/sample/name` same link as `/entry/title`:

```
    title --> /entry/instrument/bluesky/streams/baseline/user_data_sample_title/value_start
```

prjemian, 2020-07-25:
New file name: `20200610-135050-S00049-uascan-78b4801.h5` for `db["78b4801c"]`

jilavsky, 2020-07-25:
SPEC_data_file is name of the "spec" data file we are writing notes in. the one created by "NewSpecFile" command in BS. 
In soon-to-come future this will be related to user name. After this run we will change arrangement of data in 2020-09/09_21_UserName/09_21_UserName_usaxs (_saxs, _waxs). NX file needs to have "09_21_UserName" in there - and historically, this has been name of the "spec" file. I'll submit this as issue for the future so do not forget.

prjemian, 2020-07-27:
`SPEC_data_file` is named `user_data_spec_file`: `/entry/instrument/bluesky/streams/baseline/user_data_spec_file/value`

commit 9d9f89f6 (2020-07-27): fixes #356
closed 2020-07-27 by prjemian