APS-USAXS/usaxs-bluesky-ended-2023 #357: regenerate these NeXus files using new format and metadata

prjemian, 2020-07-06:
As noted in #352:

> Can you eventually regenerate the nexus files so I can finish the reader, please?

* 20200610-135050-S00049-78b4801.hdf
* 20200618-105527-S00029-355d8b5.hdf
* 20200618-121245-S00020-e904886.hdf

should be enough.

prjemian, 2020-07-06:
With `.h5` extension per #354, right?  And with additional metadata per #355, #356, right?

jilavsky, 2020-07-06:
Yes, with all changes if possible.

prjemian, 2020-07-25:
@jilavsky -- Here is everything but the `SPEC_data_file`:

```
-rw-rw-r--   1 usaxs        usaxs  2.3M Jul 24 23:46 20200610-135050-S00049-uascan-78b4801.h5
-rw-rw-r--   1 usaxs        usaxs  2.3M Jul 24 23:50 20200618-105527-S00029-uascan-355d8b5.h5
-rw-rw-r--   1 usaxs        usaxs  2.3M Jul 24 23:50 20200618-121245-S00020-uascan-e904886.h5
usaxs@usaxscontrol /share1/USAXS_data $ pwd
/share1/USAXS_data
```

prjemian, 2020-07-25:
Here's how the files were regenerated:

```
bash $ blueskyUSAXS
# ... startup ...
from instrument.callbacks.nxwriter_usaxs import NXWriterUascan

my_writer = NXWriterUascan()

run= db["78b4801c"]
replay(run, callback=my_writer.receiver)

run = db["355d8b5"]
replay(run, callback=my_writer.receiver)

run = db["e904886"]
replay(run, callback=my_writer.receiver)
```

jilavsky, 2020-07-26:
I am unable to find new metadata there. Can you provide locations?

prjemian, 2020-07-26:
To get new metadata, don't we need to collect it?  The NX writer works on
info it gets from the databroker.

prjemian, 2020-07-26:
If I understand, you found the new ,h5 files but did not find the metadata
in them, right?

prjemian, 2020-07-27:
revision on how the files were regenerated:

```
bash $ blueskyUSAXS
# ... startup ...
from instrument.callbacks.nxwriter_usaxs import NXWriterUascan

my_writer = NXWriterUascan()

for uid in "".split():
    replay(db[uid], callback=my_writer.receiver)
```

prjemian, 2020-07-27:
Here's a check of the new content:

```
import pyRestTable, h5py
addresses = []
addresses.append("/entry/SPEC_data_file")
addresses.append("/entry/program_name")
addresses.append("/entry/sample/name")
addresses.append("/entry/sample/thickness")
files = []
files.append("20200610-135050-S00049-uascan-78b4801.h5")
files.append("20200618-105527-S00029-uascan-355d8b5.h5")
files.append("20200618-121245-S00020-uascan-e904886.h5")
table = pyRestTable.Table()
table.labels = "file address value".split()
for fname in files:
    with h5py.File(fname, "r") as nx: 
        for addr in addresses: 
            table.addRow((fname, addr, nx[addr][()]))
print(table)
```

Here is the output:

```
======================================== ======================= ============================
file                                     address                 value                       
======================================== ======================= ============================
20200610-135050-S00049-uascan-78b4801.h5 /entry/SPEC_data_file   b'06_10_StepScans.dat'      
20200610-135050-S00049-uascan-78b4801.h5 /entry/program_name     bluesky                     
20200610-135050-S00049-uascan-78b4801.h5 /entry/sample/name      b'Blank'                    
20200610-135050-S00049-uascan-78b4801.h5 /entry/sample/thickness 1.0                         
20200618-105527-S00029-uascan-355d8b5.h5 /entry/SPEC_data_file   b'06_18_test.dat'           
20200618-105527-S00029-uascan-355d8b5.h5 /entry/program_name     bluesky                     
20200618-105527-S00029-uascan-355d8b5.h5 /entry/sample/name      b'AirBlank-Step'            
20200618-105527-S00029-uascan-355d8b5.h5 /entry/sample/thickness 0.0                         
20200618-121245-S00020-uascan-e904886.h5 /entry/SPEC_data_file   b'06_18_testAutoCollect.dat'
20200618-121245-S00020-uascan-e904886.h5 /entry/program_name     bluesky                     
20200618-121245-S00020-uascan-e904886.h5 /entry/sample/name      b'GlassyCarbonM5-Step'      
20200618-121245-S00020-uascan-e904886.h5 /entry/sample/thickness 1.0                         
======================================== ======================= ============================
```

prjemian, 2020-07-27:
Files re-generated now.  Confirm you find the content.

prjemian, 2020-07-27:
tree structure of `20200610-135050-S00049-uascan-78b4801.h5` (all 2,493 lines of it)

<details>

```
/share1/USAXS_data/2020-06/20200610-135050-S00049-uascan-78b4801.h5 : NeXus data file
  entry:NXentry
    SPEC_data_file --> /entry/instrument/bluesky/streams/baseline/user_data_spec_file/value_start
    duration:NX_FLOAT64[] = [ ... ]
    end_time:NX_CHAR = 2020-06-10T13:55:39.559547
    entry_identifier --> /entry/instrument/bluesky/metadata/run_start_uid
    plan_name --> /entry/instrument/bluesky/metadata/plan_name
    program_name:NX_CHAR = bluesky
    run_cycle --> /entry/instrument/bluesky/streams/baseline/aps_aps_cycle/value_start
    start_time:NX_CHAR = 2020-06-10T13:50:50.239672
    title --> /entry/instrument/bluesky/streams/baseline/user_data_sample_title/value_start
    contact:NXuser
      affiliation --> /entry/instrument/bluesky/streams/baseline/bss_user_info_institution/value_start
      email --> /entry/instrument/bluesky/streams/baseline/bss_user_info_email/value_start
      facility_user_id --> /entry/instrument/bluesky/streams/baseline/bss_user_info_badge/value_start
      name --> /entry/instrument/bluesky/streams/baseline/bss_user_info_contact/value_start
      role:NX_CHAR = contact
    data:NXdata
      EPOCH --> /entry/instrument/bluesky/streams/primary/upd_autorange_controls_reqrange/time
      I00_autorange_controls_gain --> /entry/instrument/bluesky/streams/primary/I00_autorange_controls_gain/value
      I00_autorange_controls_reqrange --> /entry/instrument/bluesky/streams/primary/I00_autorange_controls_reqrange/value
      I0_USAXS --> /entry/instrument/bluesky/streams/primary/I0_USAXS/value
      I0_autorange_controls_gain --> /entry/instrument/bluesky/streams/primary/I0_autorange_controls_gain/value
      I0_autorange_controls_reqrange --> /entry/instrument/bluesky/streams/primary/I0_autorange_controls_reqrange/value
      PD_USAXS --> /entry/instrument/bluesky/streams/primary/PD_USAXS/value
      a_stage_r --> /entry/instrument/bluesky/streams/primary/a_stage_r/value
      a_stage_y --> /entry/instrument/bluesky/streams/primary/a_stage_y/value
      d_stage_y --> /entry/instrument/bluesky/streams/primary/d_stage_y/value
      m_stage_r --> /entry/instrument/bluesky/streams/primary/m_stage_r/value
      s_stage_y --> /entry/instrument/bluesky/streams/primary/s_stage_y/value
      scaler0_display_rate --> /entry/instrument/bluesky/streams/primary/scaler0_display_rate/value
      scaler0_time --> /entry/instrument/bluesky/streams/primary/scaler0_time/value
      seconds --> /entry/instrument/bluesky/streams/primary/seconds/value
      trd_autorange_controls_gain --> /entry/instrument/bluesky/streams/primary/trd_autorange_controls_gain/value
      trd_autorange_controls_reqrange --> /entry/instrument/bluesky/streams/primary/trd_autorange_controls_reqrange/value
      upd_autorange_controls_gain --> /entry/instrument/bluesky/streams/primary/upd_autorange_controls_gain/value
      upd_autorange_controls_reqrange --> /entry/instrument/bluesky/streams/primary/upd_autorange_controls_reqrange/value
    instrument:NXinstrument
      bluesky:NXnote
        plan_name --> /entry/instrument/bluesky/metadata/plan_name
        uid --> /entry/instrument/bluesky/metadata/run_start_uid
        metadata:NXnote
          EPICS_CA_MAX_ARRAY_BYTES:NX_CHAR = b'1280000'
          EPICS_HOST_ARCH:NX_CHAR = b'linux-x86_64'
          SAD_mm:NX_FLOAT64[] = [ ... ]
          SDD_mm:NX_FLOAT64[] = [ ... ]
          ay0:NX_FLOAT64[] = [ ... ]
          beamline_id:NX_CHAR = b'APS 9-ID-C USAXS'
          center:NX_FLOAT64[] = [ ... ]
          dy0:NX_FLOAT64[] = [ ... ]
          exponent:NX_FLOAT64[] = [ ... ]
          finish:NX_FLOAT64[] = [ ... ]
          intervals:NX_INT64[] = [ ... ]
          login_id:NX_CHAR = b'usaxs@usaxscontrol.xray.aps.anl.gov'
          minStep:NX_FLOAT64[] = [ ... ]
          pid:NX_INT64[] = [ ... ]
          plan_args:NX_CHAR = b'SAD_mm: 195.0\nSDD_mm: 900.0\nay0: 0.0\ncount_time: 0.5\ndy0: 12.794893\nexponent: 0.6\nfinish: 7.131764183414561\nintervals: 300\nminStep: 2.5e-05\nreference: 8.746958600000001\nstart: 8.748573740933907\n'
          plan_name:NX_CHAR = b'uascan'
          plan_type:NX_CHAR = b'generator'
          proposal_id:NX_CHAR = b'testing'
          run_start_uid:NX_CHAR = 78b4801c-1db1-4674-b894-03c4ef4d12a5
          start:NX_FLOAT64[] = [ ... ]
          uascan_direction:NX_INT64[] = [ ... ]
          uascan_factor:NX_FLOAT64[] = [ ... ]
          useDynamicTime:NX_CHAR = b'True'
[... 2403 lines cut ...]
        energy_taper --> /entry/instrument/bluesky/streams/baseline/undulator_downstream_energy_taper/value_start
        gap --> /entry/instrument/bluesky/streams/baseline/undulator_downstream_gap/value_start
        gap_taper --> /entry/instrument/bluesky/streams/baseline/undulator_downstream_gap_taper/value_start
        harmonic_value --> /entry/instrument/bluesky/streams/baseline/undulator_downstream_harmonic_value/value_start
        location --> /entry/instrument/bluesky/streams/baseline/undulator_downstream_location/value_start
        total_power --> /entry/instrument/bluesky/streams/baseline/undulator_downstream_total_power/value_start
        type:NX_CHAR = undulator
        version --> /entry/instrument/bluesky/streams/baseline/undulator_downstream_version/value_start
    sample:NXsample
      chemical_formula --> /entry/instrument/bluesky/streams/baseline/sample_data_chemical_formula/value_start
[... cut ...]
      stress_field --> /entry/instrument/bluesky/streams/baseline/sample_data_stress_field/value_start
      temperature --> /entry/instrument/bluesky/streams/baseline/sample_data_temperature/value_start
      thickness --> /entry/instrument/bluesky/streams/baseline/user_data_sample_thickness/value_start
      volume_fraction --> /entry/instrument/bluesky/streams/baseline/sample_data_volume_fraction/value_start
      x_translation --> /entry/instrument/bluesky/streams/baseline/sample_data_x_translation/value_start
```

</details>

prjemian, 2020-07-27:
Let's check with a new data collection tomorrow and identify if anything is missing.

prjemian, 2020-07-28:
Collecting new data now ...

prjemian, 2020-07-28:
```
from instrument.callbacks.nxwriter_usaxs import NXWriterUascan 
my_writer = NXWriterUascan()
uids = "b034490 e44696d".split()
for uid in uids:
    replay(db[uid], my_writer.receiver)
```

prjemian, 2020-07-28:
We found the content we expected.

closed 2020-07-28 by prjemian